ros2-java/ros2_java #1: More documentation

Theosakamg, 2016-07-17:
Thank for your code.

But can you explain the concept of this project :
- Proof of concept
or
- Future official implementation of Java for ROS2

Other questions :
Type of license associated to this project ?
I see you implement a other implementation (High-level implementation) in .net-core. It is a other POC or reorientation ?

Thank you for your answers.

esteve, 2016-08-23:
Thank you for your interest! I just made a first release of all code needed for writing Java and Android applications for ROS2, see https://groups.google.com/d/msg/ros-sig-ng-ros/y6m51E53Nyo/LTNUs9wADwAJ

I don't know if this may become the official implementation of Java for ROS2, it just happens to be the first one :-)

The goal of this project is to run ROS2 on Android, but along the way, you can also write applications that target the JVM on a PC.

I have a .NET wrapper in the works, based on .NET Core [1], mostly for UWP applications, but I haven't had time to polish it. @firesurfer is working on another implementation [2], which uses a different approach than `rcljava` and `rclpy`. Hope he can chime in and tell you more about his implementation.

1 - https://dotnet.github.io/
2 - https://github.com/firesurfer/rclcs

firesurfer, 2016-08-23:
Could you show us your .NET wrapper code on github (or somewhere else). I'm interested on how you've done the wrapping.

esteve, 2016-08-24:
@firesurfer I'm currently travelling and I left the .NET wrapper code on my Windows machine. Anyway, what I did was basically follow the same pattern as `rcljava`, using P/Invoke (instead of JNI) to call the unmanaged code from the managed land, and viceversa.

When I wrote the first prototype of the `rclpy` bindings at OSRF, I used a similar approach as yours to avoid the overhead of converting between C and Python. However, I soon found out that using `ctypes` and `cffi` to call the `rcl` API directly was too cumbersome and implementing nested types and arrays was difficult, so I ended up writing a glue layer in Python/C. Maybe things are different in .NET, so it might be possible to avoid that overhead, I don't know. In any case, `rcljava` follows the same approach as `rclpy` (besides the fact that Java is compiled), and `rcldotnet` was fairly straightforward given the similarities with Java.

I've requested a new category in the ROS discourse [1], I think it'd be a great place to have this kind of discussion.

And it'd be great to join efforts, let me know if you're interested. My usecase for `rcldotnet` is UWP applications, and Mono (I assume that's your usecase) would be great too.

1 - http://discourse.ros.org/t/request-for-new-rosdotnet-category/448

firesurfer, 2016-08-25:
I would really like to join efforts, I think it is more convenient for users to have one good working wrapper instead of multiple wrappers. 
My `rclcs` and `rosidl_generator_cs` are following the approach to have as little as possible overhead. I'm thinking that it should be possible to reproduce the same datatypes and especially struct layouts as the `rosidl_generator_c` generates. The interop abilities of .Net are more advanced than the abilities of Java as far as I know. 

My usecase at the moment is mono only (Mainly because I haven't done any development on Windows for a long time). I think it should be fairly easy to port my code to .net core because I'm mostly using the interop libs.

esteve, 2016-08-26:
@firesurfer now we have a category in the ROS discourse for client libraries thanks to @tfoote, I think it'd best to move the discussion there:

http://discourse.ros.org/t/net-bindings-for-ros2/460

Theosakamg, 2016-08-26:
Thank @esteve .

You link https://raw.githubusercontent.com/esteve/ament_java/master/ros2_java_desktop.repos in documentation return 404...
prefere https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop.repos

(after all my test, i make a merge request)

esteve, 2016-08-26:
@Theosakamg thanks for noticing! I just fixed the URLs https://github.com/esteve/ros2_java/commit/55710be597b1f32c283b20288d6b26fbe482b8dd

Theosakamg, 2016-08-27:
After full-install, my feedback : It works perfectly !! Thank.

My environment was : Debian Testing with Java 1.8 and python 3.5.2
I need to install with APT :  `python3-yaml python3-empy gradle`
and disable 2 extra package :  `ament build --symlink-install --isolated --skip-packages ros1_bridge pendulum_control`

Theosakamg, 2016-08-27:
On ros2_java_android build command
```
ament build --isolated --cmake-args \
  -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_TOOLCHAIN_FILE=$HOME/ros2_android_ws/src/android-cmake/android-cmake/android.toolchain.cmake \
  -DANDROID_FUNCTION_LEVEL_LINKING=OFF -DANDROID_NATIVE_API_LEVEL=android-21 -DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang -DANDROID_STL=gnustl_shared \
  -DANDROID_ABI=armeabi-v7a -DANDROID_NDK=$HOME/android_ndk/android-ndk-r12b -DTHIRDPARTY=ON -DCOMPILE_EXAMPLES=OFF -- \
  --ament-cmake-args \
  -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_TOOLCHAIN_FILE=$HOME/ros2_android_ws/src/android-cmake/android-cmake/android.toolchain.cmake \
  -DANDROID_FUNCTION_LEVEL_LINKING=OFF -DANDROID_NATIVE_API_LEVEL=android-21 -DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang -DANDROID_STL=gnustl_shared \
  -DANDROID_ABI=armeabi-v7a -DANDROID_NDK=$HOME/android_ndk/android-ndk-r12b -DTHIRDPARTY=ON -DCOMPILE_EXAMPLES=OFF --
  --ament-gradle-args \
  -Pament.android_stl=gnustl_shared -Pament.android_abi=armeabi-v7a -Pament.android_ndk=$HOME/android_ndk/android-ndk-r12b --
```

result with 
`... --ament-gradle-args : command not found`

add `\` on ligne 8.

esteve, 2016-08-30:
@Theosakamg thanks for trying out the Android stuff, I just updated the command in `README.md`. As for the skipped packages, did you follow the general ROS2 instructions (https://github.com/ros2/ros2/wiki)? `pendulum_control` and `ros1_bridge` should not be affected by the `ros2_java` code.

Do you mind if we move the conversation to http://discourse.ros.org/c/rosjava and/or http://discourse.ros.org/t/announcing-java-and-android-for-ros2/445 ? Discourse is a great place to discuss this sort of development.

closed 2016-08-30 by esteve