ros2-java/ros2_java #4: Need to resync Fast-RTPS C++11

Theosakamg, 2016-10-03:
For subproject : ros2_java_android.repos
On https://github.com/esteve/Fast-RTPS branch cpp11-android  (sorry, no issues available)

When you build :
```
Process package 'rmw_fastrtps_cpp' with context:
--------------------------------------------------------------------------------
 source_space => ..../src/eProsima/ROS-RMW-Fast-RTPS-cpp/rmw_fastrtps_cpp
  build_space => ..../build_isolated/rmw_fastrtps_cpp
install_space => ..../install_isolated/rmw_fastrtps_cpp
   make_flags => -j4, -l4
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'rmw_fastrtps_cpp'
==> '..../build_isolated/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '..../build_isolated/rmw_fastrtps_cpp'
==> '..../build_isolated/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/make -j4 -l4' in '..../build_isolated/rmw_fastrtps_cpp'
[ 50%] Building CXX object CMakeFiles/rmw_fastrtps_cpp.dir/src/functions.cpp.o
..../src/eProsima/ROS-RMW-Fast-RTPS-cpp/rmw_fastrtps_cpp/src/functions.cpp:2516:46: error: no member named 'getAttributes' in 'eprosima::fastrtps::Publisher'
            client_info->request_publisher_->getAttributes().topic.getTopicName().c_str(),
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^
1 error generated.
CMakeFiles/rmw_fastrtps_cpp.dir/build.make:62: recipe for target 'CMakeFiles/rmw_fastrtps_cpp.dir/src/functions.cpp.o' failed
make[2]: *** [CMakeFiles/rmw_fastrtps_cpp.dir/src/functions.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/rmw_fastrtps_cpp.dir/all' failed
make[1]: *** [CMakeFiles/rmw_fastrtps_cpp.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

Theosakamg, 2016-10-03:
link to https://github.com/eProsima/Fast-RTPS/pull/26

esteve, 2016-10-04:
This repository is only for the ROS2 Java bindings.

Please comment on the PR you linked so the upstream developers know that there are people using that branch and that it'd be beneficial to have it merged. Thanks.

closed 2016-10-04 by esteve