ros2-java/ros2_java #21: Conflict when running Desktop setup when RTI Connext is installed

spiderkeys, 2017-03-10:
Attempted to run the setup scripts for Desktop listed in the readme. I think I am running into a conflict because I have RTI Connext. Relevant output:

```sh
spiderkeys@spiderlair:~/ros2_java_ws$ ament build --symlink-install --isolated
# Topological order
 - ament_cmake_export_jars
 - connext_cmake_module
 - console_bridge
 - fastcdr
 - fastrtps
 - fastrtps_cmake_module
...
...
...
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.0.0 (/home/spiderkeys/ament_ws/install_isolated/ament_cmake/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found RTI Connext: /home/spiderkeys/rti_connext_dds-5.2.3
CMake Warning at cmake/Modules/FindConnext.cmake:195 (message):
  NDDSHOME set to '/home/spiderkeys/rti_connext_dds-5.2.3' but found multiple
  files named
  'libnddsc.so;libnddscore.so;libnddscpp.so;librticonnextmsgcpp.so' under
  '/home/spiderkeys/rti_connext_dds-5.2.3/lib':
  armv6vfphLinux3.xgcc4.7.2/libnddsc.so;x64Linux3gcc4.8.2/libnddsc.so;armv6vfphLinux3.xgcc4.7.2/libnddscore.so;x64Linux3gcc4.8.2/libnddscore.so;armv6vfphLinux3.xgcc4.7.2/libnddscpp.so;x64Linux3gcc4.8.2/libnddscpp.so;x64Linux3gcc4.8.2/librticonnextmsgcpp.so
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

CMake Warning at cmake/Modules/FindConnext.cmake:201 (message):
  NDDSHOME set to '/home/spiderkeys/rti_connext_dds-5.2.3' but found multiple
  files named
  'libnddscd.so;libnddscored.so;libnddscppd.so;librticonnextmsgcppd.so' under
  '/home/spiderkeys/rti_connext_dds-5.2.3/lib':
  armv6vfphLinux3.xgcc4.7.2/libnddscd.so;x64Linux3gcc4.8.2/libnddscd.so;armv6vfphLinux3.xgcc4.7.2/libnddscored.so;x64Linux3gcc4.8.2/libnddscored.so;armv6vfphLinux3.xgcc4.7.2/libnddscppd.so;x64Linux3gcc4.8.2/libnddscppd.so;x64Linux3gcc4.8.2/librticonnextmsgcppd.so
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

CMake Error at cmake/Modules/FindConnext.cmake:206 (message):
  NDDSHOME set to '/home/spiderkeys/rti_connext_dds-5.2.3' but could neither
  find all optimized libraries
  'libnddsc.so;libnddscore.so;libnddscpp.so;librticonnextmsgcpp.so' nor all
  debug libraries
  'libnddscd.so;libnddscored.so;libnddscppd.so;librticonnextmsgcppd.so' under
  '/home/spiderkeys/rti_connext_dds-5.2.3/lib':

  - optimized:
  armv6vfphLinux3.xgcc4.7.2/libnddsc.so;x64Linux3gcc4.8.2/libnddsc.so;armv6vfphLinux3.xgcc4.7.2/libnddscore.so;x64Linux3gcc4.8.2/libnddscore.so;armv6vfphLinux3.xgcc4.7.2/libnddscpp.so;x64Linux3gcc4.8.2/libnddscpp.so;x64Linux3gcc4.8.2/librticonnextmsgcpp.so

  - debug:
  armv6vfphLinux3.xgcc4.7.2/libnddscd.so;x64Linux3gcc4.8.2/libnddscd.so;armv6vfphLinux3.xgcc4.7.2/libnddscored.so;x64Linux3gcc4.8.2/libnddscored.so;armv6vfphLinux3.xgcc4.7.2/libnddscppd.so;x64Linux3gcc4.8.2/libnddscppd.so;x64Linux3gcc4.8.2/librticonnextmsgcppd.so
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/spiderkeys/ros2_java_ws/build_isolated/connext_cmake_module/CMakeFiles/CMakeOutput.log".

<== Command '. /home/spiderkeys/ros2_java_ws/build_isolated/connext_cmake_module/cmake__build.sh && /usr/bin/cmake /home/spiderkeys/ros2_java_ws/src/ros2/rmw_connext/connext_cmake_module -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/spiderkeys/ros2_java_ws/install_isolated/connext_cmake_module' failed in '/home/spiderkeys/ros2_java_ws/build_isolated/connext_cmake_module' with exit code '1'
<== Command '. /home/spiderkeys/ros2_java_ws/build_isolated/connext_cmake_module/cmake__build.sh && /usr/bin/cmake /home/spiderkeys/ros2_java_ws/src/ros2/rmw_connext/connext_cmake_module -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/spiderkeys/ros2_java_ws/install_isolated/connext_cmake_module' failed in '/home/spiderkeys/ros2_java_ws/build_isolated/connext_cmake_module' with exit code '1'
```

spiderkeys, 2017-03-10:
Removing the NDDSHOME variable such that Ament couldn't find RTI solved the problem.

esteve, 2017-03-11:
It seems the issue is caused by `NDDSHOME` pointing to a directory where multiple versions of RTI Connext are installed. Could you try pointing `NDDSHOME` to a directory where only x64 or armv are installed? That may solve it.

In any case, you might receive more/better feedback in the [rmw_connext](https://github.com/ros2/rmw_connext/) repo, `ros2_java` just reuses Connext's CMake machinery from `rmw_connext`.

spiderkeys, 2017-03-13:
Ah that makes sense, as I did have it installed for multiple arches. Will kick it over to rmw_connext if I run into the issue again, though I have uninstalled Connext for now.

closed 2017-03-13 by spiderkeys

esteve, 2017-03-13:
@spiderkeys to be honest, the code for finding the Connext libraries is a bit finicky and when it was first written, we didn't test it with libraries for other architectures besides x86_64. I recall testing `ros2_java` with Connext a little bit when I first started working on the Java bindings for ROS2, but I switched to Fast-RTPS soon after. Let me know if it works with Connext currently, it'd be great to verify `ros2_java` with more `rmw` implementations.

oemergenc, 2018-12-29:
Hey, it seems like i am running into something similar or at least something related to wrong RTI Connext libs. I am on Mac OS X High Sierra and trying to build the desktop version using the ci-scripts. Here is what i am doing:

`
export OSPL_HOME=/Users/omer.genc/ros2_install/opensplice/x86_64.darwin10_clang
export NDDSHOME=/Applications/rti_connext_dds-5.3.1
git clone https://github.com/esteve/ros2_java.git
cd ros2_java
chmod 755 ci-scripts/build_java.sh
./ci-scripts/build_java.sh
`
The build fails when it comes to the package **builtin_interfaces**. See the attached log file for the error. Any hints on this, would be greatly apprechiated.
[desktop-build.log](https://github.com/esteve/ros2_java/files/2716292/desktop-build.log)