ros2-java/ros2_java #22: FastRTPS cmake build fails to find third party libs

spiderkeys, 2017-03-10:
This may be the bug in the FastRTPS cmake scripts that was introduced a few commits ago (https://github.com/eProsima/Fast-RTPS/issues/80) and was fixed in https://github.com/eProsima/Fast-RTPS/commit/96cf1ea3c5cad63504c845b6145f9367f5b68939.

Relevant output:
```sh
Process package 'fastrtps' with context:
--------------------------------------------------------------------------------
 source_space => /home/spiderkeys/ros2_java_ws/src/eProsima/Fast-RTPS
  build_space => /home/spiderkeys/ros2_java_ws/build_isolated/fastrtps
install_space => /home/spiderkeys/ros2_java_ws/install_isolated/fastrtps
   make_flags => -j8, -l8
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'fastrtps'
Running cmake because arguments have changed.
==> '. /home/spiderkeys/ros2_java_ws/build_isolated/fastrtps/cmake__build.sh && /usr/bin/cmake /home/spiderkeys/ros2_java_ws/src/eProsima/Fast-RTPS -DCMAKE_INSTALL_PREFIX=/home/spiderkeys/ros2_java_ws/install_isolated/fastrtps' in '/home/spiderkeys/ros2_java_ws/build_isolated/fastrtps'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring 
-- Version: 1.3.1
-- To change de version modify the file configure.ac
-- Performing Test SUPPORTS_CXX11
-- Performing Test SUPPORTS_CXX11 - Success
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- fastcdr library found...
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find asio (missing: ASIO_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/Findasio.cmake:12 (find_package_handle_standard_args)
  cmake/dev/eprosima_libraries.cmake:171 (find_package)
  CMakeLists.txt:195 (find_eprosima_thirdparty)
```

esteve, 2017-03-11:
@spiderkeys Thanks for trying this out. What options did you pass to `ament`?

I just triggered a CI job (https://travis-ci.org/esteve/ros2_java/builds/197344715) to check that the current post-C++11-ification master of FastRTPS works with the desktop version of `ros2_java`. This is the command I used in the Travis CI job to build the `ros2_java` workspace:

`$ ament build --symlink-install --isolated --install-space install_isolated_java --build-space build_isolated_java --cmake-args -DEPROSIMA_BUILD=ON --`

I added `-DEPROSIMA_BUILD=ON` so that Travis would fetch the external Git submodules, but haven't tried building FastRTPS without passing any flags yet.

spiderkeys, 2017-03-13:
@esteve I had passed in the arguments described on this repo's readme, which did not include the -DEPROSIMA_BUILD=ON option, which I believe ended up being responsible for the errors above. After adding that, I now get further in the build process but am getting a new error. Steps to reproduce are below.

Commands:
```sh
mkdir -p ~/ros2_java_ws/src
cd ~/ros2_java_ws

wget https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop.repos
vcs import ~/ros2_java_ws/src < ros2_java_desktop.repos

cd ~/ros2_java_ws/src/ros2/rosidl_typesupport
patch -p1 < ../../ros2_java/ros2_java/rosidl_typesupport_ros2_java.patch

cd ~/ros2_java_ws

. ~/ament_ws/install_isolated/local_setup.sh
ament build --symlink-install --isolated --install-space install_isolated_java --build-space build_isolated_java --cmake-args -DEPROSIMA_BUILD=ON --
```

Output:
```sh
Process package 'builtin_interfaces' with context:
--------------------------------------------------------------------------------
 source_space => /home/spiderkeys/ros2_java_ws/src/ros2/common_interfaces/builtin_interfaces
  build_space => /home/spiderkeys/ros2_java_ws/build_isolated_java/builtin_interfaces
install_space => /home/spiderkeys/ros2_java_ws/install_isolated_java/builtin_interfaces
   make_flags => -j8, -l8
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'builtin_interfaces'
Running cmake because arguments have changed.
==> '. /home/spiderkeys/ros2_java_ws/build_isolated_java/builtin_interfaces/cmake__build.sh && /usr/bin/cmake /home/spiderkeys/ros2_java_ws/src/ros2/common_interfaces/builtin_interfaces -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DEPROSIMA_BUILD=ON -DCMAKE_INSTALL_PREFIX=/home/spiderkeys/ros2_java_ws/install_isolated_java/builtin_interfaces' in '/home/spiderkeys/ros2_java_ws/build_isolated_java/builtin_interfaces'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.0.0 (/home/spiderkeys/ament_ws/install_isolated/ament_cmake/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rosidl_default_generators: 0.0.0 (/home/spiderkeys/ros2_java_ws/install_isolated_java/rosidl_default_generators/share/rosidl_default_generators/cmake)
-- Using single matching rosidl_typesupport_c: rosidl_typesupport_introspection_c
-- Using single matching rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp
-- Found ament_cmake_export_jars: 0.0.0 (/home/spiderkeys/ros2_java_ws/install_isolated_java/ament_cmake_export_jars/share/ament_cmake_export_jars/cmake)
-- Found rmw_implementation_cmake: 0.0.0 (/home/spiderkeys/ros2_java_ws/install_isolated_java/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake)
-- Found rmw: 0.0.0 (/home/spiderkeys/ros2_java_ws/install_isolated_java/rmw/share/rmw/cmake)
-- Found rcljava_common: 0.0.0 (/home/spiderkeys/ros2_java_ws/install_isolated_java/rcljava_common/share/rcljava_common/cmake)
-- Found Java: /usr/bin/java (found version "1.8.0.121") found components:  Development 
-- Found JNI: /usr/lib/jvm/default-java/jre/lib/amd64/libjawt.so  
-- Found rmw_fastrtps_cpp: 0.0.0 (/home/spiderkeys/ros2_java_ws/install_isolated_java/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.5") 
-- Found python_cmake_module: 0.0.0 (/home/spiderkeys/ros2_java_ws/install_isolated_java/python_cmake_module/share/python_cmake_module/cmake)
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "3.5") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.5m
-- Using PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.5m.so
-- Found PythonExtra: .so  
-- Configuring done
-- Generating done
CMake Warning:
[... 39 lines cut ...]
make[2]: *** [rosidl_generator_cpp/builtin_interfaces/msg/duration.hpp] Error 1
make[2]: *** [rosidl_generator_java/builtin_interfaces/msg/Duration.java] Error 1
CMakeFiles/Makefile2:483: recipe for target 'CMakeFiles/builtin_interfaces__cpp.dir/all' failed
make[1]: *** [CMakeFiles/builtin_interfaces__cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:304: recipe for target 'CMakeFiles/builtin_interfaces__java.dir/all' failed
make[1]: *** [CMakeFiles/builtin_interfaces__java.dir/all] Error 2
Traceback (most recent call last):
  File "/home/spiderkeys/ros2_java_ws/install_isolated_java/rosidl_generator_py/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py'
[... cut ...]
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/spiderkeys/ros2_java_ws/build_isolated_java/builtin_interfaces/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/spiderkeys/ros2_java_ws/build_isolated_java/builtin_interfaces' with exit code '2'
<== Command '. /home/spiderkeys/ros2_java_ws/build_isolated_java/builtin_interfaces/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/spiderkeys/ros2_java_ws/build_isolated_java/builtin_interfaces' with exit code '2'
```

spiderkeys, 2017-03-13:
It looks like a missing dependency on the python module 'empy', however, installing that did not resolve the issue. Wondering if this is some python2.7 vs python3 issue.

esteve, 2017-03-13:
I'm reopening this as a reminder that there's an issue in the  documentation, will fix it tomorrow. Thanks for noticing!

TODO:
- Update CMake incantation with `-DEPROSIMA_BUILD=ON`
- Mention requirement of NDK 14
- Remove reference to Fast-RTPS C++11 fork

esteve, 2017-03-14:
@spiderkeys I've pushed https://github.com/esteve/ros2_java/pull/23 which now has details about the `-DEPROSIMA_BUILD=ON` flag and NDK 14, let me know if that clarifies the build steps a bit. Thanks!

closed 2017-03-15 by esteve

spiderkeys, 2017-03-15:
Great, I'll walk through the new instructions and see if I run into any more problems with the ament build.