ros2-java/ros2_java #27: DCMA ros2java-alfred/ament_rcljava_gradle

Theosakamg, 2017-03-24:
Esteve you request a DCMA for this repository but it was write before yours !

Please modify your request... (You did not even contact us for the problem of DCMA. This is really disappointing)

esteve, 2017-03-24:
You know that's not true, I have contacted you multiple times regarding the copyright violations and other issues, both publicly and privately, and haven't received a response.

It doesn't matter if you created the project earlier, when you migrated the plugin from Java to Groovy, you took the code from my plugin, removed the copyright and claimed it as yours. Also, registering it on the Gradle Plugin registry as org.ros2.rcljava was highly sketchy, would it have been that hard to mention that it's for the Alfred project? Naming it org.ros2.rcljava_alfred would have been much less misleading.

I'm working on this in my spare time and for free, I'm giving away all the code, you can even include it in a commercial product, make money and don't have to give anything back. However, the Apache License is very clear, you can't remove the copyright, claim the work is yours and not reference the original code. Violating the terms of an opensource license is really frowned upon in the community. 

The DMCA notice was a last resort, and it's something that made me very sad, but it was the only thing left I could do since you weren't willing to act upon the license violations.

closed 2017-03-24 by esteve

esteve, 2017-03-24:
Just to clarify something, I have no interest in taking down your fork, so as I repeatedly asked you before, if you make the changes and comply with the license, I'm totally fine asking GitHub to reenable your repositories. The whole situation just makes me sad, I take absolutely no pleasure in disabling other people's code.

Theosakamg, 2017-03-25:
No problem Esteve ! I understand, and my apology for the no respect of License. I request to maintainer to make the modification.