ros2-java/ros2_java #30: RVIZ

jabrena, 2017-04-27:
Hi mate,

This weekend, I will test your project.
I would like to know if exist a version of rviz for ros2?

Cheers

esteve, 2017-04-29:
Awesome! Let me know what you think of `ros2_java` and if it fits your needs. I noticed your `ev3dev-lang-java` project, it'd be so cool to have `ros2_java` running on the EV3, I keep meaning to work on/play with my EV3 again :-)

You can try using `rviz` with the [ros1_bridge](https://github.com/ros2/ros1_bridge), which enables you to use the existing ROS1 tools and applications with ROS2.

If you have further questions about rviz and ROS2, feel free to post them on the [ros-ng forum](https://discourse.ros.org/c/ng-ros), where all of the discussions regarding ROS2 are happening.

Thanks for trying out `ros2_java` :-)

jabrena, 2017-04-29:
Hi @esteve, Yes this weekend I will test `ros_java` if the POC goes well, I will write the whole remoting support with ros2_java.

Give me 24h.

Cheers

Juan Antonio

esteve, 2017-04-29:
Sure, no problem. Unfortunately the documentation for `ros2_java` is almost non-existing, but the examples for Java and Android are fairly self explanatory and exercise all of the API:

https://github.com/esteve/ros2_java_examples
https://github.com/esteve/ros2_android_examples

Let me know if you need any help.

jabrena, 2017-04-29:
I will work with this robot to test the library:
https://www.youtube.com/watch?v=1d9q24aNMHQ

jabrena, 2017-04-29:
Hi, I am trying to install the ROS2 environment in local, but I have the following issues with the first block of instructions.

When I try to install ROS2:

```
src/ament/ament_tools/scripts/ament.py build --symlink-install --isolated
```

I had to install latest version of cmake manually and the path is:
/Applications/CMake.app/Contents/bin

When I execute:

./src/ament/ament_tools/scripts/ament.py build --symlink-install --isolated

I receive the following error:

``` bash:

``` 
➜  ament_ws python3 ./src/ament/ament_tools/scripts/ament.py build --symlink-install --isolated
# Topological order
 - ament_cppcheck
 - ament_flake8
 - ament_package
 - ament_cmake_core
 - ament_cmake_export_definitions
 - ament_cmake_export_dependencies
 - ament_cmake_export_include_directories
 - ament_cmake_export_interfaces
 - ament_cmake_export_libraries
 - ament_cmake_export_link_flags
 - ament_cmake_gmock
 - ament_cmake_gtest
 - ament_cmake_include_directories
 - ament_cmake_libraries
 - ament_cmake
 - ament_cmake_nose
 - ament_cmake_python
 - ament_cmake_target_dependencies
 - ament_cmake_test
 - ament_cmake_auto
 - ament_lint_auto
 - ament_pep8
 - ament_pep257
 - ament_pyflakes
 - ament_copyright
 - ament_clang_format
 - ament_cpplint
 - ament_index_python
 - ament_lint_cmake
 - ament_cmake_lint_cmake
 - ament_cmake_copyright
 - ament_cmake_clang_format
 - ament_cmake_cppcheck
 - ament_cmake_cpplint
 - ament_cmake_flake8
 - ament_cmake_pep257
 - ament_cmake_pep8
 - ament_cmake_pyflakes
 - ament_cmake_uncrustify
 - gmock_vendor
 - gtest_vendor
 - osrf_pycommon
 - ament_tools
 - ament_build_type_gradle
 - uncrustify
 - ament_uncrustify
 - ament_lint_common
 - ament_index_cpp
[... 106 lines cut ...]
reading manifest file 'ament_package.egg-info/SOURCES.txt'
writing manifest file 'ament_package.egg-info/SOURCES.txt'
running build_ext
Creating /Users/jabrena/ament_ws/install_isolated/ament_package/lib/python3.6/site-packages/ament-package.egg-link (link to .)
ament-package 0.0.0 is already the active version in easy-install.pth

Installed /Users/jabrena/ament_ws/build_isolated/ament_package
-- [ament] Deploying: /Users/jabrena/ament_ws/install_isolated/ament_package/local_setup.bash
-- [ament] Deploying: /Users/jabrena/ament_ws/install_isolated/ament_package/local_setup.sh
-- [ament] Deploying: /Users/jabrena/ament_ws/install_isolated/ament_package/local_setup.zsh
-- [ament] Deploying: /Users/jabrena/ament_ws/install_isolated/ament_package/setup.bash
-- [ament] Deploying: /Users/jabrena/ament_ws/install_isolated/ament_package/setup.sh
-- [ament] Deploying: /Users/jabrena/ament_ws/install_isolated/ament_package/setup.zsh
-- [ament] Deploying: /Users/jabrena/ament_ws/install_isolated/ament_package/_order_packages.py

Process package 'ament_cmake_core' with context:
--------------------------------------------------------------------------------
 source_space => /Users/jabrena/ament_ws/src/ament/ament_cmake/ament_cmake_core
  build_space => /Users/jabrena/ament_ws/build_isolated/ament_cmake_core
install_space => /Users/jabrena/ament_ws/install_isolated/ament_cmake_core
   make_flags => -j4, -l4
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'ament_cmake_core'
ament build: Could not find 'cmake' executable

jabrena, 2017-04-29:
Fixed:

``` 
PATH="/Applications/CMake.app/Contents/bin":"$PATH"
```

jabrena, 2017-04-29:
When I am finishing the installation:

``` 
➜  ros2_java_ws . ~/ament_ws/install_isolated/local_setup.sh
➜  ros2_java_ws ament build --symlink-install --isolated
zsh: command not found: ament
``` 

zsh doesn´t recognize ament. Where is the executable?

Theosakamg, 2017-04-29:
Can you try :
`cd /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS && git submodule init && git submodule update`

jabrena, 2017-04-29:
Hi @Theosakamg, with that command, I could start the building but in the last moment appeared this message:

``` 
[100%] Linking CXX shared library libfastrtps.dylib
Undefined symbols for architecture x86_64:
  "tinyxml2::XMLDocument::LoadFile(char const*)", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in EDPStaticXML.cpp.o
  "tinyxml2::XMLDocument::XMLDocument(bool, tinyxml2::Whitespace)", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in EDPStaticXML.cpp.o
  "tinyxml2::XMLDocument::~XMLDocument()", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in EDPStaticXML.cpp.o
  "tinyxml2::XMLElement::FindAttribute(char const*) const", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
  "tinyxml2::XMLElement::GetText() const", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLParticipantEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
  "tinyxml2::XMLElement::Attribute(char const*, char const*) const", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
  "tinyxml2::XMLAttribute::QueryIntValue(int*) const", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
  "tinyxml2::XMLNode::FirstChildElement(char const*) const", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLParticipantEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
  "tinyxml2::XMLNode::NextSiblingElement(char const*) const", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLParticipantEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
  "tinyxml2::XMLNode::Value() const", referenced from:
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLParticipantEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLReaderEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/cpp/libfastrtps.1.4.0.dylib] Error 1
make[1]: *** [src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
make: *** [all] Error 2

<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/fastrtps/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/fastrtps' with exit code '2'
<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/fastrtps/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/fastrtps' with exit code '2'
```

Theosakamg, 2017-04-29:
Do you clean your previous build  ? Do you have the full log of build of Fast-RTPS ?

jabrena, 2017-04-29:
Hi @Theosakamg, sorry I am not a super expert compiling this kind of software. How to to clean the whole building again? I will compile everything again.

Theosakamg, 2017-04-29:
For clean the previous fastrtps build with :
```
rm -rf /Users/jabrena/ros2_java_ws/build_isolated/fastrtps
rm -rf /Users/jabrena/ros2_java_ws/install_isolated/fastrtps
```

for clean all previous build with :
```
rm -rf /Users/jabrena/ros2_java_ws/build_isolated
rm -rf /Users/jabrena/ros2_java_ws/install_isolated
```

And what is you environment  ? ubuntu ?

jabrena, 2017-04-29:
OSX, I go with the clean process

jabrena, 2017-04-29:
I clean previous build, but the process repeated. Where is the log about full log of build of Fast-RTPS?

Console log:
```
-- Found asio: /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include  
-- Could NOT find tinyxml2 (missing:  TINYXML2_LIBRARY) 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   fastrtps

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/jabrena/ros2_java_ws/build_isolated/fastrtps
==> '. /Users/jabrena/ros2_java_ws/build_isolated/fastrtps/cmake__build.sh && /usr/bin/make -j4 -l4' in '/Users/jabrena/ros2_java_ws/build_isolated/fastrtps'
Scanning dependencies of target fastrtps
[  2%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/log/StdoutConsumer.cpp.o
[  3%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/log/Log.cpp.o
[  3%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/utils/eClock.cpp.o
[  4%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/utils/IPFinder.cpp.o
[  5%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/utils/md5.cpp.o
[  7%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/utils/StringMatching.cpp.o
[  8%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/ResourceEvent.cpp.o
[  9%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/TimedEvent.cpp.o
[ 10%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/TimedEventImpl.cpp.o
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/src/cpp/rtps/resources/ResourceEvent.cpp:20:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/include/fastrtps/rtps/resources/ResourceEvent.h:25:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio.hpp:19:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/associated_executor.hpp:21:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/system_executor.hpp:148:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/impl/system_executor.hpp:18:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/detail/executor_op.hpp:19:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/detail/fenced_block.hpp:24:
/Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/detail/macos_fenced_block.hpp:45:5: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
    OSMemoryBarrier();
    ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has been explicitly marked deprecated here
void    OSMemoryBarrier( void );
        ^
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/src/cpp/rtps/resources/ResourceEvent.cpp:20:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/include/fastrtps/rtps/resources/ResourceEvent.h:25:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio.hpp:19:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/associated_executor.hpp:21:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/system_executor.hpp:148:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/impl/system_executor.hpp:18:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/detail/executor_op.hpp:19:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/detail/fenced_block.hpp:24:
/Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/thirdparty/asio/asio/include/asio/detail/macos_fenced_block.hpp:51:5: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
    OSMemoryBarrier();
    ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has been explicitly marked deprecated here
void    OSMemoryBarrier( void );
        ^
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/src/cpp/rtps/resources/TimedEvent.cpp:20:
In file included from /Users/jabrena/ros2_java_ws/src/eProsima/Fast-RTPS/include/fastrtps/rtps/resources/TimedEvent.h:25:
[... 1599 lines cut ...]
      eprosima::fastrtps::rtps::EDPStaticXML::loadXMLWriterEndpoint(tinyxml2::XMLElement*, eprosima::fastrtps::rtps::StaticRTPSParticipantInfo*) in EDPStaticXML.cpp.o
  "tinyxml2::XMLNode::FirstChildElement(char const*) const", referenced from:
[... cut ...]
make[2]: *** [src/cpp/libfastrtps.1.4.0.dylib] Error 1
make[1]: *** [src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
make: *** [all] Error 2

<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/fastrtps/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/fastrtps' with exit code '2'
<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/fastrtps/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/fastrtps' with exit code '2'
```

jabrena, 2017-04-29:
I think that the problem is here:

```
-- Could NOT find tinyxml2 (missing:  TINYXML2_LIBRARY)
```

esteve, 2017-04-29:
@jabrena the instructions that @Theosakamg gave you were only necessary when Fast-RTPS still didn't have the changes for C+11 merged properly, but that's not the case anymore, so don't call `git submodule ...` and instead install the `asio` and `tinyxml2` packages via `brew`.

I'm sorry the README doesn't mention it explicitly, but in order to build `ros2_java` you'll need to follow the normal ROS2 steps and install all the dependencies:

https://github.com/ros2/ros2/wiki/OSX-Development-Setup#install-prerequisites

I'll update the `README.md` file tomorrow to mention those.

jabrena, 2017-04-29:
Hi @esteve, the problem with asio was fixed but not with tinyxml2, I go with tinyxml2 using brew. 
I go with the new iteration :)

esteve, 2017-04-29:
@jabrena could you remove the entire workspace and start over without the `git submodule ...` command?  And also install all the dependencies via `brew` (including `cmake`), just in case. That'd make it easier to track down any potential issue.

Theosakamg, 2017-04-29:
Other idea :
now ros2.repo embeded the tinyxml
https://github.com/ros2/ros2/blob/master/ros2.repos#L166

esteve, 2017-04-29:
@Theosakamg no, the `tinyxml2_vendor` package uses the system-installed `tinyxml2` package, so @jabrena still needs to install it via brew.

jabrena, 2017-04-29:
Hi mates,

I am repeating the whole block:

``` 
mkdir -p ~/ros2_java_ws/src
cd ~/ros2_java_ws
wget https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop.repos
vcs import ~/ros2_java_ws/src < ros2_java_desktop.repos
cd ~/ros2_java_ws/src/ros2/rosidl_typesupport
patch -p1 < ../../ros2_java/ros2_java/rosidl_typesupport_ros2_java.patch
cd ~/ros2_java_ws
. ~/ament_ws/install_isolated/local_setup.zsh
ament build --symlink-install --isolated
```

I have just installed `asio` and `tinyxml2` using brew.

Compiling now.

esteve, 2017-04-29:
@jabrena you'll have to fetch the `tinyxml_vendor` repository into the workspace (or use this updated version of the repos file https://github.com/esteve/ros2_java/blob/master/ros2_java_desktop.repos) and install the `tinyxml2` package.

jabrena, 2017-04-29:
Repeating again with `tinyxml_vendor` 
if everything run, I will help you with documentation. :)

jabrena, 2017-04-29:
Hi @esteve,

The process stopped here:

``` 
[  6%] Generating Python code for ROS interfaces
Traceback (most recent call last):
  File "/Users/jabrena/ros2_java_ws/install_isolated/rosidl_generator_c/share/rosidl_generator_c/cmake/../../../lib/rosidl_generator_c/rosidl_generator_c", line 8, in <module>
    from rosidl_generator_c import generate_c
  File "/Users/jabrena/ros2_java_ws/install_isolated/rosidl_generator_c/lib/python3.6/site-packages/rosidl_generator_c/__init__.py", line 17, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/Users/jabrena/ros2_java_ws/install_isolated/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 21, in <module>
    import em
ModuleNotFoundError: No module named 'em'
make[2]: *** [rosidl_generator_c/builtin_interfaces/msg/duration.h] Error 1
make[1]: *** [CMakeFiles/builtin_interfaces__rosidl_generator_c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/Users/jabrena/ros2_java_ws/install_isolated/rosidl_generator_cpp/share/rosidl_generator_cpp/cmake/../../../lib/rosidl_generator_cpp/rosidl_generator_cpp", line 8, in <module>
    from rosidl_generator_cpp import generate_cpp
  File "/Users/jabrena/ros2_java_ws/install_isolated/rosidl_generator_cpp/lib/python3.6/site-packages/rosidl_generator_cpp/__init__.py", line 17, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/Users/jabrena/ros2_java_ws/install_isolated/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 21, in <module>
    import em
ModuleNotFoundError: No module named 'em'
make[2]: *** [rosidl_generator_cpp/builtin_interfaces/msg/duration.hpp] Error 1
make[1]: *** [CMakeFiles/builtin_interfaces__cpp.dir/all] Error 2
Traceback (most recent call last):
  File "/Users/jabrena/ros2_java_ws/install_isolated/rosidl_generator_py/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py'
make[2]: *** [rosidl_generator_py/builtin_interfaces/_builtin_interfaces_s.ep.rosidl_typesupport_c.c] Error 1
make[1]: *** [builtin_interfaces__py/CMakeFiles/builtin_interfaces__py.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!

<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces' with exit code '2'
<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces' with exit code '2'
```

Theosakamg, 2017-04-29:
Python empy is needed. (from documentation)
`python3 -m pip install empy setuptools nose vcstool pep8 pydocstyle pyflakes flake8 mock coverage`

jabrena, 2017-04-29:
@Theosakamg many thanks, I repeat again :D 

https://en.wikipedia.org/wiki/Groundhog_Day

jabrena, 2017-04-29:
I improved the progress but I continue with problems:

``` 
-- Found rcljava_common: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/cmake)
CMake Warning at /Applications/CMake.app/Contents/share/cmake-3.8/Modules/FindJava.cmake:154 (message):
  regex not supported: java version "9-ea"

  Java(TM) SE Runtime Environment (build 9-ea+163)

  Java HotSpot(TM) 64-Bit Server VM (build 9-ea+163, mixed mode).  Please
  report
Call Stack (most recent call first):
  /Users/jabrena/ros2_java_ws/install_isolated/rosidl_generator_java/share/rosidl_generator_java/cmake/rosidl_generator_java_generate_interfaces.cmake:26 (find_package)
  /Users/jabrena/ament_ws/install_isolated/ament_cmake_core/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  /Users/jabrena/ros2_java_ws/install_isolated/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:150 (ament_execute_extensions)
  CMakeLists.txt:12 (rosidl_generate_interfaces)

-- Found Java: /Library/Java/JavaVirtualMachines/jdk-9.jdk/Contents/Home/bin/java (found version "..") found components:  Development 
-- Found JNI: /System/Library/Frameworks/JavaVM.framework  
-- Found rmw_fastrtps_cpp: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp/cmake)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.6.1", minimum required is "3.5") 
-- Found python_cmake_module: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/python_cmake_module/share/python_cmake_module/cmake)
-- Using PythonExtra_INCLUDE_DIRS: /usr/local/Cellar/python3/3.6.1/Frameworks/Python.framework/Versions/3.6/include/python3.6m;/usr/local/Cellar/python3/3.6.1/Frameworks/Python.framework/Versions/3.6/include/python3.6m
-- Using PythonExtra_LIBRARIES: /usr/local/opt/python3/Frameworks/Python.framework/Versions/3.6/lib/python3.6/config-3.6m-darwin/libpython3.6m.dylib
-- Found PythonExtra: .so  
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces
==> '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces'
[  2%] Generating C code for ROS interfaces
Scanning dependencies of target builtin_interfaces__cpp
Scanning dependencies of target builtin_interfaces__py
[  4%] Generating C++ code for ROS interfaces
[  6%] Generating Python code for ROS interfaces
[  6%] Built target builtin_interfaces__cpp
[  8%] Generating C++ introspection for ROS interfaces
Scanning dependencies of target builtin_interfaces__java
[ 10%] Generating Java code for ROS interfaces
Scanning dependencies of target builtin_interfaces__rosidl_generator_c
[ 12%] Building C object CMakeFiles/builtin_interfaces__rosidl_generator_c.dir/rosidl_generator_c/builtin_interfaces/msg/duration__functions.c.o
[ 14%] Building C object CMakeFiles/builtin_interfaces__rosidl_generator_c.dir/rosidl_generator_c/builtin_interfaces/msg/time__functions.c.o
[ 17%] Linking C shared library libbuiltin_interfaces__rosidl_generator_c.dylib
ld: unknown option: --no-undefined
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libbuiltin_interfaces__rosidl_generator_c.dylib] Error 1
make[1]: *** [CMakeFiles/builtin_interfaces__rosidl_generator_c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Scanning dependencies of target builtin_interfaces__rosidl_typesupport_introspection_cpp
[ 21%] Building CXX object CMakeFiles/builtin_interfaces__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/builtin_interfaces/msg/time__type_support.cpp.o
[ 21%] Building CXX object CMakeFiles/builtin_interfaces__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/builtin_interfaces/msg/duration__type_support.cpp.o
[ 23%] Linking CXX shared library libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.dylib
ld: unknown option: --no-undefined
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.dylib] Error 1
make[1]: *** [CMakeFiles/builtin_interfaces__rosidl_typesupport_introspection_cpp.dir/all] Error 2
[ 23%] Built target builtin_interfaces__py
[ 23%] Built target builtin_interfaces__java
make: *** [all] Error 2

<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces' with exit code '2'
<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces' with exit code '2'
➜  ros2_java_ws 
```

Theosakamg, 2017-04-29:
ROS2 (on OSX) need specific version of JDK 1.8. 
http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html

jabrena, 2017-04-29:
Tomorrow, I will switch `Java 9 ea` to a stable version of `Java 8`.

Good night @esteve & @Theosakamg 

Cheers

Juan Antonio

jabrena, 2017-04-29:
I uninstalled the early version of Java 9. 

``` 
+++ Building 'builtin_interfaces'
Running cmake because arguments have changed.
==> '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /Applications/CMake.app/Contents/bin/cmake /Users/jabrena/ros2_java_ws/src/ros2/common_interfaces/builtin_interfaces -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/Users/jabrena/ros2_java_ws/install_isolated/builtin_interfaces -G Unix Makefiles' in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces'
-- The C compiler identification is AppleClang 8.1.0.8020042
-- The CXX compiler identification is AppleClang 8.1.0.8020042
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.0.0 (/Users/jabrena/ament_ws/install_isolated/ament_cmake/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.6.1", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/local/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rosidl_default_generators: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/rosidl_default_generators/share/rosidl_default_generators/cmake)
-- Using single matching rosidl_typesupport_c: rosidl_typesupport_introspection_c
-- Using single matching rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp
-- Found ament_cmake_export_jars: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/ament_cmake_export_jars/share/ament_cmake_export_jars/cmake)
-- Found rmw_implementation_cmake: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake)
-- Found rmw: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/rmw/share/rmw/cmake)
-- Found rcljava_common: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/cmake)
-- Found Java: /Library/Java/JavaVirtualMachines/jdk1.8.0_121.jdk/Contents/Home/bin/java (found version "1.8.0.121") found components:  Development 
-- Found JNI: /System/Library/Frameworks/JavaVM.framework  
-- Found rmw_fastrtps_cpp: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp/cmake)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.6.1", minimum required is "3.5") 
-- Found python_cmake_module: 0.0.0 (/Users/jabrena/ros2_java_ws/install_isolated/python_cmake_module/share/python_cmake_module/cmake)
-- Using PythonExtra_INCLUDE_DIRS: /usr/local/Cellar/python3/3.6.1/Frameworks/Python.framework/Versions/3.6/include/python3.6m;/usr/local/Cellar/python3/3.6.1/Frameworks/Python.framework/Versions/3.6/include/python3.6m
-- Using PythonExtra_LIBRARIES: /usr/local/opt/python3/Frameworks/Python.framework/Versions/3.6/lib/python3.6/config-3.6m-darwin/libpython3.6m.dylib
-- Found PythonExtra: .so  
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces
==> '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces'
Scanning dependencies of target builtin_interfaces__py
Scanning dependencies of target builtin_interfaces__cpp
[  6%] Generating C++ code for ROS interfaces
[  6%] Generating Python code for ROS interfaces
[  6%] Generating C code for ROS interfaces
[  6%] Built target builtin_interfaces__cpp
[  8%] Generating C++ introspection for ROS interfaces
Scanning dependencies of target builtin_interfaces__java
[ 10%] Generating Java code for ROS interfaces
Scanning dependencies of target builtin_interfaces__rosidl_generator_c
[ 12%] Building C object CMakeFiles/builtin_interfaces__rosidl_generator_c.dir/rosidl_generator_c/builtin_interfaces/msg/duration__functions.c.o
[ 14%] Building C object CMakeFiles/builtin_interfaces__rosidl_generator_c.dir/rosidl_generator_c/builtin_interfaces/msg/time__functions.c.o
[ 17%] Linking C shared library libbuiltin_interfaces__rosidl_generator_c.dylib
ld: unknown option: --no-undefined
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libbuiltin_interfaces__rosidl_generator_c.dylib] Error 1
make[1]: *** [CMakeFiles/builtin_interfaces__rosidl_generator_c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Scanning dependencies of target builtin_interfaces__rosidl_typesupport_introspection_cpp
[ 21%] Building CXX object CMakeFiles/builtin_interfaces__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/builtin_interfaces/msg/duration__type_support.cpp.o
[ 21%] Building CXX object CMakeFiles/builtin_interfaces__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/builtin_interfaces/msg/time__type_support.cpp.o
[... cut ...]
[ 23%] Built target builtin_interfaces__py
[ 23%] Built target builtin_interfaces__java
make: *** [all] Error 2

<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces' with exit code '2'
<== Command '. /Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/Users/jabrena/ros2_java_ws/build_isolated/builtin_interfaces' with exit code '2'
```

esteve, 2017-04-30:
@jabrena I've set up a Travis project to build `ros2_java` on macOS:

https://travis-ci.org/esteve/ros2_java_osx
https://github.com/esteve/ros2_java_osx/tree/travis

The following steps compile `ros2_java` on macOS successfully, maybe they can help you setting up the dependencies on your computer:

https://github.com/esteve/ros2_java_osx/blob/travis/.travis.yml

unfortunately, the tests don't pass, so I'm going to investigate a bit. Let me know if you manage to compile `ros2_java` on macOS, maybe we can find out what's making the tests fail. Thanks!

esteve, 2017-04-30:
@jabrena I've found http://stackoverflow.com/a/14905563, which may be the reason why JNI libraries are not loaded on macOS. `ros2_java` uses CMake's `add_library()` to generate the JNI libraries, and I'm guessing it defaults to `.so`, instead of using `.jnilib` as extension.

jabrena, 2017-04-30:
Hi @esteve, yes I want to collaborate in this project. If we solve this issue, you have 2 additional hands for issues. :)

Another alternative is if with a docker image could solve my problems. If some linux image build the infrastructure. using ssh to that machine could enough. But this solution would be valid for a Desktop, the host, but later, we should find a solution for a Raspi (armhf) 3 & EV3Dev (armel)

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/214

esteve, 2017-04-30:
@jabrena it turns out that the tests are just flaky, seems to be an issue with DDS. Anyway, if you follow the instructions from the Travis file and the ROS2 documentation, you should be able to build `ros2_java` locally and run the tests.

If you want to give Docker a shot, here are the Dockerfiles I use for running the Travis builds:

https://github.com/esteve/ros2_java_docker

If you don't mind, I'm going to close this issue since it's deviating from the original intent, but feel free to open more if you find any problems. Thanks!

closed 2017-04-30 by esteve

jabrena, 2017-04-30:
Hi @esteve, I will review travis instructions and the docker image. I will review.

Cheers