ros2-java/ros2_java #31: RCUtils error Ubuntu 14.04

sauravsahay, 2017-10-18:
I am trying to install ros2_java and I followed all steps but get this issue. It gets stuck at the following step. I have installed empy and python3-empy. 

+++ Building 'rcutils'
==> '. /home/ssahay/ros2_java_ws/build_isolated/rcutils/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/ssahay/ros2_java_ws/build_isolated/rcutils'
==> '. /home/ssahay/ros2_java_ws/build_isolated/rcutils/cmake__build.sh && /usr/bin/make -j56 -l56' in '/home/ssahay/ros2_java_ws/build_isolated/rcutils'
[  5%] Expanding logging_macros.h.em
/home/ssahay/ros2_java_ws/src/ros2/rcutils/resource/logging_macros.h.em:192: error: <class 'SyntaxError'>: invalid syntax (logging.py, line 110)
make[2]: *** [include/rcutils/logging_macros.h] Error 1
make[2]: *** Deleting file `include/rcutils/logging_macros.h'
make[1]: *** [CMakeFiles/rcutils.dir/all] Error 2
make: *** [all] Error 2

Please suggest.

esteve, 2017-10-25:
Thanks for checking out `ros2_java`. `rcutils` is part of the main ROS2 tree and is separate from `ros2_java`. In any case, ROS2 no longer supports Ubuntu 14.04, the C/C++ compiler is too old and many other dependencies are either outdated or not available. Could you try building again on Ubuntu 16.04?

sauravsahay, 2017-11-01:
Thanks for the info - I will try it on Ubuntu 16.04 next week and update here.

closed 2017-11-01 by sauravsahay

sauravsahay, 2017-11-29:
The install works on Ubuntu 16.04 
Had to install these prereqs in the process:
vcs
libasio-dev
libtinyxml2
empy (via pip)
gradle
liborocos-kdl-dev
libeigen3-dev

Thanks!

esteve, 2017-11-29:
Thanks for feedback!