ros2-java/ros2_java #32: Could not find a configuration file for "fastcdr"

eborghi10, 2017-10-30:
Hi, I'm install `ros2_java` according to [this Installation link](https://github.com/esteve/ros2_java#android) for Android, previous installing _CMake 3.10.20171027, Python3, VCS_ and other required dependencies from [here](https://github.com/ros2/ros2/wiki/Linux-Development-Setup#how-to-setup-the-development-environment). 

I also installed **Android NDK r15c** for the last step from [here](https://developer.android.com/ndk/downloads/index.html). 

In the last step of the tutorial the path from NDK is required so I changed r14 to r15c. Besides, I'm not quite sure how to check the ANDROID_NATIVE_API_LEVEL flag.

```bash
ament build --isolated --cmake-args \
  -DPYTHON_EXECUTABLE=/usr/bin/python3 \
  -DEPROSIMA_BUILD=ON \
  -DCMAKE_FIND_ROOT_PATH="$HOME/ament_ws/install_isolated;$HOME/ros2_android_ws/install_isolated" \
  -DANDROID_FUNCTION_LEVEL_LINKING=OFF \
  -DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang -DANDROID_STL=gnustl_shared \
  -DANDROID_ABI=armeabi-v7a \
  -DANDROID_NDK=$HOME/android_ndk/android-ndk-r15c \
  -DANDROID_NATIVE_API_LEVEL=android-21 \
  -DCMAKE_TOOLCHAIN_FILE=$HOME/android_ndk/android-ndk-r15c/build/cmake/android.toolchain.cmake \
  -- \
  --ament-gradle-args \
  -Pament.android_stl=gnustl_shared -Pament.android_abi=armeabi-v7a -Pament.android_ndk=$HOME/android_ndk/android-ndk-r15c --
```

After executing this step, I get the following error:

```bash
[ 43%] Creating directories for 'SampleConfig_Multimedia'
-- Configuring incomplete, errors occurred!
See also "/home/smyrno/ros2_android_ws/build_isolated/fastrtps/examples/C++/OwnershipStrengthQoSExample/CMakeFiles/CMakeOutput.log".
examples/CMakeFiles/OwnershipStrengthQoSExample.dir/build.make:105: recipe for target 'examples/OwnershipStrengthQoSExample-prefix/src/OwnershipStrengthQoSExample-stamp/OwnershipStrengthQoSExample-configure' failed
make[2]: *** [examples/OwnershipStrengthQoSExample-prefix/src/OwnershipStrengthQoSExample-stamp/OwnershipStrengthQoSExample-configure] Error 1
CMakeFiles/Makefile2:1502: recipe for target 'examples/CMakeFiles/OwnershipStrengthQoSExample.dir/all' failed
make[1]: *** [examples/CMakeFiles/OwnershipStrengthQoSExample.dir/all] Error 2
[ 43%] No download step for 'SampleConfig_Multimedia'
[ 44%] No update step for 'SampleConfig_Multimedia'
[ 44%] No patch step for 'SampleConfig_Multimedia'
[ 45%] Performing configure step for 'SampleConfig_Multimedia'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:19 (find_package):
  Could not find a configuration file for package "fastcdr" that is
  compatible with requested version "".
 

 The following configuration files were considered but not accepted:

    /home/smyrno/ros2_android_ws/build_isolated/fastrtps/external/install/lib/fastcdr/cmake/fastcdrConfig.cmake, version: 1.0.7 (32bit)

-- Configuring incomplete, errors occurred!
See also "/home/smyrno/ros2_android_ws/build_isolated/fastrtps/examples/C++/StaticHelloWorldExample/CMakeFiles/CMakeOutput.log".
examples/CMakeFiles/StaticHelloWorldExample.dir/build.make:105: recipe for target 'examples/StaticHelloWorldExample-prefix/src/StaticHelloWorldExample-stamp/StaticHelloWorldExample-configure' failed
make[2]: *** [examples/StaticHelloWorldExample-prefix/src/StaticHelloWorldExample-stamp/StaticHelloWorldExample-configure] Error 1
-- Detecting CXX compile features - done
CMakeFiles/Makefile2:1465: recipe for target 'examples/CMakeFiles/StaticHelloWorldExample.dir/all' failed
make[1]: *** [examples/CMakeFiles/StaticHelloWorldExample.dir/all] Error 2
1 warning generated.
CMake Error at CMakeLists.txt:19 (find_package):
  Could not find a configuration file for package "fastcdr" that is
  compatible with requested version "".

  The following configuration files were considered but not accepted:

    /home/smyrno/ros2_android_ws/build_isolated/fastrtps/external/install/lib/fastcdr/cmake/fastcdrConfig.cmake, version: 1.0.7 (32bit)

-- Configuring incomplete, errors occurred!
See also "/home/smyrno/ros2_android_ws/build_isolated/fastrtps/examples/C++/RTPSTest_registered/CMakeFiles/CMakeOutput.log".
examples/CMakeFiles/RTPSTest_registered.dir/build.make:105: recipe for target 'examples/RTPSTest_registered-prefix/src/RTPSTest_registered-stamp/RTPSTest_registered-configure' failed
make[2]: *** [examples/RTPSTest_registered-prefix/src/RTPSTest_registered-stamp/RTPSTest_registered-configure] Error 1
CMakeFiles/Makefile2:1576: recipe for target 'examples/CMakeFiles/RTPSTest_registered.dir/all' failed
make[1]: *** [examples/CMakeFiles/RTPSTest_registered.dir/all] Error 2
[ 53%] Linking CXX executable SimpleCommunicationPublisher
[ 53%] Linking CXX executable SimpleCommunicationSubscriber
[ 53%] Built target SimpleCommunicationPublisher
[ 53%] Built target SimpleCommunicationSubscriber
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/smyrno/ros2_android_ws/build_isolated/fastrtps/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/smyrno/ros2_android_ws/build_isolated/fastrtps' with exit code '2'
<== Command '. /home/smyrno/ros2_android_ws/build_isolated/fastrtps/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/smyrno/ros2_android_ws/build_isolated/fastrtps' with exit code '2'
```

Thanks in advance.

esteve, 2017-11-03:
@eborghi10 sorry for the slow response and thank for trying this out :-) Yeah, it's a bug in the README.md I forgot to update it with the latest version of the NDK and the CMake flags I use for the [CI builds](https://travis-ci.org/esteve/ros2_android_travis). Could you try building the Android bindings again with the following instead:

```
ament build --parallel \
  --symlink-install \
  --isolated \
  --cmake-args \
  -DPYTHON_EXECUTABLE=/usr/bin/python3 \
  -DCMAKE_TOOLCHAIN_FILE=$HOME/android_ndk/android-ndk-r15c/build/cmake/android.toolchain.cmake \
  -DANDROID_FUNCTION_LEVEL_LINKING=OFF \
  -DANDROID_NATIVE_API_LEVEL=android-21 \
  -DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang \
  -DANDROID_STL=gnustl_shared \
  -DANDROID_ABI=armeabi-v7a \
  -DANDROID_NDK=$HOME/android_ndk/android-ndk-r15c \
  -DTHIRDPARTY=ON \
  -DCOMPILE_EXAMPLES=OFF \
  -DCMAKE_FIND_ROOT_PATH="$HOME/ament_ws/install_isolated;$HOME/ros2_android_ws/install_isolated" \
  -- \
  --ament-gradle-args \
  -Pament.android_stl=gnustl_shared -Pament.android_abi=armeabi-v7a -Pament.android_ndk=$HOME/android_ndk/android-ndk-r15c --
```

Alternatively, you can also give the [Docker images](https://hub.docker.com/r/esteve/ros2-ubuntu-xenial-travisci/) I use for [Travis](https://travis-ci.org/esteve/ros2_android_travis) a try. Just fetch them like this:

```sh
$ docker run -it --rm esteve/ros2-ubuntu-xenial-travisci:android
$ sh /build_android.sh
```

Let me know if that helped!

esteve, 2017-11-20:
I'm closing this since you got it working with `-DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang`. Thanks for the report!

closed 2017-11-20 by esteve