ros2-java/ros2_java #33: unrecognized command line options '-fno-integrated-as' and '-fno-limit-debug-info'

eborghi10, 2017-11-01:
Hi, I'm having the following issue when I install ROS2. I have installed **CMake 3.10** and **Android NDK r14b**.

```bash
# Topological order
 - ament_cmake_export_jars
 - ament_cmake_ros
 - fastcdr
 - fastrtps
 - fastrtps_cmake_module
 - poco_vendor
 - rcljava_common
 - rcutils
 - rmw
 - rmw_implementation_cmake
 - rosidl_default_generators
 - rosidl_default_runtime
 - rosidl_generator_dds_idl
 - rosidl_parser
 - rosidl_cmake
 - rosidl_typesupport_interface
 - rosidl_generator_c
 - rosidl_generator_cpp
 - rosidl_typesupport_c
 - rosidl_typesupport_cpp
 - rosidl_typesupport_introspection_c
 - rosidl_typesupport_introspection_cpp
 - rmw_fastrtps_cpp
 - rmw_implementation
 - rosidl_generator_java
 - builtin_interfaces
 - example_interfaces
 - lifecycle_msgs
 - rcl_interfaces
 - std_msgs
 - actionlib_msgs
 - geometry_msgs
 - diagnostic_msgs
 - nav_msgs
 - rcl
 - rcl_lifecycle
 - rcljava
 - rclandroid
 - ros2_listener_android
 - ros2_talker_android
 - sensor_msgs
 - shape_msgs
 - std_srvs
 - stereo_msgs
 - test_msgs
 - trajectory_msgs
 - visualization_msgs
 - common_interfaces

Process package 'ament_cmake_export_jars' with context:
--------------------------------------------------------------------------------
 source_space => /home/emiliano/ros2_android_ws/src/ros2_java/ros2_java/ament_cmake_export_jars
  build_space => /home/emiliano/ros2_android_ws/build_isolated/ament_cmake_export_jars
install_space => /home/emiliano/ros2_android_ws/install_isolated/ament_cmake_export_jars
   make_flags => -j4, -l4
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'ament_cmake_export_jars'
Running cmake because arguments have changed.
==> '. /home/emiliano/ros2_android_ws/build_isolated/ament_cmake_export_jars/cmake__build.sh && /usr/bin/cmake /home/emiliano/ros2_android_ws/src/ros2_java/ros2_java/ament_cmake_export_jars -DBUILD_TESTING=0 -DPYTHON_EXECUTABLE=/usr/bin/python3 -DEPROSIMA_BUILD=ON -DCMAKE_FIND_ROOT_PATH=/home/emiliano/ament_ws/install_isolated;/home/emiliano/ros2_android_ws/install_isolated -DANDROID_FUNCTION_LEVEL_LINKING=OFF -DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang -DANDROID_STL=gnustl_shared -DANDROID_ABI=armeabi-v7a -DANDROID_NDK=/home/emiliano/android_ndk/android-ndk-r14b -DANDROID_NATIVE_API_LEVEL=android-23 -DCMAKE_TOOLCHAIN_FILE=/home/emiliano/android_ndk/android-ndk-r14b/build/cmake/android.toolchain.cmake -DCMAKE_INSTALL_PREFIX=/home/emiliano/ros2_android_ws/install_isolated/ament_cmake_export_jars' in '/home/emiliano/ros2_android_ws/build_isolated/ament_cmake_export_jars'
-- Found ament_cmake_core: 0.0.3 (/home/emiliano/ament_ws/install_isolated/ament_cmake_core/share/ament_cmake_core/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Configuring done
-- Generating done
-- Build files have been written to: /home/emiliano/ros2_android_ws/build_isolated/ament_cmake_export_jars
==> '. /home/emiliano/ros2_android_ws/build_isolated/ament_cmake_export_jars/cmake__build.sh && /usr/bin/make -j4 -l4' in '/home/emiliano/ros2_android_ws/build_isolated/ament_cmake_export_jars'
[... 95 lines cut ...]
    make[1]: Entering directory '/home/emiliano/ros2_android_ws/build_isolated/fastcdr/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_7b27b.dir/testCCompiler.c.o
    /home/emiliano/android_ndk/android-ndk-r14b/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/bin/arm-linux-androideabi-gcc --target=armv7-none-linux-androideabi --sysroot=/home/emiliano/android_ndk/android-ndk-r14b/platforms/android-23/arch-arm  -isystem /home/emiliano/android_ndk/android-ndk-r14b/platforms/android-23/arch-arm/usr/include -isystem /home/emiliano/android_ndk/android-ndk-r14b/platforms/android-23/arch-arm/usr/include/arm-linux-androideabi  -g -DANDROID -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -march=armv7-a -mfloat-abi=softfp -mfpu=vfpv3-d16 -fno-integrated-as -mthumb -Wa,--noexecstack -Wformat -Werror=format-security   -O0 -fno-limit-debug-info  -fPIE   -o CMakeFiles/cmTC_7b27b.dir/testCCompiler.c.o   -c /home/emiliano/ros2_android_ws/build_isolated/fastcdr/CMakeFiles/CMakeTmp/testCCompiler.c
    arm-linux-androideabi-gcc: error: unrecognized command line option '-fno-integrated-as'
    arm-linux-androideabi-gcc: error: unrecognized command line option '-fno-limit-debug-info'
    CMakeFiles/cmTC_7b27b.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_7b27b.dir/testCCompiler.c.o' failed
    make[1]: *** [CMakeFiles/cmTC_7b27b.dir/testCCompiler.c.o] Error 1
    make[1]: Leaving directory '/home/emiliano/ros2_android_ws/build_isolated/fastcdr/CMakeFiles/CMakeTmp'
    Makefile:126: recipe for target 'cmTC_7b27b/fast' failed
    make: *** [cmTC_7b27b/fast] Error 2
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:31 (project)

-- Configuring incomplete, errors occurred!
See also "/home/emiliano/ros2_android_ws/build_isolated/fastcdr/CMakeFiles/CMakeOutput.log".
See also "/home/emiliano/ros2_android_ws/build_isolated/fastcdr/CMakeFiles/CMakeError.log".

<== Command '. /home/emiliano/ros2_android_ws/build_isolated/fastcdr/cmake__build.sh && /usr/bin/cmake /home/emiliano/ros2_android_ws/src/eProsima/Fast-CDR -DPYTHON_EXECUTABLE=/usr/bin/python3 -DEPROSIMA_BUILD=ON '-DCMAKE_FIND_ROOT_PATH=/home/emiliano/ament_ws/install_isolated;/home/emiliano/ros2_android_ws/install_isolated' -DANDROID_FUNCTION_LEVEL_LINKING=OFF -DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang -DANDROID_STL=gnustl_shared -DANDROID_ABI=armeabi-v7a -DANDROID_NDK=/home/emiliano/android_ndk/android-ndk-r14b -DANDROID_NATIVE_API_LEVEL=android-23 -DCMAKE_TOOLCHAIN_FILE=/home/emiliano/android_ndk/android-ndk-r14b/build/cmake/android.toolchain.cmake -DCMAKE_INSTALL_PREFIX=/home/emiliano/ros2_android_ws/install_isolated/fastcdr' failed in '/home/emiliano/ros2_android_ws/build_isolated/fastcdr' with exit code '1'
<== Command '. /home/emiliano/ros2_android_ws/build_isolated/fastcdr/cmake__build.sh && /usr/bin/cmake /home/emiliano/ros2_android_ws/src/eProsima/Fast-CDR -DPYTHON_EXECUTABLE=/usr/bin/python3 -DEPROSIMA_BUILD=ON '-DCMAKE_FIND_ROOT_PATH=/home/emiliano/ament_ws/install_isolated;/home/emiliano/ros2_android_ws/install_isolated' -DANDROID_FUNCTION_LEVEL_LINKING=OFF -DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang -DANDROID_STL=gnustl_shared -DANDROID_ABI=armeabi-v7a -DANDROID_NDK=/home/emiliano/android_ndk/android-ndk-r14b -DANDROID_NATIVE_API_LEVEL=android-23 -DCMAKE_TOOLCHAIN_FILE=/home/emiliano/android_ndk/android-ndk-r14b/build/cmake/android.toolchain.cmake -DCMAKE_INSTALL_PREFIX=/home/emiliano/ros2_android_ws/install_isolated/fastcdr' failed in '/home/emiliano/ros2_android_ws/build_isolated/fastcdr' with exit code  #'1'
```

esteve, 2017-11-03:
@eborghi10 this seems to be caused by using GCC instead of Clang as the compiler, could you pass `-DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang` as an argument to `--cmake-args` and try building again? Thanks!

esteve, 2017-11-20:
Closing this one as well since you got it all working now :-)

closed 2017-11-20 by esteve