ros2-java/ros2_java #34: Desktop build failing

jubeira, 2017-11-11:
Hi @esteve!
First of all, thanks for publishing your work!

Today I tried the build instructions without Docker in my computer, and it fails when it tries to build `rcljava_examples`.
When I run `ament build --symlink-install --isolated` (under the header "Desktop" in the readme), I get:
```
* What went wrong:
A problem occurred evaluating root project 'rcljava_examples'.
> Failed to apply plugin [id 'org.ros2.tools.gradle']
   > Could not create an instance of type org.ros2.tools.gradle.JavaAmentPluginExtension_Decorated.
```
I got Ubuntu 16.04 with CMake 3.6, Gradle 2.10 and Java 8 installed. Any ideas?

esteve, 2017-11-11:
Thanks for the feedback!

Could you paste a longer log? Gradle should output a bit more info about
what went wrong. Also, could you try upgrading Gradle? The Docker images
use Gradle from the following PPA:

https://launchpad.net/~cwchien/+archive/ubuntu/gradle?field.series_filter=xenial

IIRC the Gradle plug-in requires at least version 3.2

Thanks!

jubeira, 2017-11-11:
Thanks for the quick reply!
Upgrading Gradle did the trick. Just for the record, I followed [this guide](https://www.vultr.com/docs/how-to-install-gradle-on-ubuntu-16-10) to upgrade my Gradle version. I'm now running the examples in my desktop :)

Perhaps the minimum Gradle version can be specified as a pre-requisite?

vaasudEva, 2017-11-20:
Hi, I am still get build error while building Desktop Java
```
$ gradle -v
------------------------------------------------------------
Gradle 4.3.1
------------------------------------------------------------

Build time:   2017-11-08 08:59:45 UTC
Revision:     e4f4804807ef7c2829da51877861ff06e07e006d

Groovy:       2.4.12
Ant:          Apache Ant(TM) version 1.9.6 compiled on June 29 2015
JVM:          1.8.0_131 (Oracle Corporation 25.131-b11)
OS:           Linux 4.4.0-96-generic amd64
```
```
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] 
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] FAILURE: Build failed with an exception.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] 
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] * What went wrong:
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] A problem occurred configuring root project 'rcljava_examples'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] > Could not resolve all files for configuration ':classpath'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]    > Could not resolve gradle.plugin.org.ros2.tools.gradle:ament:0.6.1.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]      Required by:
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]          project :
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]       > Could not resolve gradle.plugin.org.ros2.tools.gradle:ament:0.6.1.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]          > Could not get resource 'https://jcenter.bintray.com/gradle/plugin/org/ros2/tools/gradle/ament/0.6.1/ament-0.6.1.pom'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]             > Could not HEAD 'https://jcenter.bintray.com/gradle/plugin/org/ros2/tools/gradle/ament/0.6.1/ament-0.6.1.pom'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]                > Connect to proxy01.iind.tel.com:443 [proxy01.iind.tel.com/10.1.192.48] failed: Connection refused (Connection refused)
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]       > Could not resolve gradle.plugin.org.ros2.tools.gradle:ament:0.6.1.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]          > Could not get resource 'https://repo1.maven.org/maven2/gradle/plugin/org/ros2/tools/gradle/ament/0.6.1/ament-0.6.1.pom'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]             > Could not HEAD 'https://repo1.maven.org/maven2/gradle/plugin/org/ros2/tools/gradle/ament/0.6.1/ament-0.6.1.pom'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]                > Connect to proxy01.iind.tel.com:443 [proxy01.iind.tel.com/10.1.192.48] failed: Connection refused (Connection refused)
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]       > Could not resolve gradle.plugin.org.ros2.tools.gradle:ament:0.6.1.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]          > Could not get resource 'https://plugins.gradle.org/m2/gradle/plugin/org/ros2/tools/gradle/ament/0.6.1/ament-0.6.1.pom'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]             > Could not HEAD 'https://plugins.gradle.org/m2/gradle/plugin/org/ros2/tools/gradle/ament/0.6.1/ament-0.6.1.pom'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]                > Connect to proxy01.iind.tel.com:443 [proxy01.iind.tel.com/10.1.192.48] failed: Connection refused (Connection refused)
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]    > Could not resolve me.tatarka:gradle-retrolambda:3.6.1.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]      Required by:
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]          project :
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]       > Could not resolve me.tatarka:gradle-retrolambda:3.6.1.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]          > Could not get resource 'https://jcenter.bintray.com/me/tatarka/gradle-retrolambda/3.6.1/gradle-retrolambda-3.6.1.pom'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]             > Could not HEAD 'https://jcenter.bintray.com/me/tatarka/gradle-retrolambda/3.6.1/gradle-retrolambda-3.6.1.pom'.
03:49:15.103 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]                > Connect to proxy01.iind.tel.com:443 [proxy01.iind.tel.com/10.1.192.48] failed: Connection refused (Connection refused)

```
If its, proxy issue too, I had set proxy info like this (see below). But its not working
```
-Dhttps.proxyHost=proxy01.iind.tel.com -Dhttps.ProxyPort=800 -Dhttp.proxyHost=proxy01.iind.tel.com -Dhttp.ProxyPort=800
``

esteve, 2017-11-20:
@jubeira noted, closing this and I'll update the README.md to mention that Gradle >= 3.2 needs to be installed. Thanks for digging into it.

closed 2017-11-20 by esteve

esteve, 2017-11-20:
@vaasudEva the problem you're having appears to be unrelated to the report that @jubeira filed and to ros2_java (see how it's failing to fetch the retrolambda plugin as well), it seems there's an issue with your network. In any case, can you try creating a file named test.gradle with these contents:

```buildscript {
  repositories {
    jcenter()
    mavenCentral()
    mavenLocal()
    maven {
      url "https://plugins.gradle.org/m2/"
    }
  }

  dependencies {
    classpath 'gradle.plugin.org.ros2.tools.gradle:ament:0.7.0'
    classpath 'me.tatarka:gradle-retrolambda:3.6.1'
  }
}

repositories {
   mavenCentral()
}
```
and type `gradle -b test.gradle` so that Gradle will attempt downloading the dependencies. Also, I think you have an older version of the examples (the latest revision uses the 0.7.0 version of the Ament plugin), I highly recoomend you to update your local copy of the ros2_java_examples repo.

vaasudEva, 2017-11-21:
Thank you so much @esteve. Actually I had to set proxy in `~/gradle.properties`.

esteve, 2017-11-21:
@vaasudEva no problem :-) Let me know if you have any other question about ros2_java