ros2-java/ros2_java #35: Android build details and documentation

jubeira, 2017-11-11:
First of all, I can confirm that the installation instructions for Android examples worked on my desktop.
Here's some details I found during the installation:
- Pre-requirements: Python 3, `python-vcstool`, Gradle > 3.2 (e.g. [here](https://www.vultr.com/docs/how-to-install-gradle-on-ubuntu-16-10)) and Android NDK are needed (r15 as stated in the instructions is OK). `gradle -v` should indicate a version higher than 3.2. Maybe it would be nice to have this documented in the readme too.
- Details: at some point of the installation when doing `ament build ....`, CMake tries to create a directory in the root folder `/test_msgs`, and it doesn't have access to it. The directory can be created manually with standard user access beforehand, and then the build works. Perhaps it would be nice to find that part of the buildscript and create the directory inside the workspace.

Other than that, this looks good! Thanks again for the good work!

Pedestrian1671022, 2018-04-13:
when i observe the following instructions
mkdir -p ~/ros2_java_ws/src
cd ~/ros2_java_ws
wget https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop.repos
vcs import ~/ros2_java_ws/src < ros2_java_desktop.repos
cd ~/ros2_java_ws/src/ros2/rosidl_typesupport
patch -p1 < ../../ros2_java/ros2_java/rosidl_typesupport_ros2_java.patch
cd ~/ros2_java_ws
. ~/ament_ws/install_isolated/local_setup.sh
ament build --symlink-install --isolated
error occurred
/home/pedestrian/ros2_java_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:367:16: error: ‘rmw_log_severity_t’ was not declared in this scope
   1, ARG_TYPES(rmw_log_severity_t))
                ^
/home/pedestrian/ros2_java_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:176:19: note: in definition of macro ‘EXPAND’
 #define EXPAND(x) x
                   ^
/home/pedestrian/ros2_java_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:212:26: note: in expansion of macro ‘ARGS_1’
   ReturnType name(EXPAND(ARGS_ ## _NR(__VA_ARGS__))) \
                          ^
~/ros2_java_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:365:1: note: in expansion of macro ‘RMW_INTERFACE_FN’
 RMW_INTERFACE_FN(rmw_set_log_severity,
 ^
~/ros2_java_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:367:6: note: in expansion of macro ‘ARG_TYPES’
   1, ARG_TYPES(rmw_log_severity_t))
      ^
~/ros2_java_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:213:3: error: expected ‘,’ or ‘;’ before ‘{’ token
   { \
   ^
~/ros2_java_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:365:1: note: in expansion of macro ‘RMW_INTERFACE_FN’
 RMW_INTERFACE_FN(rmw_set_log_severity,
 ^
~/ros2_java_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:421:1: error: expected ‘}’ at end of input
 }
 ^
CMakeFiles/rmw_implementation.dir/build.make:62: recipe for target 'CMakeFiles/rmw_implementation.dir/src/functions.cpp.o' failed
make[2]: *** [CMakeFiles/rmw_implementation.dir/src/functions.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/rmw_implementation.dir/all' failed
make[1]: *** [CMakeFiles/rmw_implementation.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. ~/ros2_java_ws/build_isolated/rmw_implementation/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '~/ros2_java_ws/build_isolated/rmw_implementation' with exit code '2'
<== Command '. ~/ros2_java_ws/build_isolated/rmw_implementation/cmake__build.sh && /usr/bin/make -j8 -l8' failed in ~/ros2_java_ws/build_isolated/rmw_implementation' with exit code '2'

esteve, 2018-04-14:
@Pedestrian1671022 please don't cross-post, you already submitted https://github.com/esteve/ros2_android_examples/issues/2 and replied there. I left this ticket open as a reminder for fleshing out the Android instructions.

jacobperron, 2021-11-13:
Tracking here now: https://github.com/ros2-java/ros2_java/issues/81
Closing as duplicate.

closed 2021-11-13 by jacobperron