ros2-java/ros2_java #36: rcljava_examples build failure

mallanmba, 2017-12-22:
When attempting to follow the build instructions, ament bails with an error when building rcljava_examples:
```
* Where:
Build file '/home/mallan/ros/ros2_java_ws/build_isolated/rcljava_examples/build.gradle' line: 2

* What went wrong:
A problem occurred evaluating root project 'rcljava_examples'.
> Failed to apply plugin [id 'org.ros2.tools.gradle']
   > Could not create an instance of type org.ros2.tools.gradle.JavaAmentPluginExtension_Decorated.
```
the root exception is 
```
Caused by: org.gradle.api.internal.MissingMethodException: Could not find method findProperty() for arguments [ament.source_space] on root project 'rcljava_examples'.
```
The full error is attached below - this is on Ubuntu 16.04. Let me know if you need more info or if I can help.

[examples_build_failure.txt](https://github.com/esteve/ros2_java/files/1583488/examples_build_failure.txt)

esteve, 2017-12-22:
Thanks for filing this issue. What version of Gradle do you have? The ros2_java Gradle plugin needs 3.2 or higher. If you're using the package that comes with 16.04, you can install a more recent version of Gradle via the following PPA:

```
sudo add-apt-repository ppa:cwchien/gradle
sudo apt install -y gradle
```

let me know if that helps.

mallanmba, 2017-12-22:
Ah, thanks for the quick reply - yes, Gradle version was the issue. I had the 16.04 default (2.1), after updating the build succeeded.  It'd be helpful to have that in the install instructions (or is it there and I missed it?)  Cheers

closed 2017-12-22 by mallanmba

esteve, 2017-12-22:
Awesome! I kept meaning to update the instructions (I had left #35 open to remind myself). I've finally made a quick edit with the instructions for installing Gradle on Ubuntu https://github.com/esteve/ros2_java/commit/053c8b4329364dbdf5331d570cbb9cdd86f009c0