ros2-java/ros2_java #38: build on OSX failed

qiao-tw, 2018-04-21:
Hi @esteve,

I tried to build this repo for android on my machine but failed on package "ament_cmake_export_jars".
My environment is macOS High Sierra 10.13.4. 
The steps I did are as follows:
```
git clone git@github.com:esteve/ros2_java.git
cd ros2_java
chmod 755 ci-scripts/build-android.sh
./ci-scripts/build-android.sh
```
The Fast-RTPS seems built successfully. However, the final message is
`Failed packages: ament_cmake_export_jars`

Please give some pointers, thanks.

esteve, 2018-04-21:
Hi @qiao-tw, thanks for trying out `ros2-java`!

I use those scripts for the CI builds (https://travis-ci.org/esteve/ros2_java), but they are not meant to be used by users directly. Could you try the steps in https://github.com/esteve/ros2_java#sounds-great-how-can-i-try-this-out and https://github.com/esteve/ros2_java#android and tell me if they work on macOS? I recall trying them on macOS a while ago, but I don't know if they'll still work. Thanks!

qiao-tw, 2018-04-21:
Thank you @esteve for the quick reply!  However, I just tried but it makes no difference.
The only change I made is a newer NDK and put it at a different place (`/opt/android/android_ndk_r16b`)
I put the build log [here](https://www.dropbox.com/s/4l05maqgd17a33e/ros2_java.build-android.log?dl=0) for your reference.
Thanks for the help.

esteve, 2018-04-21:
It seems to be failing because CMake can't properly find the Python interpreter, I think it's because `brew` installs Python in `/usr/local`. Could you pass `-DPYTHON_EXECUTABLE=/usr/local/bin/python3` to `ament build ...` and try again? I hope that should solve it. Thanks!

esteve, 2018-04-23:
Great! Happy to know it works for you :-) The `test_msgs` package is a bit different than the other message packages because it installs the generated Python messages directly, so there's not much it can be done besides skipping it. You can either add an `AMENT_IGNORE` file in its directory or pass `--skip-packages test_msgs` to `ament`

qiao-tw, 2018-05-03:
Thanks for the help again! I believe I'm very close to the end point of this building process.
The error I just met came from gradle, I did some research and suspected that it relates to java version, but sadly I don't know how to fix it. The error message as following:

```
Process package 'rclandroid' with context:
--------------------------------------------------------------------------------
 source_space => /Users/qiao/ros2_android_ws/src/ros2_java/ros2_android/rclandroid
  build_space => /Users/qiao/ros2_android_ws/build_isolated/rclandroid
install_space => /Users/qiao/ros2_android_ws/install_isolated/rclandroid
   make_flags => -j8, -l8
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'rclandroid'
==> '/usr/local/bin/gradle -Pament.source_space=/Users/qiao/ros2_android_ws/src/ros2_java/ros2_android/rclandroid -Pament.build_space=/Users/qiao/ros2_android_ws/build_isolated/rclandroid -Pament.install_space=/Users/qiao/ros2_android_ws/install_isolated/rclandroid -Pament.dependencies=/Users/qiao/ros2_android_ws/install_isolated/ament_cmake_export_jars/share/ament_cmake_export_jars:/Users/qiao/ros2_android_ws/install_isolated/ament_cmake_export_jni_libraries/share/ament_cmake_export_jni_libraries:/Users/qiao/ros2_android_ws/install_isolated/fastcdr/share/fastcdr:/Users/qiao/ros2_android_ws/install_isolated/fastrtps/share/fastrtps:/Users/qiao/ros2_android_ws/install_isolated/fastrtps_cmake_module/share/fastrtps_cmake_module:/Users/qiao/ros2_android_ws/install_isolated/poco_vendor/share/poco_vendor:/Users/qiao/ros2_android_ws/install_isolated/rcljava_common/share/rcljava_common:/Users/qiao/ros2_android_ws/install_isolated/rcutils/share/rcutils:/Users/qiao/ros2_android_ws/install_isolated/rmw/share/rmw:/Users/qiao/ros2_android_ws/install_isolated/rmw_implementation_cmake/share/rmw_implementation_cmake:/Users/qiao/ros2_android_ws/install_isolated/rosidl_parser/share/rosidl_parser:/Users/qiao/ros2_android_ws/install_isolated/rosidl_cmake/share/rosidl_cmake:/Users/qiao/ros2_android_ws/install_isolated/rosidl_typesupport_interface/share/rosidl_typesupport_interface:/Users/qiao/ros2_android_ws/install_isolated/rosidl_generator_c/share/rosidl_generator_c:/Users/qiao/ros2_android_ws/install_isolated/rosidl_generator_cpp/share/rosidl_generator_cpp:/Users/qiao/ros2_android_ws/install_isolated/rosidl_typesupport_introspection_c/share/rosidl_typesupport_introspection_c:/Users/qiao/ros2_android_ws/install_isolated/rosidl_typesupport_c/share/rosidl_typesupport_c:/Users/qiao/ros2_android_ws/install_isolated/rosidl_typesupport_introspection_cpp/share/rosidl_typesupport_introspection_cpp:/Users/qiao/ros2_android_ws/install_isolated/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp:/Users/qiao/ros2_android_ws/install_isolated/rmw_implementation/share/rmw_implementation:/Users/qiao/ros2_android_ws/install_isolated/rosidl_default_generators/share/rosidl_default_generators:/Users/qiao/ros2_android_ws/install_isolated/rosidl_default_runtime/share/rosidl_default_runtime:/Users/qiao/ros2_android_ws/install_isolated/rosidl_generator_java/share/rosidl_generator_java:/Users/qiao/ros2_android_ws/install_isolated/rosidl_typesupport_cpp/share/rosidl_typesupport_cpp:/Users/qiao/ros2_android_ws/install_isolated/builtin_interfaces/share/builtin_interfaces:/Users/qiao/ros2_android_ws/install_isolated/rcl_interfaces/share/rcl_interfaces:/Users/qiao/ros2_android_ws/install_isolated/rcl/share/rcl:/Users/qiao/ros2_android_ws/install_isolated/rcljava/share/rcljava -Pament.build_tests=False -Pament.package_manifest.name=rclandroid -Pament.exec_dependency_paths_in_workspace=/Users/qiao/ros2_android_ws/install_isolated/rcljava/share/rcljava -Pament.gradle_recursive_dependencies=True -Pament.gradle_isolated=True -Pament.android_stl=gnustl_shared -Pament.android_abi=armeabi-v7a -Pament.android_ndk=/Users/qiao/Tools/android-ndk-r16b -Pament.android_variant=release assemble --stacktrace' in '/Users/qiao/ros2_android_ws/build_isolated/rclandroid'

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring root project 'rclandroid'.
> Failed to notify project evaluation listener.
   > Could not initialize class com.android.sdklib.repository.AndroidSdkHandler

* Try:
Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Exception is:
org.gradle.api.ProjectConfigurationException: A problem occurred configuring root project 'rclandroid'.
        at org.gradle.configuration.project.LifecycleProjectEvaluator.addConfigurationFailure(LifecycleProjectEvaluator.java:94)
        at org.gradle.configuration.project.LifecycleProjectEvaluator.notifyAfterEvaluate(LifecycleProjectEvaluator.java:89)
        at org.gradle.configuration.project.LifecycleProjectEvaluator.doConfigure(LifecycleProjectEvaluator.java:70)
        at org.gradle.configuration.project.LifecycleProjectEvaluator.access$100(LifecycleProjectEvaluator.java:34)
        at org.gradle.configuration.project.LifecycleProjectEvaluator$ConfigureProject.run(LifecycleProjectEvaluator.java:110)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:317)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:309)
[... cut ...]
        at org.gradle.internal.event.ListenerBroadcast.dispatch(ListenerBroadcast.java:140)
        at org.gradle.internal.event.ListenerBroadcast.dispatch(ListenerBroadcast.java:37)
        at org.gradle.internal.dispatch.ProxyDispatchAdapter$DispatchingInvocationHandler.invoke(ProxyDispatchAdapter.java:93)
        at com.sun.proxy.$Proxy28.afterEvaluate(Unknown Source)
        at org.gradle.configuration.project.LifecycleProjectEvaluator.notifyAfterEvaluate(LifecycleProjectEvaluator.java:76)
        ... 85 more
```

Any idea? Thanks!

esteve, 2018-05-03:
@qiao-tw Almost there! :-) I haven't tested `ros2-java` on OSX myself, but I found this on StackOverlfow which raises the same exception:

https://stackoverflow.com/a/48341857

What version of the JDK do you have installed? 8 or 9? The answer on StackOverflow recommends installing 8, as it seems that 9 is not supported by the Android SDK yet, but I don't know, I've only tested `ros2-java` with JDK 8.

qiao-tw, 2018-05-04:
@esteve Many thanks for your big help! I finally got there!

closed 2018-05-04 by qiao-tw

esteve, 2018-05-06:
@qiao-tw that's great! Looking forward to see what you build with the Android bindings :-)

So, let me summarize what you did so I can update the documentation for OSX:

- Set `-DPYTHON_EXECUTABLE` to `/usr/local/bin/python3` when invoking `ament` to build the workspace
- Ignore the `test_msgs` package
- Downgrade JDK to 8
is that all or is there anything else?

qiao-tw, 2018-05-06:
@esteve how about this - let me revise the steps and send you a pull request?  :-)

esteve, 2018-05-06:
@qiao-tw that'd be fantastic :-) Thank you so much!