ros2-java/ros2_java #43: Android build can't build examples (libfastcdr not found)

johnmarkwayve, 2018-06-11:
I can't build for android following the README instructions.

The build fails trying to link fastrtps with the following error:

```
[ 38%] Building C object src/cpp/CMakeFiles/fastrtps.dir/__/__/thirdparty/android-ifaddrs/ifaddrs.c.o
72 warnings generated.
73 warnings generated.
1 warning generated.
1 warning generated.
1 warning generated.
1 warning generated.
[ 39%] Linking CXX shared library libfastrtps.so
/home/johnmark/Android/Ndk/android-ndk-r17/toolchains/arm-linux-androideabi-4.9/prebuilt/linux-x86_64/lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: cannot find -lfastcdr
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
src/cpp/CMakeFiles/fastrtps.dir/build.make:1403: recipe for target 'src/cpp/libfastrtps.so' failed
make[2]: *** [src/cpp/libfastrtps.so] Error 1
CMakeFiles/Makefile2:85: recipe for target 'src/cpp/CMakeFiles/fastrtps.dir/all' failed
make[1]: *** [src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/johnmark/code/ros2_android/build_isolated/fastrtps/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/johnmark/code/ros2_android/build_isolated/fastrtps' with exit code '2'
Failed packages: rcutils
```

The script that produces this error is:

```
#!/bin/bash

set -e

AMENT_WS_INSTALL=$HOME/code/ament_java/install_isolated
ANDROID_NDK_ROOT=$HOME/Android/Ndk/android-ndk-r17
ROS2_INSTALL=$HOME/ros2_android/install_isolated

source $AMENT_WS_INSTALL/local_setup.bash

ament build --isolated --cmake-args \
  -DPYTHON_EXECUTABLE=/usr/bin/python3 \
  -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK_ROOT/build/cmake/android.toolchain.cmake \
  -DANDROID_FUNCTION_LEVEL_LINKING=OFF \
  -DANDROID_NATIVE_API_LEVEL=android-21 \
  -DANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang \
  -DANDROID_STL=gnustl_shared \
  -DANDROID_ABI=armeabi-v7a \
  -DANDROID_NDK=$ANDROID_NDK_ROOT \
  -DTHIRDPARTY=ON \
  -DCOMPILE_EXAMPLES=ON \
  -DCMAKE_FIND_ROOT_PATH="$AMENT_WS_INSTALL;$ROS2_INSTALL" \
  -- \
  --parallel \
  --ament-gradle-args \
  -Pament.android_stl=gnustl_shared -Pament.android_abi=armeabi-v7a -Pament.android_ndk=$ANDROID_NDK_ROOT --
```

The only change I've made from the README is `-DCOMPILE_EXAMPLES=ON` to build the example apks.

Building for desktop Java works fine.

esteve, 2018-06-11:
Thanks for trying out `ros2_java` and `ros2_android`!

The `COMPILE_EXAMPLES` flag controls whether the Fast-RTPS examples are built, which are not needed for Android. The .apks are in the `ros2_android_examples` repository, which are always built.

Can you paste a longer log with where `rcutils` failed?

johnmarkwayve, 2018-06-11:
Thanks for your prompt reply, here's the whole output of the script I posted (first run): https://pastebin.com/Kwt5hHUz

johnmarkwayve, 2018-06-11:
No problem. This time it causes the linker error (and is too long for pastebin)
https://gist.github.com/johnmarkwayve/a8b6ccf3069f076335bae882a211781e

esteve, 2018-06-11:
That's weird, because Fast-RTPS fetches a local copy of Fast-CDR and link against it, so it shouldn't have any problem finding it. Sorry for all the trouble, but could you remove both workspaces and try from scratch and without `--parallel`? I think there were some files left over from the first CMake run that are causing the link error.

johnmarkwayve, 2018-06-11:
Thanks for all of your help. While I was waiting I did a fresh build of the ament package, desktop package, and android package in a completely different tree. I'm no longer seeing these errors, which suggests that something may have been messed up with my first ament package build. Sorry for the wild goose chase.

I'm still working through some errors (including the `test_msgs` thing mentioned in #35 ), but I will open them as separate issues if they need addressing.

Thanks again for your help.

esteve, 2018-06-12:
Great to hear it worked! `test_msgs` is a bit special, so it's probably best to pass it to `--skip-packages` for now when crosscompiling. Let me know if there's anything I can help with. Shall I close this ticket or do you prefer to keep it open?

johnmarkwayve, 2018-06-12:
Feel free to close it. While I'm here, just a couple of suggestions for the README:

- Put in the `--skip-packages test_msgs` (or equivalent `AMENT_IGNORE`) to the android build instructions, with a sentence indicating why
- Note that the `ANDROID_HOME` environment variable needs to be defined and pointing at the root of the SDK distribution

esteve, 2018-06-15:
@johnmarkwayve thanks! I've updated the README (https://github.com/esteve/ros2_java/commit/d6077dde972a3e984073b5bf6953094886fa8292). Let me know if there's anything else that needs to be improved/fixed.

> The repos at https://github.com/ros2java are behind the ones at https://github.com/esteve, but the readme there still refers to the repos at /esteve, this may cause some confusion/conflicts. Why are there two versions at all?

The `ros2java` organization has a bit of an unfortunate history, a squatter registered it a while ago before me and forked my repositories there, so I had to ask GitHub to grant me access to it. I forgot about it all, but I just cleaned it and removed the repositories. Thanks for the heads up.

closed 2018-06-15 by esteve