ros2-java/ros2_java #44: Error installing on Ubuntu 16.04

fidget324, 2018-07-02:
I'm trying to build using
```
$ ament build --symlink-install --isolated
```
It builds several repos without any issues, but then fails when it gets to the `test_msgs` installation. I get the error:
```
cannot create directory '//test_msgs'
```
I dug into the CMake a little bit, and it seems that somewhere the path is not "isolated" and becomes the root of the filesystem: "/". I'm not sure if this is an issue with this repo, or one of the subrepos...

esteve, 2018-07-27:
Sorry for the really slow response. I've submitted https://github.com/ros2/rcl_interfaces/pull/37 which should fix this.

esteve, 2018-07-31:
Now that ros2/rcl_interfaces#37 has been merged, I'll close this ticket. Please reopen it if you still see this error. Thanks.

closed 2018-07-31 by esteve

fidget324, 2018-08-02:
I'm still getting the error. I also see the same issue in your latest Travis build for xenial. Please reopen.

fsal33m, 2018-08-02:
@esteve I'm seeing the same issue.

esteve, 2018-08-02:
Given the bug is in `rcl_interfaces` and not in `ros2-java`, there's not much that can be done except switching to the master branch in `rcl_interfaces` to pick up the change. I've changed the repos files to use the `master` branch of `rcl_interfaces`. Please let me know if it works for you.