ros2-java/ros2_java #45: ROS2 bouncy release

johnmarkwayve, 2018-07-06:
I've been trying to get and android build to work with the new `bouncy` release of ROS2, not yet with great success. My guess is that the switch to `colcon` as a build system is the main problem (I've been able to get an "ordinary" arm cross-compilation build working https://github.com/ros2-for-arm/ros2/issues/5)

Do you have any suggestions for how to complete the process?

esteve, 2018-07-09:
CI tracks ROS2 master (https://travis-ci.org/esteve/ros2_java/jobs/401430849 https://travis-ci.org/esteve/ros2_java/jobs/401430850), so it `ros2-android` work with Bouncy (which is just a few days older than master). Could you elaborate a bit about the issues you had building `ros2-android` for Bouncy?

Regarding `colcon`, for now `ament_tools` is still fine, but I'm migrating `ament_java` to `colcon-ros-gradle`. If you don't use the [ROS2 Gradle](https://plugins.gradle.org/plugin/org.ros2.tools.gradle) plugin, `colcon` should work (but I unfortunately haven't had time to test it) as is with `ros2-android`

johnmarkwayve, 2018-07-09:
Apologies for the confusion, it turns out I was accidentally using ardent on my (linux) publisher, so the messages weren't being picked up by the bouncy (android) subscriber. Thanks for your help.

closed 2018-07-09 by johnmarkwayve

esteve, 2018-07-09:
No worries. Thanks for the quick response!

flewber, 2018-07-23:
So just to clarify, as @johnmarkwayve alluded to, a machine running Ardent will not be able to communicate with a machine running Bouncy?

johnmarkwayve, 2018-07-24:
@flewber I've not seen anything official to say that this is by design, but in my experience over the last few weeks it appears to be true. Or, at least, a publisher using bouncy will not be received by a subscriber using ardent (I don't think I checked the inverse)

calvertdw, 2018-07-24:
It might depend on the DDS vendor you choose. I have had problems mixing Fast-RTPS with OpenSplice. See https://github.com/eProsima/Fast-RTPS/issues/142.

flewber, 2018-07-24:
I have only been using the bundled Fast-RTPS on all of my machines. When I upgraded one to Bouncy, the demo_nodes_cpp talker and listener could no longer communicate in either direction from a Bouncy machine to an Ardent machine. Upgrading the Ardent machine to Bouncy now... hopefully that will fix the issue.

johnmarkwayve, 2018-07-25:
@flewber That's similar to my experience with Linux amd64, Linux arm64, and Java on android

flewber, 2018-07-25:
Upgrading the Ardent machine to Bouncy DID fix the issue.