ros2-java/ros2_java #47: Tag for bouncy needed

calvertdw, 2018-07-20:
Let's make a tag for bouncy and get it working as an overlay on a bouncy binary release.

esteve, 2018-07-20:
This is a bit difficult, since the generators are plugged in when all the repos are in the workspace. Correct me if I'm wrong, but it seems the goal of this ticket is not so much about the tag itself, but about being able to combine `ros2-java` with binary releases. I've changed the title, but please revert it if I'm overstepping here. Thanks!

calvertdw, 2018-07-23:
Is there a reason you don't build ros2_java as an overlay ws? I tried and I get ament errors:
```
Starting >>> rosidl_generator_java
--- stderr: rosidl_generator_java
CMake Error at F:/dev/ros2java/install/share/rcljava_common/cmake/Modules/JavaExtra.cmake:128 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMakeLists.txt:94 (ament_add_junit_tests)

CMake Error at CMakeLists.txt:110 (add_dependencies):
  Cannot add target-level dependencies to non-existent target
  "rosidl_generator_java_tests_org.ros2.generator.InterfacesTest".

  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.

---
Failed   <<< rosidl_generator_java      [ Exited with code 1 ]
```

esteve, 2018-07-23:
> Is there a reason you don't build ros2_java as an overlay ws? I tried and I get errors:

It's not a deliberate decision. When I started `ros2_java` there were no binary releases of ROS2. `ros2_java` follows the same structure as the Python generator (which I initially wrote), so there are many things I just "reapplied" here.

In any case, I think it'd be really useful to be able to combine `ros2_java` with an existing binary release. There are some hurdles, though, since there needs to be a way to build the bindings for base messages for which there are already binaries in the overlay (i.e. `std_msgs`, `builtin_interfaces`, etc.). Pull requests are more than welcome :-)

calvertdw, 2018-07-25:
@esteve Can you create a branch for `bouncy` so I can PR?

esteve, 2018-07-25:
Done: https://github.com/esteve/ros2_java/tree/bouncy

calvertdw, 2018-07-31:
I noticed this is essential fixed by https://github.com/esteve/ros2_java/commit/2b256f8ef935b48b0a6c7cd255d4652edf757825. Can you reset the `bouncy` branch to that commit?