ros2-java/ros2_java #49: Errors in `console_bridge`

calvertdw, 2018-07-23:
```log
Command line warning D9002: ignoring unknown option '-fvisibility=hidden'

error C2065: '__deprecated__': undeclared identifier
error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
error C2448: '__attribute__': function-style initializer appears to be a function definition
```

[CMakeError.log](https://github.com/esteve/ros2_java/files/2220665/CMakeError.log)

esteve, 2018-07-23:
Were you able to build ROS2 without `ros2_java`? That is, did you manage to build all of ROS2 with https://github.com/ros2/ros2/wiki/Windows-Development-Setup ?

In any case, `console_bridge` is outside the scope of `ros2_java`, so any issue should be filed in its own repository https://github.com/ros/console_bridge

calvertdw, 2018-07-23:
Yes I can build ROS 2 and run everythings, including all the DDS vendors and RViz 2.

Submitted https://github.com/ros/console_bridge/issues/57

calvertdw, 2018-07-25:
Building console_bridge by itself works fine. As well as `ament build --only-packages console_bridge` in the ros2java_ws.

calvertdw, 2018-07-25:
`ament build --only-packages console_bridge class_loader` fails with this error:

```
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Findconsole_bridge_vendor.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "console_bridge_vendor", but CMake did not find one.

  Could not find a package configuration file provided by
  "console_bridge_vendor" with any of the following names:

    console_bridge_vendorConfig.cmake
    console_bridge_vendor-config.cmake

  Add the installation prefix of "console_bridge_vendor" to CMAKE_PREFIX_PATH
  or set "console_bridge_vendor_DIR" to a directory containing one of the
  above files.  If "console_bridge_vendor" provides a separate development
  package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "F:/dev/ros2javabundle/build/class_loader/CMakeFiles/CMakeOutput.log".
```

esteve, 2018-07-25:
`console_bridge_vendor` is provided by the following repo:

```
  ros2/console_bridge_vendor:
    type: git
    url: https://github.com/ros2/console_bridge_vendor.git
    version: master
```

calvertdw, 2018-07-25:
Waiting for https://github.com/esteve/ros2_java/pull/52

PR #52 (2018-07-27): Fix class_loader & console_bridge_vendor errors
Fixes #49 and #51.
closed 2018-07-27 by esteve