ros2-java/ros2_java #50: Windows CI build

calvertdw, 2018-07-24:
@esteve If I remember correctly you said you had this project building with VSTS. I'm having errors building on my machine and want to compare with the CI, but I can't find a link to it anywhere.

If not, can we get an AppVeyor build going?

esteve, 2018-07-25:
I've set Appveyor for Linux, but I'm happy to add a Windows job. Could you submit a PR for that? Thanks.

calvertdw, 2018-07-25:
Will do! Thanks

calvertdw, 2018-07-27:
I haven't forgotten about this. I've been busy with other stuff and I've never worked with AppVeyor before so I have to make time to get up to speed.

esteve, 2018-07-27:
No worries! Thanks a lot for your contributions :-)

jacobperron, 2021-11-13:
Duplicate of https://github.com/ros2-java/ros2_java/issues/102

closed 2021-11-13 by jacobperron