ros2-java/ros2_java #54: Building 'rcljava' fails on Ubuntu 16.04

FatimaSaleem, 2018-07-28:
I get the following errors while building ros2_java when it gets to rcljava package
```
[ 19%] Building CXX object CMakeFiles/org_ros2_rcljava_node__node_impl.dir/src/main/cpp/org_ros2_rcljava_node_NodeImpl.cpp.o
/home/test/ros2_java_ws/src/ros2_java/ros2_java/rcljava/src/main/cpp/org_ros2_rcljava_node_NodeImpl.cpp: In function ‘jlong Java_org_ros2_rcljava_node_NodeImpl_nativeCreateTimerHandle(JNIEnv*, jclass, jlong)’:
/home/test/ros2_java_ws/src/ros2_java/ros2_java/rcljava/src/main/cpp/org_ros2_rcljava_node_NodeImpl.cpp:223:88: error: invalid conversion from ‘jlong {aka long int}’ to ‘rcl_clock_t*’ [-fpermissive]
   rcl_ret_t ret = rcl_timer_init(timer, timer_period, NULL, rcl_get_default_allocator());
                                                                                        ^
/home/test/ros2_java_ws/src/ros2_java/ros2_java/rcljava/src/main/cpp/org_ros2_rcljava_node_NodeImpl.cpp:223:88: warning: converting to non-pointer type ‘long int’ from NULL [-Wconversion-null]
/home/test/ros2_java_ws/src/ros2_java/ros2_java/rcljava/src/main/cpp/org_ros2_rcljava_node_NodeImpl.cpp:223:88: error: cannot convert ‘rcutils_allocator_t’ to ‘rcl_timer_callback_t {aka void (*)(rcl_timer_t*, long int)}’ for argument ‘4’ to ‘rcl_ret_t rcl_timer_init(rcl_timer_t*, rcl_clock_t*, int64_t, rcl_timer_callback_t, rcl_allocator_t)’
CMakeFiles/org_ros2_rcljava_node__node_impl.dir/build.make:62: recipe for target 'CMakeFiles/org_ros2_rcljava_node__node_impl.dir/src/main/cpp/org_ros2_rcljava_node_NodeImpl.cpp.o' failed
make[2]: *** [CMakeFiles/org_ros2_rcljava_node__node_impl.dir/src/main/cpp/org_ros2_rcljava_node_NodeImpl.cpp.o] Error 1
CMakeFiles/Makefile2:173: recipe for target 'CMakeFiles/org_ros2_rcljava_node__node_impl.dir/all' failed
make[1]: *** [CMakeFiles/org_ros2_rcljava_node__node_impl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
1 warning
[ 23%] Generating CMakeFiles/rcljava_jar.dir/java_class_filelist
[ 28%] Creating Java archive rcljava.jar
[ 28%] Built target rcljava_jar
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/test/ros2_java_ws/build_isolated/rcljava/cmake__build.sh && /usr/bin/make -j2 -l2' failed in '/home/test/ros2_java_ws/build_isolated/rcljava' with exit code '2'
<== Command '. /home/test/ros2_java_ws/build_isolated/rcljava/cmake__build.sh && /usr/bin/make -j2 -l2' failed in '/home/test/ros2_java_ws/build_isolated/rcljava' with exit code '2'
```

songshan0321, 2018-07-31:
I'm facing exactly the same issue too. Hope it can be solved soon!

esteve, 2018-07-31:
Thanks for trying out ros2-java and for the report. Unfortunately the interface for timers changed between `Bouncy` and `master`, so until I have more time to update `rcljava` for master, I've switched all the repositories to the same branches that Bouncy has.

esteve, 2018-08-01:
Can you try building ros2-java again? I temporarily reverted the repositories to Bouncy in https://github.com/esteve/ros2_java/commit/2b256f8ef935b48b0a6c7cd255d4652edf757825 so that CI passes.

songshan0321, 2018-08-02:
Yup, ros2-java is built successfully on desktop with skipping test_msgs.
But I have issue when compiling on Android:
```
Failed packages: ament_cmake_ros
Traceback (most recent call last):
  File "/home/rmf/ament_ws/install_isolated/ament_tools/bin/ament", line 11, in <module>
    load_entry_point('ament-tools', 'console_scripts', 'ament')()
  File "/home/rmf/ament_ws/build_isolated/ament_tools/ament_tools/commands/ament.py", line 88, in main
    rc = args.main(args)
  File "/home/rmf/ament_ws/build_isolated/ament_tools/ament_tools/verbs/build/cli.py", line 172, in main
    return iterate_packages(opts, packages, per_package_main)
  File "/home/rmf/ament_ws/build_isolated/ament_tools/ament_tools/verbs/build/cli.py", line 331, in iterate_packages
    os.symlink(template_path, dst)
FileNotFoundError: [Errno 2] No such file or directory: '/home/rmf/ament_ws/build_isolated/ament_package/ament_package/template/isolated_prefix_level/local_setup.bash' -> '/home/rmf/ros2_android_ws/install_isolated/local_setup.bash'
```

fsal33m, 2018-08-02:
Thanks, Cl passed.

closed 2018-08-02 by fsal33m