ros2-java/ros2_java #55: NDK is missing a "platforms" directory.

songshan0321, 2018-08-06:
Error when compile ros2_listener_android.

These are some path I have set.
export JAVA_HOME="/usr/lib/jvm/default-java"
export ANDROID_HOME="~/Android/Sdk"

```
+++ Building 'ros2_listener_android'
==> '/usr/bin/gradle -Pament.source_space=/home/rmf/ros2_android_ws/src/ros2_java/ros2_android_examples/ros2_listener_android -Pament.build_space=/home/rmf/ros2_android_ws/build_isolated/ros2_listener_android -Pament.install_space=/home/rmf/ros2_android_ws/install_isolated/ros2_listener_android -Pament.dependencies=/home/rmf/ros2_android_ws/install_isolated/ament_cmake_export_jars/share/ament_cmake_export_jars:/home/rmf/ros2_android_ws/install_isolated/ament_cmake_export_jni_libraries/share/ament_cmake_export_jni_libraries:/home/rmf/ros2_android_ws/install_isolated/fastcdr/share/fastcdr:/home/rmf/ros2_android_ws/install_isolated/fastrtps/share/fastrtps:/home/rmf/ros2_android_ws/install_isolated/fastrtps_cmake_module/share/fastrtps_cmake_module:/home/rmf/ros2_android_ws/install_isolated/poco_vendor/share/poco_vendor:/home/rmf/ros2_android_ws/install_isolated/rcljava_common/share/rcljava_common:/home/rmf/ros2_android_ws/install_isolated/rcutils/share/rcutils:/home/rmf/ros2_android_ws/install_isolated/rmw/share/rmw:/home/rmf/ros2_android_ws/install_isolated/rmw_implementation_cmake/share/rmw_implementation_cmake:/home/rmf/ros2_android_ws/install_isolated/rosidl_parser/share/rosidl_parser:/home/rmf/ros2_android_ws/install_isolated/rosidl_cmake/share/rosidl_cmake:/home/rmf/ros2_android_ws/install_isolated/rosidl_typesupport_interface/share/rosidl_typesupport_interface:/home/rmf/ros2_android_ws/install_isolated/rosidl_generator_c/share/rosidl_generator_c:/home/rmf/ros2_android_ws/install_isolated/rosidl_generator_cpp/share/rosidl_generator_cpp:/home/rmf/ros2_android_ws/install_isolated/rosidl_typesupport_introspection_c/share/rosidl_typesupport_introspection_c:/home/rmf/ros2_android_ws/install_isolated/rosidl_typesupport_c/share/rosidl_typesupport_c:/home/rmf/ros2_android_ws/install_isolated/rosidl_typesupport_introspection_cpp/share/rosidl_typesupport_introspection_cpp:/home/rmf/ros2_android_ws/install_isolated/rmw_fastrtps_cpp/share/rmw_fastrtps_cpp:/home/rmf/ros2_android_ws/install_isolated/rmw_implementation/share/rmw_implementation:/home/rmf/ros2_android_ws/install_isolated/rosidl_generator_java/share/rosidl_generator_java:/home/rmf/ros2_android_ws/install_isolated/rosidl_default_generators/share/rosidl_default_generators:/home/rmf/ros2_android_ws/install_isolated/rosidl_typesupport_cpp/share/rosidl_typesupport_cpp:/home/rmf/ros2_android_ws/install_isolated/rosidl_default_runtime/share/rosidl_default_runtime:/home/rmf/ros2_android_ws/install_isolated/builtin_interfaces/share/builtin_interfaces:/home/rmf/ros2_android_ws/install_isolated/rcl_interfaces/share/rcl_interfaces:/home/rmf/ros2_android_ws/install_isolated/std_msgs/share/std_msgs:/home/rmf/ros2_android_ws/install_isolated/rcl/share/rcl:/home/rmf/ros2_android_ws/install_isolated/rcljava/share/rcljava:/home/rmf/ros2_android_ws/install_isolated/rclandroid/share/rclandroid -Pament.build_tests=False -Pament.package_manifest.name=ros2_listener_android -Pament.exec_dependency_paths_in_workspace=/home/rmf/ros2_android_ws/install_isolated/rclandroid/share/rclandroid:/home/rmf/ros2_android_ws/install_isolated/builtin_interfaces/share/builtin_interfaces:/home/rmf/ros2_android_ws/install_isolated/std_msgs/share/std_msgs -Pament.gradle_recursive_dependencies=False -Pament.gradle_isolated=True -Pament.android_stl=gnustl_shared -Pament.android_abi=armeabi-v7a -Pament.android_ndk=~/Android/Sdk/ndk-bundle -Pament.android_variant=release assemble --stacktrace' in '/home/rmf/ros2_android_ws/build_isolated/ros2_listener_android'

> Configure project :
NDK is missing a "platforms" directory.
If you are using NDK, verify the ndk.dir is set to a valid NDK directory.  It is currently set to /home/rmf/ros2_android_ws/build_isolated/ros2_listener_android/~/Android/Sdk/ndk-bundle.
If you are not using NDK, unset the NDK variable from ANDROID_NDK_HOME or local.properties to remove this warning.

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring root project 'ros2_listener_android'.
> The SDK directory '~/Android/Sdk' does not exist.
```

closed 2018-08-07 by songshan0321

esteve, 2018-08-08:
How did you solve it? Is it because the tilde (`~`) is not a valid character?

songshan0321, 2018-08-17:
It was solved after I set my ANDROID_NDK.