ros2-java/ros2_java #57: Destroy/Shutdown a ROS2 Node in Android

songshan0321, 2018-08-21:
I am trying to restart nodes from Android, but it doesn't kill nodes, wondering if there is any function to do it?
```c++
      // Kill nodes
      baseControllerNode = null;
      listenerNode = null;
      // Start a ROS2 publisher node
      baseControllerNode = new BaseControllerNode("teleop_twist_joy", "cmd_vel");
      // Start a ROS2 subscriber node
      listenerNode = new ListenerNode("ldr_listener", "ldr_value", listenerView, dateView);

      // Start listening to /ldr_value
      getExecutor().addNode(listenerNode);
```

johnmarkwayve, 2018-08-29:
Java objects are not immediately destroyed when you null out the corresponding variable. At some unpredictable point (assuming there's no living reference left to the object) the Java garbage collector will come around and actually destroy it, that's why your node isn't getting destroyed when you expect.

It seems that a `dispose` method has been provided on `Node` for this very purpose though.

songshan0321, 2018-08-29:
Thanks for ur response! I have tried that but it seems like there is no dispose() method on Node object.

johnmarkwayve, 2018-08-29:
I don't have an installation to hand to check, but it really should. See, for example, https://github.com/esteve/ros2_java/blob/master/rcljava/src/test/java/org/ros2/rcljava/node/NodeTest.java#L176

jacobperron, 2021-11-13:
Certainly, newer versions of ros2_java have a dispose method.

Since this ticket is very old I'm going to close it. Please open a new  ticket if you run into the same issue.

closed 2021-11-13 by jacobperron