ros2-java/ros2_java #60: Support for ROS2 Crystal

ruffsl, 2018-12-30:
While following along with the readme, I encountered some build issues when building from ROS2 Crystal. @esteve , perhaps you might see the IDL issue generated from the following example Dockerfiles? I'd like to eventually get to making a dockerfile to build android apk for ROS2 with.

### Dockerfile

<details>

``` Dockerfile
FROM ros:crystal-ros-base-bionic

# install ros build tools
RUN apt-get update && apt-get install -y \
      default-jdk \
      gradle \
      python3-colcon-common-extensions && \
    rm -rf /var/lib/apt/lists/*

# clone ros package repo
ENV ROS_WS /opt/ros_ws
RUN mkdir -p $ROS_WS/src
WORKDIR $ROS_WS
RUN git -C src clone \
      https://github.com/esteve/ament_java.git
RUN git -C src clone \
      https://github.com/esteve/ros2_java.git
RUN git -C src clone \
      https://github.com/esteve/ros2_java_examples.git

# install ros package dependencies
RUN apt-get update && \
    rosdep update && \
    rosdep install -y \
      --from-paths \
        src/ \
      --ignore-src && \
    rm -rf /var/lib/apt/lists/*

# build ros package source
RUN . /opt/ros/$ROS_DISTRO/setup.sh && \
    colcon build \
      --cmake-args \
        -DCMAKE_BUILD_TYPE=Release

# source ros package from entrypoint
RUN sed --in-place --expression \
      '$isource "$ROS_WS/install/setup.bash"' \
      /ros_entrypoint.sh

# run ros package launch file
CMD ["ros2", "run", "rcljava_examples", "publisher_lambda"]
```

</details>

### Build output:

<details>

```
$ docker build .
Sending build context to Docker daemon  3.072kB
Step 1/9 : FROM ros:crystal-ros-base-bionic
 ---> f7af64b98e6e
Step 2/9 : RUN apt-get update && apt-get install -y       default-jdk       gradle       python3-colcon-common-extensions &&     rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> e32cdb65afc8
Step 3/9 : ENV ROS_WS /opt/ros_ws
 ---> Using cache
 ---> 2aac0bf2ddcd
Step 4/9 : RUN mkdir -p $ROS_WS/src
 ---> Using cache
[... 188 lines cut ...]
Call Stack (most recent call first):
  /opt/ros/crystal/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  /opt/ros/crystal/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:280 (ament_execute_extensions)
  CMakeLists.txt:73 (rosidl_generate_interfaces)

CMake Error at cmake/rosidl_generator_java_generate_interfaces.cmake:251 (add_dependencies):
  The dependency target
  "rosidl_generator_java__rosidl_typesupport_fastrtps_c" of target
  "msgUint64fastrtps_c" does not exist.
Call Stack (most recent call first):
  /opt/ros/crystal/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  /opt/ros/crystal/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:280 (ament_execute_extensions)
  CMakeLists.txt:73 (rosidl_generate_interfaces)

---
Failed   <<< rosidl_generator_java	[ Exited with code 1 ]

Summary: 5 packages finished [8.00s]
  1 package failed: rosidl_generator_java
  2 packages had stderr output: rcljava_common rosidl_generator_java
  1 package not processed
The command '/bin/sh -c . /opt/ros/$ROS_DISTRO/setup.sh &&     colcon build       --cmake-args         -DCMAKE_BUILD_TYPE=Release' returned a non-zero code: 1
```

</details>

esteve, 2018-12-30:
I haven't tried `ros2_java` with Crystal yet, but the problem seems to come from using the binary distribution of ROS2 instead of building the workspace from scratch (see https://github.com/esteve/ros2_java/issues/47).

The Java code generator is integrated into the `rosidl_generator_packages` virtual package, but at the moment it can't be mixed with an existing ROS2 installation. Could you try building Crystal and `ros2_java` in the same workspace? Thanks.

ruffsl, 2018-12-31:
> Could you try building Crystal and ros2_java in the same workspace?

About the same issue, please see the following build:

### Dockerfile

<details>

``` Dockerfile
FROM ubuntu:bionic

# setup timezone
RUN echo 'Etc/UTC' > /etc/timezone && \
    ln -s /usr/share/zoneinfo/Etc/UTC /etc/localtime && \
    apt-get update && apt-get install -q -y tzdata && rm -rf /var/lib/apt/lists/*

# install packages
RUN apt-get update && apt-get install -q -y \
      bash-completion \
      dirmngr \
      git \
      gnupg2 \
      libasio-dev \
      libtinyxml2-dev \
      lsb-release \
      python3-pip \
      wget \
    && rm -rf /var/lib/apt/lists/*

# setup ros2 keys
RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 421C365BD9FF1F717815A3895523BAEEB01FA116

# setup sources.list
RUN echo "deb http://packages.ros.org/ros2/ubuntu `lsb_release -sc` main" > /etc/apt/sources.list.d/ros2-latest.list

# setup environment
ENV LANG C.UTF-8
ENV LC_ALL C.UTF-8

# install packages from the ROS repositories
RUN apt-get update && apt-get install -y --no-install-recommends \
      python3-catkin-pkg-modules \
      python3-colcon-common-extensions \
      python3-rosdep \
      python3-vcstool \
    && rm -rf /var/lib/apt/lists/*

# install python packages
RUN pip3 install -U \
      argcomplete \
      flake8 \
      flake8-blind-except \
      flake8-builtins \
      flake8-class-newline \
      flake8-comprehensions \
      flake8-deprecated \
      flake8-docstrings \
      flake8-import-order \
      flake8-quotes \
      pytest-repeat \
      pytest-rerunfailures

# bootstrap rosdep
ENV ROSDISTRO_INDEX_URL https://raw.githubusercontent.com/ros2/rosdistro/ros2/index.yaml
RUN rosdep init \
    && rosdep update

# install ros build tools
RUN apt-get update && apt-get install -y \
      default-jdk \
[... 484 lines cut ...]
  /root/ros2_ws/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:280 (ament_execute_extensions)
  CMakeLists.txt:73 (rosidl_generate_interfaces)

CMake Error at cmake/rosidl_generator_java_generate_interfaces.cmake:251 (add_dependencies):
  The dependency target
  "rosidl_generator_java__rosidl_typesupport_fastrtps_c" of target
  "msgUint64fastrtps_c" does not exist.
Call Stack (most recent call first):
  /root/ros2_ws/install/ament_cmake_core/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  /root/ros2_ws/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:280 (ament_execute_extensions)
  CMakeLists.txt:73 (rosidl_generate_interfaces)

---
Failed   <<< rosidl_generator_java	[ Exited with code 1 ]
[0m]0;colcon build [87/102 done] [2 ongoing]]0;colcon build [88/102 done] [1 ongoing]Aborted  <<< rosidl_generator_py
]0;colcon build [89/102 done] [0 ongoing]
Summary: 87 packages finished [2min 3s]
  1 package failed: rosidl_generator_java
  1 package aborted: rosidl_generator_py
  6 packages had stderr output: rcljava_common rmw_connext_cpp rmw_connext_shared_cpp rosidl_generator_java rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp
  13 packages not processed
The command '/bin/sh -c colcon     build     --cmake-args       -DSECURITY=ON     --no-warn-unused-cli     --symlink-install     --packages-up-to       rcljava_examples' returned a non-zero code: 1
```

</details>

kerliang0801, 2019-10-03:
> I've started updating `rosidl_generator_java` to work with the new pipeline. Specifically, I plan to target Dashing (skipping Crystal); making changes to the rest of `ros2_java` as necessary.
> 
> Ultimately, I'll be working towards feature parity with `rclcpp` and `rclpy`.

@jacobperron Really appreciate your effort and contributions!! You're awesome:D Just wondering if there's a general timeline on the update?

jacobperron, 2019-10-04:
> Just wondering if there's a general timeline on the update?

Timelines are hard :sweat_smile: 

I've opened up #66, which should let you compile `rosidl_generator_java` for Dashing. I'm working on `rcljava` now, hopefully you'll see a PR for that one soon.

jacobperron, 2020-02-05:
Considering Crystal is EOL, I'm going to close this.

I'll be putting in some effort to get ROS 2 Java support for Foxy (and Dashing/Eloquent along the way). Here's a metaticket tracking progress: https://github.com/ros2-java/ros2_java/issues/82

closed 2020-02-05 by jacobperron