ros2-java/ros2_java #63: Android NDK support?

machinekoder, 2019-08-21:
I haven't found any documentation on how to use ROS2 natively on Android with C++/NDK applications.

Does `ros2_java` also build and deploy the native libs? If so, it would help to mention it in the README for other folks that want to use ROS2 natively, for example in their C++/Qt apps.

esteve, 2019-08-21:
You need the NDK in order to build `ros2_java` for Android, as detailed in https://github.com/esteve/ros2_java#android

`ros2_java` is a mix of C++ and Java, it builds upon a subset of ROS 2 crosscompiled for Android and a layer written in C++ and Java for developing applications using the Android API.

jacobperron, 2021-11-13:
Closing in favor of tracking Android build issues in https://github.com/ros2-java/ros2_java/issues/81 and #137

closed 2021-11-13 by jacobperron