ros2-java/ros2_java #67: "Built target builtin_interfaces__rosidl_typesupport_introspection_cpp" failed

dimasoktanugraha, 2019-10-08:
when i'm in desktop installation at ament build --symlink-install --isolated, i got this error
[image]

charlotteheggem, 2019-11-25:
Hi, I have faced a similar problem with "action_msgs__java" instead of "builtin_interfaces__java".
Did you find a solution for this?

Jconn, 2019-11-25:
I haven't gotten opensplice (or cyclone) to compile when targeting android. Do you get the same issue building with only fastrtps?

dimasoktanugraha, 2019-11-26:
still don't know how to solve this,

charlotteheggem, 2019-11-26:
@jacobperron @esteve Do you know what could be wrong?

jacobperron, 2019-11-27:
@charlotteheggem It sounds related to https://github.com/ros2/ros2/issues/752. If it's the same issue, it was [fixed](https://github.com/ros2/rosidl_typesupport_opensplice/pull/33) in later versions of rosidl_typesupport_opensplice (Crystal onwards). I suspect that since rcljava assumes the ROS 2 Bouncy, it is subsequently not working with newer version of OpenSplice.

esteve, 2019-12-04:
@charlotteheggem @jacobperron is right about ros2_java assuming Bouncy. Fortunately, he has submitted #66 to update ros2_java to Dashing :-)

esteve, 2019-12-04:
@Jconn is Android a supported platform for Cyclone? I haven't tested it, though.

Jconn, 2019-12-04:
cyclone and opensplice failed to build against the android ndk. I moved on to fastRTPS

year112, 2020-02-21:
you can find rosidl_typesupport_opensplice_cpp/__init__.py
and delete that part
```
else:
        assert False, "Failed to find '// Generated: ' line"
```

jacobperron, 2021-11-13:
Closing since this is old and we are focused on supporting ROS Galactic. Please checkout the `main` branch :)

closed 2021-11-13 by jacobperron