ros2-java/ros2_java #69: Shared Maintainership

jacobperron, 2020-01-13:
@esteve Currently, you are the sole maintainer of the core ROS 2 Java repositories. A couple of us at Open Robotics (myself and @clalancette) are willing to share the load :slightly_smiling_face: 

I'd like to request administrative access for us so that we can help develop and maintain ROS 2 Java.

esteve, 2020-01-14:
@jacobperron I added you to the team so you could review and merge code, and create new repositories, so that should be enough for developing ros2-java.

Regarding adding @clalancette, in the spirit of keeping ros2-java an open source project where contributors push their code and eventually become committers, I don't think it's fair to add someone to the team without them having contributed code first.

Long story short, I'd rather keep this project separate and community-driven, and not under the umbrella of Open Robotics.

jacobperron, 2020-01-14:
> Long story short, I'd rather keep this project separate and community-driven, and not under the umbrella of Open Robotics.

That's fair.

I can add new repositories, but am not able to push or merge code to the existing repositories. I also don't have access to configure CI, which might be useful.

jacobperron, 2020-02-05:
I have authority to merge/close tickets now. Thanks!

closed 2020-02-05 by jacobperron

esteve, 2020-02-05:
@jacobperron no worries, thanks for your help!