ros2-java/ros2_java #71: Action Msgs

mhubii, 2020-01-16:
Hi Steve, and Jacob,

when running the build script in ci-scipts, the action_msgs package failes to build. I am on Ubuntu 18.04 with Gradle 6.0.1, openjdk version 11.0.5. Sry I don't know much about Java. 

Best
Martin

mhubii, 2020-01-16:
Adding

```
  ros2/unique_identifier_msgs:
    type: git
    url: https://github.com/ros2/unique_identifier_msgs.git
    version: master
```

to ros2_java_desktop.repos resolves it but then another package is missing

```
CMake Error at CMakeLists.txt:21 (find_package):
  By not providing "Findtest_interface_files.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "test_interface_files", but CMake did not find one.
```

However, then the compiler complains that

```
  rosidl_generate_interfaces() the passed idl file
  'ros2_java_ws/src/ros2_java/ci-scripts/output/install_isolated_java/test_interface_files/share/test_interface_files:msg/Arrays.msg'
  does not exist
```

But actually the file sits there as a symbolic link. Anyway, I don't understand why the test_interface_files are required

jacobperron, 2020-01-29:
The master branch is only compatibility with ROS 2 Bouncy. Are you using Bouncy versions for all ROS 2 packages?

Regarding the dashing branch, it isn't yet fully compatible with Dashing. Only changes for message generation have been merged, I've just opened a PR that should make the `rcljava` package work with Dashing as well (https://github.com/ros2-java/ros2_java/pull/75). I'm still debugging issues related to services though.

mhubii, 2020-02-03:
thanks for the hint @jacobperron, no I haven't used bouncy so far. I'll try that

talregev, 2020-02-03:
https://github.com/ros2-java/ros2_java/tree/master/rcljava/src/main/cpp
is not competible with ros dashing. the api was change.

jacobperron, 2021-11-13:
Please try the `main` branch with ROS Galactic.

Dashing is EOL.

closed 2021-11-13 by jacobperron