ros2-java/ros2_java #81: Colcon and ament_java

talregev, 2020-02-03:
Hi all,
I compile the ros2_java_android.repos with colcon: the official build system for ros2.
How I can compile ament_java projects and ament_android projects with colcon?

Thanks for your help.

talregev, 2020-02-04:
I found a solution although i didn't succeed to compile android example. (Why it so hard?)
Instead of ament_tools you can have 2 repositories: 

https://github.com/colcon/colcon-gradle
https://github.com/colcon/colcon-ros-gradle

I will add them to pr

talregev, 2020-02-15:
Can you  change the guid how I can compile ros2 java for android
when compile it with ros2 dashing, and colcon?
Please help me.

jacobperron, 2021-11-13:
If I'm not mistaken, building Android packages (like the examples) needs some work and depends on https://github.com/ros2-java/ros2_java/issues/137.

Any help in with this is much appreciated.