ros2-java/ros2_java #82: Foxy Fitzroy Roadmap

jacobperron, 2020-02-05:
This is a meta-ticket with a list of tasks for ROS 2 Java support targeting [Foxy](https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/). These are tasks that I'm scheduling for myself, posting here for visibility. I'm not imposing a specific timeline on anything, but it would be nice to get things in for the release date.

If anyone would like to help out on any of the items (implementation or review), please comment on this ticket!

## Roadmap

*Will be updated as work progresses and tickets are opened*

- [ ] Dashing / Eloquent support :construction_worker_man:
  - [ ] rosidl_generator_java (https://github.com/ros2-java/ros2_java/pull/66) (https://github.com/ros2-java/ros2_java/pull/68) (https://github.com/ros2-java/ros2_java/pull/76) :mag: 
  - [ ] rcljava (https://github.com/ros2-java/ros2_java/pull/75) :mag:
  - [ ] Get CI running
- [ ] Refactor as necessary to support many nodes per ROS context
  - Depends on https://github.com/ros2/design/pull/250 and https://github.com/ros2/rmw/issues/183 for more info. We may not need to make any changes in rcljava.
- [ ] Retroactive interface generation
  - It would be good to have a mechanism to invoke the rosidl_generator_java pipeline for packages already installed. This would enable rcljava to be installed without having to build all of ROS 2 from source. On a somewhat related note, it sounds like [rclnodejs accomplishes this, perhaps with some manual steps required](https://github.com/RobotWebTools/rclnodejs/issues/483).
- [ ] Figure out how to package Gradle packages
  - In particular, AMENT_PACKAGE_PREFIX and CLASSPATH are not being set (see https://github.com/ros2-java/ament_java/issues/9)
- [ ] Architecture / API review
  - Compare and contract with [rclcpp]() and [rclpy]()
  - Propose changes that make rcljava more consistent with the other client libraries
  - Fix/deprecate API that no longer makes sense.
- [ ] [Actions](http://design.ros2.org/articles/actions.html)
  - [ ] Action server
  - [ ] Action client
- [ ] Intra-process communication

nielstiben, 2020-05-07:
Hello Jacob,

My name is Niels, a fourth-year Software Engineering bachelor student. I’m the person who offered to contribute to action support for rcljava (see ["Official” Java Client Library](https://discourse.ros.org/t/official-java-client-library/13389/20?u=niels)). 
I’m new to ROS 2. Also, I’ve never worked with real C++ projects before, which means that the development can take longer. But I’m willing to learn!

I was wondering if you could recommend a particular IDE for rcljava development.
Today, I tried to configure my IDE (IntelliJ) for rcljava development. The Java files and its dependencies were recognised, I was even able to run the examples from the example project. Rcljava’s testing package is the only package that gives problems now; it does not seem to find “_rcljava.msg._” classes. It appears that they (I followed the official [ros2-java instructions](https://github.com/ros2-java/ros2_java/tree/dashing)). 
In the end, IntelliJ does not seem to be great for C++ development; so I tried CLion too. CLion seems to lack proper Java development support, so I’m curious what tooling/IDE you are using for rcljava-development.

Also, I would like to hear if you already started on certain functionality I can build upon too. 
Due to my ROS2 experience, it would help if you can steer me in any direction. 

Thank you in advance for your help and time. In the meantime, I'll be trying to get more familiar with rcljava. 

Niels Tiben

esteve, 2020-05-07:
@nielstiben nice to see you contributing to `ros2-java`, thanks! :slightly_smiling_face: 

This is tricky, because `rcljava` uses four languages C, C++, Java and Python, so you'd need a really complete IDE if you expect to have the same features for all the languages.

Personally, I use VSCode, but mostly because it's fast and has an adequate set of features for all these languages (e.g. its support for Java is not as good as IntelliJ nor it's as good for C++ as CLion, but it's good enough for me overall).

For developing application that use `rcljava`, you may use any IDE that supports Java, `rcljava` has support for the more traditional Java toolchain, such as integration with Gradle.

What would you like to work on? I recall your advisor mentioned you're using actions in your project, would you like to give it a shot?

Thanks!

nielstiben, 2020-05-07:
@esteve 
Thank you for sharing your experiences with IDEs.
You are right about the note of my advisor. I'm indeed planning to work on rcljava action support. Due to my experience, it might be good to start with action client support. I assume that action-client implementation is easier than action-server implementation. 

Although it is (probably) going to be difficult, I'm really willing to give it a shot.

jacobperron, 2020-05-07:
I don't typically use an IDE, but when I do I reach for VSCode as well.

Regarding a starting point, starting with the client sounds good to me. My suggestion is to first familiarize yourself with the [design doc](http://design.ros2.org/articles/actions.html) about actions, and then maybe take a look at the [Python](https://github.com/ros2/examples/tree/master/rclpy/actions) and [C++](https://github.com/ros2/examples/tree/master/rclcpp/actions) examples. From there, it would be nice to propose some Java examples. Having a draft PR open gives us a place to discuss what we'd like the API to look like (they don't have to compile).

nielstiben, 2020-05-20:
@jacobperron  @esteve 
I [forked](https://github.com/nielstiben/ros2_java/tree/action_client_support_bug) the [ros2_java](https://github.com/ros2-java/ros2_java/tree/dashing) and its corresponding [examples](https://github.com/ros2-java/ros2_java_examples) project. The feature branch that I created contains the initial version of rcljava's action client support on which I'm working on.

I implemented boilerplate code for the action client implementation and I managed to get a simple action started via an example. I tried it together with the cpp [action server of the ROS2 rclcpp example](https://github.com/ros2/examples/blob/dashing/rclcpp/minimal_action_server/not_composable.cpp). So from **rcljava**, an action is started that is effectively picked up by an **rclcpp** action server, which is good. For this I use the [Fibonacci example](https://github.com/nielstiben/ros2_java_examples/blob/action_client_support/rcljava_examples/src/main/java/org/ros2/rcljava/examples/action_client/FibonacciActionClient.java).

However, I have a problem sending the action goal message. It seems that the action goal message loses its payload when it is sent from the java implementation (via JNI) to the cpp implementation. In the (java) example I [set the fibonacci 'order' to 8](https://github.com/nielstiben/ros2_java_examples/blob/5bd4d300668dce7c19d1e601a93aa6ab27e79ba4/rcljava_examples/src/main/java/org/ros2/rcljava/examples/action_client/FibonacciActionClient.java#L43). However, in cpp 'order' [always returns 0](https://github.com/nielstiben/ros2_java/blob/e6ec40e946dd64ed5f99b30175747b4fb6339d9b/rcljava/src/main/cpp/org_ros2_rcljava_action_client_ActionClientImpl.cpp#L65). 

When I hard-code a _Fibonacci goal message_ in the action client cpp implementation with a specified _order_, it does actually work. So it really seems that something goes wrong when the action goal message gets passed from java to cpp.

Would you mind having a look at my feature branch to determine what's wrong?

wilcobonestroo, 2020-05-25:
Dear @jacobperron or @esteve, Would it be possible to have a short telco (skype / teams / etc.) to get us on track on this topic? We are trying to get our head around it, but we are still not sure if we are thinking in the right direction. We would really appreciate it.

esteve, 2020-05-25:
@nielstiben @wilcobonestroo 

> Would it be possible to have a short telco (skype / teams / etc.) to get us on track on this topic?

Sure thing, send us email to esteve.fernandez (at) gmail.com and jacob (at) openrobotics.org and we can schedule a time. Jacob is based in the west coast of the US, but I'm here in Europe, so it may be easier for you guys.

esteve, 2020-05-25:
@nielstiben 

> Would you mind having a look at my feature branch to determine what's wrong?

It sounds weird indeed, especially because passing objects between C++ and Java is well tested elsewhere in the codebase, let me have a look at it and get back to you.

esteve, 2020-05-25:
@nielstiben 

You shouldn't be using a `reinterpret_cast` here https://github.com/nielstiben/ros2_java/blob/e6ec40e946dd64ed5f99b30175747b4fb6339d9b/rcljava/src/main/cpp/org_ros2_rcljava_action_client_ActionClientImpl.cpp#L62-L63

Have you tried with a `static_cast` instead?

nielstiben, 2020-05-27:
> @nielstiben
> 
> You shouldn't be using a `reinterpret_cast` here https://github.com/nielstiben/ros2_java/blob/e6ec40e946dd64ed5f99b30175747b4fb6339d9b/rcljava/src/main/cpp/org_ros2_rcljava_action_client_ActionClientImpl.cpp#L62-L63
> 
> Have you tried with a `static_cast` instead?

Yes, I have tried that. It returns the same result, as in always returns value `0`. 

I just tried the **exact** same approach of printing an object's value in the (already finished) [client implementation](https://github.com/ros2-java/ros2_java/blob/dashing/rcljava/src/main/cpp/org_ros2_rcljava_client_ClientImpl.cpp), and printing the request object values is working perfectly fine there. 
When I run the Client Example (AddTwoInts), I am able to print `a` and `b` values from the `AddTwoInts_Request object` in the C++ implementation. The C++ code is showing returning the request object's actual value.

For some reason, the ActionClient implementation does _not_ recognise the actual value of the request object and always returns value `0`.

jacobperron, 2020-05-27:
Sorry, I haven't had a chance to take a look (I've been busy with Foxy releases tasks). I'm happy to join a teleconference. I should have more time to invest in rcljava after the initial release of Foxy.

nielstiben, 2020-05-29:
Yesterday, @wilcobonestroo and I had an online conversation with @esteve to get more familiar with the source code and to walk through an issue I am currently facing ([see my comment above](https://github.com/ros2-java/ros2_java/issues/82#issuecomment-631480901)).

The issue seemed related to a bug in generating code for action requests because:
-  The `<action_name>_SendGoal_Request__convert_from_java` and other functions where not generated. 
- Also, generated code that belonged to `<action_name>_FeedbackMessage` was not only placed inserted into the `_FeedbackMessage`-files, but also into the `_SendGoal`-files. 

I created an [MR](https://github.com/ros2-java/ros2_java/pull/108), and these issues should be fixed.
___
However, this MR does not seem to fix my [initial issue](https://github.com/ros2-java/ros2_java/issues/82#issuecomment-631480901). Luckily, I have figured some way to get it functional.

In the C++ implementation, I'm am receiving an `test_msgs__action__Fibonacci_Goal` from the JNI.
The `rcl_action_send_goal_request` expects `test_msgs__action__Fibonacci_SendGoal_Request` instead. AFAIK, `_SendGoal_Request` is a container that holds the `_Goal` object.

It is not difficult to generate a `_SendGoal_Request` object that holds a `_Goal`, as I did in [my feature branch](https://github.com/nielstiben/ros2_java/blob/bfffc65d658b6411015672d2533a95841236ab1d/rcljava/src/main/cpp/org_ros2_rcljava_action_client_ActionClientImpl.cpp#L66). I got it working for the Fibonacci-example, but my implementation is very static and not working with any other type at the moment.

Is it possible to generate an 'uknown' `_SendGoal_Request` that works with can hold any goal type? I tried do something like [this](https://github.com/ros2/rclcpp/blob/3e5b6a47ea66505ecaa4896802095c7229466fa2/rclcpp_action/include/rclcpp_action/client.hpp#L349) (rclcpp), but that doesn't seem to work in my case.

jacobperron, 2021-11-13:
I've let this metaticket go stale, so I'm going to close it out.

To wrap up the current state: we've done a lot of work on the osrf fork and have started migrating changes upstream. Due to resource limitations, the focus will be on supporting the latest ROS distro (currently Galactic).

closed 2021-11-13 by jacobperron