ros2-java/ros2_java #86: Build in dashing version failed,in Ubuntu 18.0.4

year112, 2020-02-21:
In README.md 
`mkdir -p ament_ws/src
cd ament_ws
curl -skL https://raw.githubusercontent.com/esteve/ament_java/master/ament_java.repos -o ament_java.repos
vcs import src < ament_java.repos
src/ament/ament_tools/scripts/ament.py build --symlink-install --isolated`
and 
`mkdir -p ros2_java_ws/src
cd ros2_java_ws
curl -skL https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop.repos -o ros2_java_desktop.repos
vcs import src < ros2_java_desktop.repos
. ../ament_ws/install_isolated/local_setup.sh
ament build --symlink-install --isolated
`
ament_java.repos and ros2_java_desktop.repos is out of date in 2018 .
When i build in dashing version ,there has too many mistakes.

[image]

how can i fix it？ by the way ，when you will maintain the latest version of the instructions（dashing 2020 or 2019）

nielstiben, 2020-03-26:
I experienced similar issues. Have you tried cloning the `dashing` branch instead of the `master` branch? 

Replace `curl -skL https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop.repos -o ros2_java_desktop.repos`
with `curl -skL https://raw.githubusercontent.com/esteve/ros2_java/dashing/ros2_java_desktop.repos -o ros2_java_desktop.repos`

wilcobonestroo, 2020-03-29:
When I did the steps described by @nielstiben I ended up with two packages that could not build: `rcljava_example`s and `test_msgs`. The issue with `test_msgs` can be solved with #100 I can now build everything except for the examples with `ament build --skip-packages rcljava_examples --symlink-install --isolated`

wilcobonestroo, 2020-03-29:
When I try to build jcljava_examples, the error I get is 
```
/home/ros/ros2_java_ws/build_isolated/rcljava_examples/src/main/java/org/ros2/rcljava/examples/Talker.java:18: error: cannot find symbol
import org.ros2.rcljava.Node;
                       ^
```
So, it seems that including the `jcljava` packages is not going well. I don't have much experience with Gradle, so I don't know how to solve this yet. Where can I update the settings for `rcljava_examples` so it can find the `rcljava` packages?

wilcobonestroo, 2020-03-29:
@esteve I also see the example_interfaces.srv in your ROSCon presentation. However, I don't understand where it is defined. Can you help us with the questions above?

wilcobonestroo, 2020-03-29:
@nielstiben If you delete the "addTwoInts" java files, then the other examples build. The `subscriber_lambda` and `publisher_lambda` work. With a [little change](https://github.com/ros2-java/ros2_java_examples/pull/7) also the `subscriber_node` and the `publisher_node` work.

esteve, 2020-03-30:
@wilcobonestroo ros2_alpha8 branch is an old branch that predates the first official version of ROS 2, so it's not meant to be used, I've just removed it. The latest ROS 2 distro supported on master is Bouncy, but @jacobperron is working on adding support for Dashing and Eloquent in https://github.com/ros2-java/ros2_java/tree/dashing

wilcobonestroo, 2020-03-30:
Thank you for your answers. Can you also give a hint about where `example_interfaces.srv` is defined? Once that is gone, the dashing version will build.

jacobperron, 2020-04-07:
See https://github.com/ros2-java/ros2_java/pull/101 for adding `example_interfaces` to the repos file.

jacobperron, 2021-11-13:
Dashing is EOL. Please try out the `main` branch with ROS Galactic.

closed 2021-11-13 by jacobperron