ros2-java/ros2_java #105: Unsubscribe from topic safely

jacobperron, 2020-05-01:
Currently, there is no safe way to unsubscribe from a topic. If the user calls `dispose()` on a `Subscription`, then it will destroy the native subscription (effectively unsubscribing), but the `Node` is not aware of the subscriptions disposal. E.g. [getSubscriptions](https://github.com/ros2-java/ros2_java/blob/1190aeb909e051e79b2b43325f924f42e5c96160/rcljava/src/main/java/org/ros2/rcljava/node/NodeImpl.java#L227-L229) will return disposed subscriptions that may be [accessed by executors](https://github.com/ros2-java/ros2_java/blob/1190aeb909e051e79b2b43325f924f42e5c96160/rcljava/src/main/java/org/ros2/rcljava/executors/BaseExecutor.java#L170) leading an exception:

    org.ros2.rcljava.exceptions.RCLException: Failed to add subscription to wait set

I think we need a way to safely remove `Subscription` (and `Publisher`) objects from a node. Whether that is through the existing `Disposable` interface or API added to the node (e.g. `removeSubscription`).

## Implementation considerations

I think having the node detect when a subscription has been disposed (perhaps using [WeakReference](https://docs.oracle.com/javase/7/docs/api/java/lang/ref/WeakReference.html)) would be nice as I don't think it would require additional API.

jacobperron, 2020-07-29:
This was fixed in #110

closed 2020-07-29 by jacobperron

esteve, 2020-07-29:
@jacobperron thanks for catching that!

commit b16be6c8 (2020-06-24): Add methods to remove entities from Node

When an entity is disposed, make sure to also remove it from the Node.
This resolves an issue where invalid entities may be used by other classes or users.
For example, a disposed Subscription being accessed by the executor.

Fixes #105
PR #110 (2020-06-30): Add methods to remove entities from Node
When an entity is disposed, make sure to also remove it from the Node.
This resolves an issue where invalid entities may be used by other classes or users.
For example, a disposed Subscription being accessed by the executor.

Fixes #105
commit cf2ef070 (2020-06-30): Add methods to remove entities from Node (#110)

* Add methods to remove entities from Node

When an entity is disposed, make sure to also remove it from the Node.
This resolves an issue where invalid entities may be used by other classes or users.
For example, a disposed Subscription being accessed by the executor.

Fixes #105

* Add tests for disposing publishers, services, and clients

* Rename test
commit d270e675 (2021-05-17): Add methods to remove entities from Node (#110)

* Add methods to remove entities from Node

When an entity is disposed, make sure to also remove it from the Node.
This resolves an issue where invalid entities may be used by other classes or users.
For example, a disposed Subscription being accessed by the executor.

Fixes #105

* Add tests for disposing publishers, services, and clients

* Rename test