ros2-java/ros2_java #112: About ros2_java foxy version

jayhou, 2020-06-12:
Hi, I found there is a branch named "jacob/foxy". Is this branch available to ros2 "foxy" branch? I've tried to use this ros2_java branch build rcljava with ros2 foxy branch, but It's seems not work.

jayhou, 2020-06-12:
I  add these repo：
  ros2_java/ros2_java:
    type: git
    url: https://github.com/ros2-java/ros2_java.git
    version: jacob/foxy
  ament_java/ament_java:
    type: git
    url: https://github.com/esteve/ament_java.git
    version: master

and cause the colcon build error:

if there is cycle dependence in packages.xml?

ERROR:colcon:colcon build: Unable to order packages topologically:

action_msgs: ['builtin_interfaces', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_generators', 'rosidl_default_runtime', 'rosidl_generator_java', 'unique_identifier_msgs']
action_tutorials_cpp: ['action_msgs', 'action_tutorials_interfaces', 'builtin_interfaces', 'composition_interfaces', 'libstatistics_collector', 'rcl', 'rcl_action', 'rcl_interfaces', 'rclcpp', 'rclcpp_action', 'rclcpp_components', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosgraph_msgs', 'rosidl_default_runtime', 'rosidl_generator_java', 'statistics_msgs', 'std_msgs', 'unique_identifier_msgs']
action_tutorials_interfaces: ['action_msgs', 'builtin_interfaces', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_generators', 'rosidl_default_runtime', 'rosidl_generator_java', 'unique_identifier_msgs']
action_tutorials_py: ['action_msgs', 'action_tutorials_interfaces', 'builtin_interfaces', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_runtime', 'rosidl_generator_java', 'unique_identifier_msgs']
actionlib_msgs: ['builtin_interfaces', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_generators', 'rosidl_default_runtime', 'rosidl_generator_java', 'std_msgs']
builtin_interfaces: ['rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_generators', 'rosidl_default_runtime', 'rosidl_generator_java']
common_interfaces: ['actionlib_msgs', 'builtin_interfaces', 'diagnostic_msgs', 'geometry_msgs', 'nav_msgs', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_runtime', 'rosidl_generator_java', 'sensor_msgs', 'shape_msgs', 'std_msgs', 'std_srvs', 'stereo_msgs', 'trajectory_msgs', 'visualization_msgs']
composition: ['action_msgs', 'builtin_interfaces', 'composition_interfaces', 'example_interfaces', 'launch_ros', 'launch_testing_ros', 'libstatistics_collector', 'lifecycle_msgs', 'rcl', 'rcl_interfaces', 'rclcpp', 'rclcpp_components', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosgraph_msgs', 'rosidl_default_runtime', 'rosidl_generator_java', 'statistics_msgs', 'std_msgs', 'unique_identifier_msgs']
composition_interfaces: ['builtin_interfaces', 'rcl_interfaces', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_generators', 'rosidl_default_runtime', 'rosidl_generator_java']
demo_nodes_cpp: ['action_msgs', 'builtin_interfaces', 'composition_interfaces', 'example_interfaces', 'launch_ros', 'launch_testing_ros', 'libstatistics_collector', 'lifecycle_msgs', 'rcl', 'rcl_interfaces', 'rclcpp', 'rclcpp_components', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosgraph_msgs', 'rosidl_default_runtime', 'rosidl_generator_java', 'statistics_msgs', 'std_msgs', 'unique_identifier_msgs']
demo_nodes_cpp_native: ['builtin_interfaces', 'composition_interfaces', 'launch_ros', 'launch_testing_ros', 'libstatistics_collector', 'lifecycle_msgs', 'rcl', 'rcl_interfaces', 'rclcpp', 'rclcpp_components', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosgraph_msgs', 'rosidl_default_runtime', 'rosidl_generator_java', 'statistics_msgs', 'std_msgs']
demo_nodes_py: ['action_msgs', 'builtin_interfaces', 'example_interfaces', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_runtime', 'rosidl_generator_java', 'std_msgs', 'unique_identifier_msgs']
diagnostic_msgs: ['builtin_interfaces', 'geometry_msgs', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_generators', 'rosidl_default_runtime', 'rosidl_generator_java', 'std_msgs']
dummy_map_server: ['builtin_interfaces', 'geometry_msgs', 'libstatistics_collector', 'nav_msgs', 'rcl', 'rcl_interfaces', 'rclcpp', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosgraph_msgs', 'rosidl_default_runtime', 'rosidl_generator_java', 'statistics_msgs', 'std_msgs']
[... cut ...]
visualization_msgs: ['builtin_interfaces', 'geometry_msgs', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_generators', 'rosidl_default_runtime', 'rosidl_generator_java', 'std_msgs']
xcu_msgs: ['builtin_interfaces', 'rmw_cyclonedds_cpp', 'rmw_dds_common', 'rmw_fastrtps_cpp', 'rmw_fastrtps_dynamic_cpp', 'rmw_fastrtps_shared_cpp', 'rmw_implementation', 'rosidl_default_generators', 'rosidl_default_runtime', 'rosidl_generator_java']

jacobperron, 2021-11-13:
I've removed that development branch. Please use the `main` branch with ROS Galactic now. You might also have luck with `main` + Foxy, but you may have to try an earlier commit.

closed 2021-11-13 by jacobperron