ros2-java/ros2_java #113: Build error on Windows 10 with Visual Studio 2019

mefoster, 2020-06-15:
I have installed ROS 2 on my Windows 10 machine and have downloaded the latest ros2-java (checked out everything with vcs)

I am trying to build as follows:
C:\dev\ament_ws>python src/ament/ament_tools/scripts/ament.py build

I get the following error:
Process package 'ament_cmake_core' with context:
--------------------------------------------------------------------------------
 source_space => C:\dev\ament_ws\src\ament\ament_cmake\ament_cmake_core
  build_space => C:\dev\ament_ws\build\ament_cmake_core
install_space => C:\dev\ament_ws\install
   make_flags => -j8, -l8
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'ament_cmake_core'
ament build: Unknown / unsupported VS version: 16.0

Any suggestions on where to fix this?

mefoster, 2020-06-16:
Have been trying to get this to work a bit more but I'm not an expert on Visual Studio or CMake so I'm heading down a rabbit hole. Has anyone actually ever built this successfully with Visual Studio 2019?

esteve, 2020-06-16:
@mefoster thanks for trying out `ros2-java`. We're adding a CI workflow for Windows in https://github.com/ros2-java/ros2_java/pull/109 to make sure that `ros2-java` supports Windows. However, we're now targeting ROS 2 Dashing and Foxy in separately in non-master branches, and will soon update the `master` branch of `ros2-java`, which right now only supports ROS 2 Bouncy. What version of ROS 2 did you install?

mefoster, 2020-06-16:
Foxy ... ah yes, now I see what might have happened. I'll look for the Foxy branch and try again.

mefoster, 2020-06-16:
Right, I see that you opened a new request about describing how to build with colcon instead of ament. I was just in the process of discovering that ament is obsolete ... I'll stay tuned and try again when there are new instructions

jacobperron, 2021-11-13:
I'm going to close this since we have this ticket to track getting CI turning over on Windows https://github.com/ros2-java/ros2_java/issues/102

closed 2021-11-13 by jacobperron