ros2-java/ros2_java #126: how to replace ament with colcon command?

liushichao1, 2020-08-04:
i want compile android project, but the instruction use the comand like this:

`ament build --isolated --skip-packages test_msgs \
  --cmake-args \
  -DPYTHON_EXECUTABLE=${PYTHON3_EXEC} \
  -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake \
  -DANDROID_FUNCTION_LEVEL_LINKING=OFF \
  -DANDROID_NATIVE_API_LEVEL=${ANDROID_NATIVE_API_LEVEL} \
  -DANDROID_TOOLCHAIN_NAME=${ANDROID_TOOLCHAIN_NAME} \
  -DANDROID_STL=gnustl_shared \
  -DANDROID_ABI=${ANDROID_ABI} \
  -DANDROID_NDK=${ANDROID_NDK} \
  -DTHIRDPARTY=ON \
  -DCOMPILE_EXAMPLES=OFF \
  -DCMAKE_FIND_ROOT_PATH="$AMENT_WORKSPACE/install_isolated;$ROS2_ANDROID_WORKSPACE/install_isolated" \
  -- \
  --parallel \
  --ament-gradle-args \
  -Pament.android_stl=gnustl_shared -Pament.android_abi=$ANDROID_ABI -Pament.android_ndk=$ANDROID_NDK --`

when i run `colcon build -h`, there is no `  --ament-gradle-args` params, so how to do it with colcon command, any help with thanks.

jacobperron, 2020-08-05:
There's some work being done to update instructions for ROS 2 Dashing and newer in https://github.com/ros2-java/ros2_java/pull/125

liu-shichao, 2020-08-16:
> There's some work being done to update instructions for ROS 2 Dashing and newer in #125

ok, thanks very much , good luck for you.

closed 2020-08-19 by liu-shichao