ros2-java/ros2_java #131: Could not fund 'gradle' executable

aitazhixin, 2020-09-16:
run script
`  ament build --isolated --skip-packages test_msgs \
  --cmake-args \
  -DPYTHON_EXECUTABLE=${PYTHON3_EXEC} \
   -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake \
   -DANDROID_FUNCTION_LEVEL_LINKING=OFF \
    -DANDROID_NATIVE_API_LEVEL=${ANDROID_NATIVE_API_LEVEL} \
    -DANDROID_TOOLCHAIN_NAME=${ANDROID_TOOLCHAIN_NAME} \
    -DANDROID_STL=c++_shared \
    -DANDROID_ABI=${ANDROID_ABI} \
    -DANDROID_NDK=${ANDROID_NDK} \
    -DTHIRDPARTY=ON \
    -DCOMPILE_EXAMPLES=OFF \
    -DCMAKE_FIND_ROOT_PATH="$AMENT_WORKSPACE/install_isolated;$ROS2_ANDROID_WORKSPACE/install_isol    ated" \
    -- \
    --ament-gradle-args \
    -Pament.android_stl="c++_shared" -Pament.android_abi=$ANDROID_ABI -Pament.android_ndk=$ANDROID    _NDK --`

Process package 'rclandroid' with context:
--------------------------------------------------------------------------------
 source_space => /home/ak/Downloads/ros2build/ros2_android_ws/src/ros2_java/ros2_android/rclandroid
  build_space => /home/ak/Downloads/ros2build/ros2_android_ws/build_isolated/rclandroid
install_space => /home/ak/Downloads/ros2build/ros2_android_ws/install_isolated/rclandroid
   make_flags => -j16, -l16
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'rclandroid'
ament build: Could not find 'gradle' executable

esteve, 2020-09-16:
@aitazhixin thanks for trying ros2-java out. Do you have Gradle installed? It is necessary for building rclandroid and the examples.

aitazhixin, 2020-09-17:
now, using downloaded gradle, it can be built.

closed 2020-09-17 by aitazhixin