ros2-java/ros2_java #132: JNI ERROR when receiving sensor_msgs.msg.Image in Android.

aitazhixin, 2020-09-17:
E/zygote64: JNI ERROR (app bug): expected jbyte but got value of 206 as argument 1 to void java.lang.Byte.<init>(byte)
A/zygote64: java_vm_ext.cc:534] JNI DETECTED ERROR IN APPLICATION: bad arguments passed to void java.lang.Byte.<init>(byte) (see above for details)
A/zygote64: java_vm_ext.cc:534]     from org.ros2.rcljava.interfaces.MessageDefinition org.ros2.rcljava.executors.BaseExecutor.nativeTake(long, java.lang.Class)
A/zygote64: java_vm_ext.cc:534] "ForkJoinPool.commonPool-worker-2" daemon prio=5 tid=13 Runnable
A/zygote64: java_vm_ext.cc:534]   | group="main" sCount=0 dsCount=0 flags=0 obj=0x1374c328 self=0x72ce0bf400
A/zygote64: java_vm_ext.cc:534]   | sysTid=31016 nice=0 cgrp=default sched=0/0 handle=0x72b51004f0
A/zygote64: java_vm_ext.cc:534]   | state=R schedstat=( 2581074205 10282787 194 ) utm=250 stm=7 core=5 HZ=100
A/zygote64: java_vm_ext.cc:534]   | stack=0x72b4ffe000-0x72b5000000 stackSize=1037KB
A/zygote64: java_vm_ext.cc:534]   | held mutexes= "mutator lock"(shared held)
A/zygote64: java_vm_ext.cc:534]   native: #00 pc 00000000003ccb5c  /system/lib64/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, int, BacktraceMap*, char const*, art::ArtMethod*, void*)+208)
A/zygote64: java_vm_ext.cc:534]   native: #01 pc 000000000049cdf4  /system/lib64/libart.so (art::Thread::DumpStack(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, bool, BacktraceMap*, bool) const+348)
A/zygote64: java_vm_ext.cc:534]   native: #02 pc 00000000002fe388  /system/lib64/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+1048)
A/zygote64: java_vm_ext.cc:534]   native: #03 pc 00000000002fe878  /system/lib64/libart.so (art::JavaVMExt::JniAbortF(char const*, char const*, ...)+180)
A/zygote64: java_vm_ext.cc:534]   native: #04 pc 000000000046f730  /system/lib64/libart.so (art::InvokeWithArgArray(art::ScopedObjectAccessAlreadyRunnable const&, art::ArtMethod*, art::ArgArray*, art::JValue*, char const*)+2508)
A/zygote64: java_vm_ext.cc:534]   native: #05 pc 000000000046e9ec  /system/lib64/libart.so (art::InvokeWithVarArgs(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jmethodID*, std::__va_list)+412)
A/zygote64: java_vm_ext.cc:534]   native: #06 pc 0000000000360f38  /system/lib64/libart.so (art::JNI::CallNonvirtualVoidMethodV(_JNIEnv*, _jobject*, _jclass*, _jmethodID*, std::__va_list)+616)
A/zygote64: java_vm_ext.cc:534]   native: #07 pc 0000000000344e70  /system/lib64/libart.so (art::JNI::NewObjectV(_JNIEnv*, _jclass*, _jmethodID*, std::__va_list)+752)
A/zygote64: java_vm_ext.cc:534]   native: #08 pc 00000000000fc39c  /system/lib64/libart.so (art::CheckJNI::NewObjectV(_JNIEnv*, _jclass*, _jmethodID*, std::__va_list)+744)
A/zygote64: java_vm_ext.cc:534]   native: #09 pc 0000000000007420  /data/app/com.pudutech.mirsdk.hardware.mock-yBF4X0jVOAjKUaYKAEGGow==/lib/arm64/libsensor_msgs_msg__image__jni__rosidl_typesupport_c.so (_JNIEnv::NewObject(_jclass*, _jmethodID*, ...)+208)
A/zygote64: java_vm_ext.cc:534]   native: #10 pc 0000000000007228  /data/app/com.pudutech.mirsdk.hardware.mock-yBF4X0jVOAjKUaYKAEGGow==/lib/arm64/libsensor_msgs_msg__image__jni__rosidl_typesupport_c.so (sensor_msgs_msg_Image__convert_to_java(sensor_msgs__msg__Image*, _jobject*)+920)
A/zygote64: java_vm_ext.cc:534]   native: #11 pc 0000000000004408  /data/app/com.pudutech.mirsdk.hardware.mock-yBF4X0jVOAjKUaYKAEGGow==/lib/arm64/liborg_ros2_rcljava_executors__base_executor__jni.so (Java_org_ros2_rcljava_executors_BaseExecutor_nativeTake+588)
A/zygote64: java_vm_ext.cc:534]   native: #12 pc 0000000000001394  /data/app/com.pudutech.mirsdk.hardware.mock-yBF4X0jVOAjKUaYKAEGGow==/oat/arm64/base.odex (Java_org_ros2_rcljava_executors_BaseExecutor_nativeTake__JLjava_lang_Class_2+180)
A/zygote64: java_vm_ext.cc:534]   at org.ros2.rcljava.executors.BaseExecutor.nativeTake(Native method)
A/zygote64: java_vm_ext.cc:534]   at org.ros2.rcljava.executors.BaseExecutor.executeAnyExecutable(BaseExecutor.java:83)
A/zygote64: java_vm_ext.cc:534]   at org.ros2.rcljava.executors.BaseExecutor.spinSome(BaseExecutor.java:338)
A/zygote64: java_vm_ext.cc:534]   at org.ros2.rcljava.executors.SingleThreadedExecutor.spinSome(SingleThreadedExecutor.java:42)
A/zygote64: java_vm_ext.cc:534]   at com.pudutech.mirsdk.ros2.ROS2.spinSome(ROS2.kt:58)

    *******	STACKDUMP *******
    	stack dump [1]  
    	stack dump [2]  __kernel_rt_sigreturn
    	stack dump [3]  abort
    	stack dump [4]  _ZN3art7Runtime13PreZygoteForkEv
    	stack dump [5]  _ZN7android4base10LogMessageD1Ev
    	stack dump [6]  _ZN3art9JavaVMExt8JniAbortEPKcS2_
    	stack dump [7]  _ZN3art9JavaVMExt9JniAbortFEPKcS2_z
    	stack dump [8]  
    	stack dump [9]  _ZN3art17InvokeWithVarArgsERKNS_33ScopedObjectAccessAlreadyRunnableEP8_jobjectP10_jmethodIDSt9__va_list
    	stack dump [10]  
    	stack dump [11]  
    	stack dump [12]  
    	stack dump [13]  _ZN7_JNIEnv9NewObjectEP7_jclassP10_jmethodIDz
    	stack dump [14]  _Z38sensor_msgs_msg_Image__convert_to_javaP23sensor_msgs__msg__ImageP8_jobject
    	stack dump [15]  Java_org_ros2_rcljava_executors_BaseExecutor_nativeTake
    	stack dump [16]  oatexec
    
    Exiting after fatal event  (FATAL_SIGNAL). Fatal type:  SIGABRT
    Log content flushed sucessfully to sink

aitazhixin, 2020-09-17:
Image data value range is 0-255, it has over byte data range.

aitazhixin, 2020-09-17:
Code from Image.ep.****.cpp

jobject sensor_msgs_msg_Image__convert_to_java(sensor_msgs__msg__Image * _ros_message, jobject _jmessage_obj)
{
........
  for (size_t i = 0; i < _ros_message->data.size; ++i) {
    auto _ros_data_element = _ros_message->data.data[i];
    jobject _jlist_data_element = env->NewObject(
      _jjava__lang__Byte_class_global, _jjava__lang__Byte_constructor_global, _ros_data_element);
    // TODO(esteve): replace ArrayList with a jobjectArray to initialize the array beforehand
.......
}

Here _ros_data_element type is uint8_t, _jjava__lang__Byte_class_global type is java.lang.Byte. May this constructor _jjava__lang__Byte_constructor_global be not used to create Byte object with initilization value _ros_data_element.

jacobperron, 2021-11-13:
Please try with the latest changes on `main` (ROS Galactic). If the issue persists, comment here (or open a new issue) with precise steps to reproduce. Thanks.

closed 2021-11-13 by jacobperron