ros2-java/ros2_java #136: Which version should run?

davidgith, 2020-09-30:
I am a bit confused in which system with which branch the code should run. I tried it now under different Ubuntu versions with different branches and unfortunately I had no luck.
If I see it right, according to readme, it only runs under Ubuntu 16?
Which branch should run? Then I could search for errors and send error messages if necessary. Of course it would be nice if it would run under Ubuntu 20. But I also have Ubuntu 16 and 18 as VM.
Sorry if I made a stupid mistake, I'm still a beginner...
Thanks a lot!

AlexKov13, 2020-10-06:
I ran master on 18.04 with ros-eloquent.

davidgith, 2020-10-06:
Thanks for the advice.

Unfortunately I get the following error in the step "ament build --symlink-install --isolated":

By not providing "Findunique_identifier_msgs.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "unique_identifier_msgs", but CMake did not find one.

  Could not find a package configuration file provided by
  "unique_identifier_msgs" with any of the following names:

    unique_identifier_msgsConfig.cmake
    unique_identifier_msgs-config.cmake

  Add the installation prefix of "unique_identifier_msgs" to
  CMAKE_PREFIX_PATH or set "unique_identifier_msgs_DIR" to a directory
  containing one of the above files.  If "unique_identifier_msgs" provides a
  separate development package or SDK, be sure it has been installed.

Do you have an idea what I am doing wrong?

Thanks!

davidgith, 2020-10-06:
Ah, I think one mistake I made was not sourcing the workspace. Maybe a hint for the beginners would be helpful;)
But since then I get stuck building the action_msgs, where this error appears several times.
error: redefinition of 'struct rosidl_generator_traits::has_bounded_size<T>'
 struct has_bounded_size : std::false_type {};

jacobperron, 2020-10-08:
AFAIK, the `master` branch is only compatible with ROS Bouncy.

We're in the process of updating for newer versions. I recommend trying the `dashing` branch on Ubuntu 18.04. Instructions are being updated in https://github.com/ros2-java/ros2_java/pull/119 and https://github.com/ros2-java/ros2_java/pull/125.

davidgith, 2020-10-11:
Thanks for the hint for the new readme, now it looks much better. The build process runs through except for the warning "ament_build_type_gradle doesn't explicitly install a marker in the package index (colcon-ros currently does it implicitly but that fallback will be removed in the future)" and the hint that colcon_core.task.checkall should be replaced by colcon_core.task.run, I tried that too. But I don't get the package rcljava_examples when running it, the other packages look good. But actually the build process runs without any error. Do you have an idea?
Thanks a lot!

zjyspp, 2020-11-04:
I have a similar problem. In the dashing branch, Android is compiled according to the new build method. The build process runs without any error， but it can not be generated gradle.properties file, cannot be opened with Android studio。Can  you help me？
Thanks a lot！

jacobperron, 2021-11-13:
Maintainers are now focused on keeping the `main` branch up-to-date to support the latest ROS distro (current Galactic). So please refer to the `main` branch from now on.

There will be no active support for older ROS distros due to resource constraints.

closed 2021-11-13 by jacobperron