ros2-java/ros2_java #138: ROS Actions

jacobperron, 2020-10-07:
This ticket is to track adding support for ROS actions to ros2_java.

Some early work related to code generation for actions was done in #66, #76, and #108.
There has also been some work done towards the action client (discussed in https://github.com/ros2-java/ros2_java/issues/82#issuecomment-635995720).

I think we can break this task into two parts:

- [ ] Action server implementation
- [ ] Action client implementation

Additional resources:

- [ROS 2 action design](http://design.ros2.org/articles/actions.html)
- [C++ implementation](https://github.com/ros2/rclcpp/tree/master/rclcpp_action)
- [Python implementation](https://github.com/ros2/rclpy/tree/master/rclpy/rclpy/action)

jacobperron, 2020-10-07:
@nielstiben Sorry for the delay, but I'm ready to tackle action support now. I'll start by implementing the action server. Do you mind opening a PR to the `dashing` branch containing the work you've done so far for the action client? I can help troubleshoot any issues you have, or take over where you left off if you don't have the bandwidth to work on it.

jacobperron, 2020-10-20:
I've opened a couple PRs that are prerequisites for the action server implementation I have: 
  - Access Node clock (https://github.com/ros2-java/ros2_java/pull/139)
  - Fixes to action code generation (https://github.com/ros2-java/ros2_java/pull/140)

jacobperron, 2021-11-13:
An implementation for the Action Server was added to our fork in https://github.com/osrf/ros2_java/pull/65. We will add it upstream to the `main` branch soon.

marlow-fawn, 2022-08-15:
Any updates on this? Would love actions in the main branch.