ros2-java/ros2_java #143: unable to build Android

zjyspp, 2020-11-05:
Hello
 
A few days ago, I started to try Android for my own application. I built it according to the  script  on dashing branch。but it always failed.  
[100%] Built target rosidl_typesupport_introspection_cpp
[100%] Built target rosidl_typesupport_c
+++ Installing 'rosidl_typesupport_introspection_cpp'
+++ Installing 'rosidl_typesupport_c'
==> '. /home/rosjava/ros2_java/ros2_android_ws/build_isolated/rosidl_typesupport_introspection_cpp/cmake__install.sh && /usr/bin/make install' in '/home/rosjava/ros2_java/ros2_android_ws/build_isolated/rosidl_typesupport_introspection_cpp'
==> '. /home/rosjava/ros2_java/ros2_android_ws/build_isolated/rosidl_typesupport_c/cmake__install.sh && /usr/bin/make install' in '/home/rosjava/ros2_java/ros2_android_ws/build_isolated/rosidl_typesupport_c'
[100%] Built target rosidl_typesupport_introspection_cpp
[100%] Built target rosidl_typesupport_c
CMake Error at CMakeLists.txt:38 (find_package):
  By not providing "Findrcpputils.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "rcpputils", but CMake did not find one.

  Could not find a package configuration file provided by "rcpputils" with
  any of the following names:

    rcpputilsConfig.cmake
    rcpputils-config.cmake

  Add the installation prefix of "rcpputils" to CMAKE_PREFIX_PATH or set
  "rcpputils_DIR" to a directory containing one of the above files.  If
  "rcpputils" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/rosjava/ros2_java/ros2_android_ws/build_isolated/rmw_fastrtps_shared_cpp/CMakeFiles/CMakeOutput.log".

Can somebody help ？
Thanks a lot！！

ubuntu: 18.04 Bionic
NDK: 21b
SDK: 29
gradle: 6.7
jdk: 8

fiaren, 2020-11-09:
@zjyspp  could do it by cloning https://github.com/ros2/rcpputils/tree/dashing to src

jacobperron, 2021-11-13:
Please try the latest changes on `main` with ROS Galactic and let me know if there are still issues.