ros2-java/ros2_java #144: Dasing Android build failed

xingjl6280, 2020-11-14:
Hi team,

I've tried several days but still no luck. Btw, the java version is working.

Here's my latest attempt tried today 
Steps: 
1. Set up a new ubuntu-18.04.3-desktop-amd64 instance by VirtualBox

2. Install ROS2(ros-dashing-ros-base) via Debian Packages 
https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/

3. Install development tools and ROS tools followed by corresponding section of the page.
https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/

4. Then follow the instruction of jacob/update_readme branch to setup and build for Anroid.
Branch: https://github.com/ros2-java/ros2_java/tree/jacob/update_readme
AndroidNDK: android-ndk-r21d-linux-x86_64
AndroidSDK: android-21

Error:
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

The full build log is here:
[log.zip](https://github.com/ros2-java/ros2_java/files/5540280/log.zip)

Please kindly advise.
Many thanks

Regards

xingjl6280, 2020-11-17:
Hi team, 

Please kindly help. 
We are still struggling. 

I saw some previous disucssion but not sure if it's latest. For now, there's no branch for dashing could work with Android?

thanks

jacobperron, 2020-11-19:
We haven't sorted out all the issues related to Android support for Dashing (hence the development branches), but CI was passing a while ago with these instructions: https://github.com/ros2-java/ros2_java/blob/a3fa46df61aeee47d8bad5f0c7397d3f0f5da389/.github/workflows/build_and_test.yml#L23

They *should* be more-or-less the same as the README you referenced, but it's possible there is a bug in the documentation.

xingjl6280, 2020-11-21:
thanks, Jacob, it works now.

closed 2020-11-21 by xingjl6280