ros2-java/ros2_java #145: Path change for uuid__struct.h and uuid__functions.h?

stuart-k-rogers, 2020-11-17:
I've been trying to build for desktop per the README and ran into a error:

> /home/kevin/Projects/ros2_java_ws/build_isolated/action_msgs/rosidl_generator_c/action_msgs/msg/goal_info__struct.h:18:10: fatal error: unique_identifier_msgs/msg/uuid__struct.h: No such file or directory
>    18 | #include "unique_identifier_msgs/msg/uuid__struct.h"

Similar message for uuid_functions.h.

I found the actual file in the foxy installation as:

> /opt/ros/foxy/include/unique_identifier_msgs/msg/**detail**/uuid__struct.h
> /opt/ros/foxy/include/unique_identifier_msgs/msg/**detail**/uuid__functions.h

It appears someone has moved those files recently.

jacobperron, 2020-11-19:
Which version of the README are you looking at? The version on the `master` branch is only compatible with ROS Bouncy.

stuart-k-rogers, 2020-11-20:
> Which version of the README are you looking at? The version on the `master` branch is only compatible with ROS Bouncy.

Thanks for getting back to me@

I have just the version that comes up at https://github.com/ros2-java/ros2_java.  I see now that it doesn't explicitly call out the ROS version, but of course it would make sense that if it hasn't really been updated since 2018, it's probably only compatible with the ROS version from back then.  

Is there any intent to maintain it for more recent versions?  At a different time in my life I might have taken a cut at it myself, but I'm really more interested in using ROS than in developing framework for it.  I'm trying to build a java ROS node with embedded neo4j database and this looked like a promising path to solve latency problems in the native neo4j API.

stuart-k-rogers, 2020-11-20:
And I see you have a recently active branch that looks like it's touching some of the areas for which I ran into problems with foxy.

jacobperron, 2020-11-20:
I'm interested in supporting more recent versions of ROS 2. I'm working on landing changes for Dashing in https://github.com/ros2-java/ros2_java/pull/117 and then I'd like to follow-up with changes for Foxy and the upcoming Galactic release. Work towards this effort can be found on our fork: https://github.com/osrf/ros2_java

The documentation on that fork is not updated though; I'd like to merge everything back here ultimately.

jacobperron, 2021-11-13:
I think this issue should be resolved on `main` (supporting ROS Galactic). I'm going to close it, but please comment/reopen if the issue persists.

closed 2021-11-13 by jacobperron