ros2-java/ros2_java #146: Dependency reference problem for rosidl_typesupport_interface?

stuart-k-rogers, 2020-11-17:
In attempting to build for Desktop I found build errors associated with rosidl_typesupport_interface.  I noticed that the CMake files all refer to it with like this: rosidl_typesupport_interface::rosidl_typesupport_interface.  The double-colon repeated text seems unique to rosidl_typesupport_interface, so I'm wondering if whatever generated it as a dependency has introduced an error.

> build_isolated/action_msgs/CMakeCache.txt:srvCancelGoalc_LIB_DEPENDS:STATIC=general;action_msgs__rosidl_typesupport_c;general;-Wl,--no-undefined;general;action_msgs__rosidl_typesupport_c;general;/home/kevin/Projects/ros2_java_ws/install_isolated/rosidl_generator_c/lib/librosidl_generator_c.so;general;/home/kevin/Projects/ros2_java_ws/install_isolated/rcljava_common/lib/librcljava_common.so;general; ... ;general;**rosidl_typesupport_interface::rosidl_typesupport_interface**;general;/home/kevin/Projects/ros2_java_ws/install_isolated/rosidl_typesupport_c/lib/librosidl_typesupport_c.so; ...

jacobperron, 2021-11-13:
We've made many changes since this issue was opened. I encourage you to try the `main` branch with ROS Galactic; I hope the same issue does not exist.

jacobperron, 2021-11-13:
I'm going to close the issue, please comment if this issue persists with the latest development on `main`.

closed 2021-11-13 by jacobperron

stuart-k-rogers, 2021-11-15:
I’ll have a look.

.—K

From: Jacob Perron
Sent: Friday, November 12, 2021 5:56 PM
To: ros2-java/ros2_java
Cc: stuart-k-rogers; Author
Subject: Re: [ros2-java/ros2_java] Dependency reference problem for rosidl_typesupport_interface? (#146)

I'm going to close the issue, please comment if this issue persists with the latest development on main.

stuart-k-rogers, 2021-11-18:
It builds without errors (yay!), but I can't run the rcljava_examples.  I see the source code for rcljava_examples in the package, but after sourcing all of the setup_local files and attempting to run the examples, ros2 says it can't find the rcljava_examples.  They don't appear in the colcon output for "Starting" and "Finishing".

jacobperron, 2021-11-18:
@stuart-k-rogers Make sure you have the colcon-gradle dependencies installed before building:

```
 python3 -m pip install -U git+https://github.com/colcon/colcon-gradle
 python3 -m pip install --no-deps -U git+https://github.com/colcon/colcon-ros-gradle
```

stuart-k-rogers, 2021-11-19:
Using "ros2 run..." still gives the same result, but invoke with java works perfectly.  Thanks!

jacobperron, 2021-11-19:
> Using "ros2 run..." still gives the same result, but invoking with the java command works perfectly. Thanks!

Great, thanks for checking!
I think running with "ros2 run" depends on https://github.com/colcon/colcon-ros-gradle/pull/7.

Spleenix, 2023-06-01:
> Using "ros2 run..." still gives the same result, but invoking with the java command works perfectly. Thanks!

I have the same problem. What exactly did you do to make it work for you?