ros2-java/ros2_java #149: Performance issue for using List in JNI

pluris, 2020-11-20:
I was using ros2 java and noticed that the performance was very slow in List.

When communicating with ros2 native, there is a serious performance difference in ros2java side.
So, I compared the performance by creating a Android JNI project.

- Array TIME : 17.532115(ms)
- List TIME : 1419.957922(ms)

**How about using Array instead of List?**
The usability is not good, but the difference in performance is significant, so I suggest.

Need some more modifications, but to me it also has an improved code.

jacobperron, 2021-11-13:
We have a fix https://github.com/ros2-java/ros2_java/pull/165 that will land on `main` as we start porting changes from our fork.

pluris, 2022-05-19:
@jacobperron  I'm sorry. I saw too late because didn't follow up this. I think that's a great job about #165 .

closed 2023-08-10 by pluris