ros2-java/ros2_java #152: How to publish Android APP with ros2 client?

xingjl6280, 2020-11-21:
Hi team,

There's no guild for Android exmples.
Please kindly advise what is the process to publish an Android App with ros2 client? 
Just to open the android exmpale project by Android Studio, build APK file, then install on Android device? 
Is there anything need to be done on Android device?

Another question is, how to intgrate ros2 client to an existing Android project, just modify the build.gradle?

thank you

xingjl6280, 2020-11-22:
I tried to copy the buildscript section of build.gradle from example project to my newly created Android project. 
I also executed ~/ros2_java_ws/install/local_setup.sh in the same Android studio terminal.

The build apk process failed with below error:

Execution failed for task ':app:checkAmentProperties'.
> Missing Ament Gradle property: ament.build_space

vkuehn, 2020-12-02:
my two scent...look at the introduction
https://github.com/esteve/ros2_java_examples, examples for the Java Runtime Environment

closed 2021-01-06 by xingjl6280