ros2-java/ros2_java #153: setting ROS_DOMAIN_ID for Android

AlexKov13, 2020-12-04:
Is there any way to set domain ID for android, to isolated different phones & robots?

Luriwei, 2020-12-19:
@AlexKovalchukSomatic I meeted the same question, have you resolved that?
I haved built the ros2_java in dashing branch and create an android application through android studio, but the android publisher can not be found by the computer side(ubuntu 18.0.4).
@esteve need your help, thank you very much.

AlexKovalchukSomatic, 2020-12-21:
@Luriwei 
Yes, I modified code to set DomainID while creating node (You should change signature). 
You should change "nativeCreateNodeHandle" function:

> default_options.domain_id = domain;

Luriwei, 2020-12-22:
@AlexKovalchukSomatic 
I changed the default_options.domain_id into 42, and i also changed the RCL_NODE_OPTIONS_DEFAULT_DOMAIN_ID into 42,
after that my android app node still can not be found by my unbuntu(ROS_DOMAIN_ID is 42 and they use the same wifi).

Do i lost something? What do you mean 'change signature' ? Thanks.

AlexKovalchukSomatic, 2020-12-22:
@Luriwei 
You should change default_options.domain_id in sources in ros2_android_workspace and rebuild sources  + reinstall app.

esteve, 2020-12-22:
@AlexKovalchukSomatic sorry I missed this one. It looks like you figured it out, thank you so much for looking into it. Could you post your changes somewhere? I can update the API signature so that ROS_DOMAIN can be passed as an option so that users don't have to change ros2_android themselves. Thanks.

Luriwei, 2020-12-23:
@AlexKovalchukSomatic After uninstall and reinstall app, i can communicate between two phones. Thanks.

Luriwei, 2020-12-24:
@AlexKovalchukSomatic  Now i can communicate between two phones, but my phone still can not communicate with my unbuntu or my robot, have you encountered this problem before? (My ubuntu can communicate with my robot, they all use the same domain_id and network.) Thanks.

Luriwei, 2020-12-24:
@AlexKovalchukSomatic  After i reset the variable RCL_NODE_OPTIONS_DEFAULT_DOMAIN_ID to SIZE_MAX, i resolved that quesion. Sorry to bother you, thanks again.

cfasana, 2023-12-01:
@esteve **were you able to update the API to allow specifying the ROS_DOMAIN_ID?** I am currently trying to build an Android application that publishes messages on a ROS2 topic, and I would like to read them from a laptop. I am using ROS2 Humble and I was able to let two smartphones exchange messages through ROS2Java. Unfortunately, the same does not hold for the communication with a PC and this could be related to the ROS_DOMAIN_ID.

StevenNick88, 2023-12-01:
您好，您的邮件已收到，感谢您的来信。祝您生活愉快！

mathankumar-dotworld, 2023-12-22:
@esteve @AlexKovalchukSomatic . Please share the Java and C code for setting the domain ID.

jocacace, 2025-03-10:
> [@Luriwei](https://github.com/Luriwei) Yes, I modified code to set DomainID while creating node (You should change signature). You should change "nativeCreateNodeHandle" function:
> 
> > default_options.domain_id = domain;

Hello,

This post was a bit old, but I hope you can reply. I am trying to hardcode the domain id to prevent to share all the topics of TEMI. However, I am not able to find neither the "nativeCreateNodeHandle" function either  the default_options.domain_id line. Can you drive me on how to setup the domain id?