ros2-java/ros2_java #155: Possible to remove commercial dependency on rti-connext-dds-5.3.1

xingjl6280, 2020-12-08:
Hi team,

I'm using the dashing branch.
Everything works fine but below step requires commercial license acceptance

 rosdep install --from-paths src -y -i --skip-keys "ament_tools"

Is there a way to use eProsima’s Fast RTPS to remove the dependency? 

thank you

jacobperron, 2020-12-08:
I believe Fast-RTPS is used by default. Connext is certainly not a requirement. It looks like it is being transitively brought in as a dependency during the `rosdep` install command. Try adding these additional packages to "skip keys":

        rosdep install --from-paths src -y -i --skip-keys "ament_tools rti-connext-dds-5.3.1 rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp"

If that works, we can update the instructions in the README.

xingjl6280, 2020-12-09:
Hi Jacob,

thanks for your quick response.

I've tried your new command to skip these packages.
It works to skip rti-context related dependencies.

But I got below error. 
Actually same error happens no matter skip rti-context or not. 
Previously, i followed update-readme branch successfully built ros2-java. 
But now, after the branch merged into dashing branch, something wrong happened.
Shall I raise a new ticket for the error? Is there a workaround now I can fix the error?

Starting >>> rosidl_generator_java
--- stderr: rosidl_generator_java                                    
In file included from /ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/empty__type_support.cpp:19:0:
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:37:31: error: ‘rosidl_generator_java::msg::dds_’ has not been declared
   rosidl_generator_java::msg::dds_::Empty_ & dds_message);
                               ^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:37:44: error: expected ‘,’ or ‘...’ before ‘&’ token
   rosidl_generator_java::msg::dds_::Empty_ & dds_message);
                                            ^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:46:37: error: ‘dds_’ in namespace ‘rosidl_generator_java::msg’ does not name a type
   const rosidl_generator_java::msg::dds_::Empty_ & dds_message,
                                     ^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:46:50: error: expected unqualified-id before ‘&’ token
   const rosidl_generator_java::msg::dds_::Empty_ & dds_message,
                                                  ^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:46:50: error: expected ‘)’ before ‘&’ token
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:46:52: error: expected initializer before ‘dds_message’
   const rosidl_generator_java::msg::dds_::Empty_ & dds_message,
                                                    ^~~~~~~~~~~
In file included from /ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/defaults__type_support.cpp:19:0:
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:37:31: error: ‘rosidl_generator_java::msg::dds_’ has not been declared
   rosidl_generator_java::msg::dds_::Defaults_ & dds_message);
                               ^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:37:47: error: expected ‘,’ or ‘...’ before ‘&’ token
   rosidl_generator_java::msg::dds_::Defaults_ & dds_message);
                                               ^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:46:37: error: ‘dds_’ in namespace ‘rosidl_generator_java::msg’ does not name a type
   const rosidl_generator_java::msg::dds_::Defaults_ & dds_message,
                                     ^~~~
In file included from /ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/nested__type_support.cpp:19:0:
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:37:31: error: ‘rosidl_generator_java::msg::dds_’ has not been declared
   rosidl_generator_java::msg::dds_::Nested_ & dds_message);
                               ^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:37:45: error: expected ‘,’ or ‘...’ before ‘&’ token
   rosidl_generator_java::msg::dds_::Nested_ & dds_message);
                                             ^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:46:37: error: ‘dds_’ in namespace ‘rosidl_generator_java::msg’ does not name a type
   const rosidl_generator_java::msg::dds_::Nested_ & dds_message,
                                     ^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:46:51: error: expected unqualified-id before ‘&’ token
   const rosidl_generator_java::msg::dds_::Nested_ & dds_message,
                                                   ^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:46:51: error: expected ‘)’ before ‘&’ token
[... cut ...]
                                                     ^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:46:53: error: expected ‘)’ before ‘&’ token
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:46:55: error: expected initializer before ‘dds_message’
   const rosidl_generator_java::msg::dds_::Defaults_ & dds_message,

xingjl6280, 2020-12-09:
Is it due to the recent ros version update?

I compared package with the environment I installed previously before Nov, found below packages got updated 

ros-dashing-ament-pep8                             0.7.12-1bionic.20201125.034318       amd64
ros-dashing-ament-pyflakes                         0.7.12-1bionic.20201125.034705       amd64
ros-dashing-opensplice-cmake-module                0.7.3-1bionic.20201125.042344        amd64
ros-dashing-rosidl-generator-dds-idl               0.7.1-1bionic.20201125.042556        amd64
ros-dashing-rosidl-typesupport-opensplice-c        0.7.3-1bionic.20201125.045517        amd64
ros-dashing-rosidl-typesupport-opensplice-cpp      0.7.3-1bionic.20201125.044859        amd64
ros-dashing-test-interface-files                   0.7.1-1bionic.20201125.034013        amd64

xingjl6280, 2020-12-10:
The command provided could remove the commercial dependency, so close the ticket.
For the build error, will raise a new issue.

closed 2020-12-10 by xingjl6280