ros2-java/ros2_java #156: Dashbing branch build error: ‘rosidl_generator_java::msg::dds_’ has not been declared

xingjl6280, 2020-12-10:
Hi team,

I was able to build and run examples previously followed by the branch: update-readme

But now, for a new environment,  I got below build error. 
I guess it's due to the 25thNov ros upgrade, as I found below package version difference.
Is there a workaround for now?

thank you

Version diff:
ros-dashing-ament-pep8 0.7.12-1bionic.20201125.034318 amd64
ros-dashing-ament-pyflakes 0.7.12-1bionic.20201125.034705 amd64
ros-dashing-opensplice-cmake-module 0.7.3-1bionic.20201125.042344 amd64
ros-dashing-rosidl-generator-dds-idl 0.7.1-1bionic.20201125.042556 amd64
ros-dashing-rosidl-typesupport-opensplice-c 0.7.3-1bionic.20201125.045517 amd64
ros-dashing-rosidl-typesupport-opensplice-cpp 0.7.3-1bionic.20201125.044859 amd64
ros-dashing-test-interface-files 0.7.1-1bionic.20201125.034013 amd64

Build error:
Starting >>> rosidl_generator_java
--- stderr: rosidl_generator_java
In file included from /ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/empty__type_support.cpp:19:0:
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:37:31: error: ‘rosidl_generator_java::msg::dds_’ has not been declared
rosidl_generator_java::msg::dds_::Empty_ & dds_message);
^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:37:44: error: expected ‘,’ or ‘...’ before ‘&’ token
rosidl_generator_java::msg::dds_::Empty_ & dds_message);
^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:46:37: error: ‘dds_’ in namespace ‘rosidl_generator_java::msg’ does not name a type
const rosidl_generator_java::msg::dds_::Empty_ & dds_message,
^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:46:50: error: expected unqualified-id before ‘&’ token
const rosidl_generator_java::msg::dds_::Empty_ & dds_message,
^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:46:50: error: expected ‘)’ before ‘&’ token
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/empty__rosidl_typesupport_opensplice_cpp.hpp:46:52: error: expected initializer before ‘dds_message’
const rosidl_generator_java::msg::dds_::Empty_ & dds_message,
^~~~~~~~~~~
In file included from /ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/defaults__type_support.cpp:19:0:
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:37:31: error: ‘rosidl_generator_java::msg::dds_’ has not been declared
rosidl_generator_java::msg::dds_::Defaults_ & dds_message);
^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:37:47: error: expected ‘,’ or ‘...’ before ‘&’ token
rosidl_generator_java::msg::dds_::Defaults_ & dds_message);
^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:46:37: error: ‘dds_’ in namespace ‘rosidl_generator_java::msg’ does not name a type
const rosidl_generator_java::msg::dds_::Defaults_ & dds_message,
^~~~
In file included from /ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/nested__type_support.cpp:19:0:
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:37:31: error: ‘rosidl_generator_java::msg::dds_’ has not been declared
rosidl_generator_java::msg::dds_::Nested_ & dds_message);
^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:37:45: error: expected ‘,’ or ‘...’ before ‘&’ token
rosidl_generator_java::msg::dds_::Nested_ & dds_message);
^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:46:37: error: ‘dds_’ in namespace ‘rosidl_generator_java::msg’ does not name a type
const rosidl_generator_java::msg::dds_::Nested_ & dds_message,
^~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:46:51: error: expected unqualified-id before ‘&’ token
const rosidl_generator_java::msg::dds_::Nested_ & dds_message,
^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:46:51: error: expected ‘)’ before ‘&’ token
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/nested__rosidl_typesupport_opensplice_cpp.hpp:46:53: error: expected initializer before ‘dds_message’
const rosidl_generator_java::msg::dds_::Nested_ & dds_message,
^~~~~~~~~~~
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:46:53: error: expected unqualified-id before ‘&’ token
const rosidl_generator_java::msg::dds_::Defaults_ & dds_message,
^
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:46:53: error: expected ‘)’ before ‘&’ token
/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/defaults__rosidl_typesupport_opensplice_cpp.hpp:46:55: error: expected initializer before ‘dds_message’
const rosidl_generator_java::msg::dds_::Defaults_ & dds_message,

ahmadnav, 2020-12-14:
I am getting the same error, any luck?

xingjl6280, 2020-12-14:
no luck at all :( I'm giving up and considering to use the ros1 solution: zmq + protobuf + zeroconfWe spent too much time on dashing java and Android client-------- 原始邮件 --------发件人： ahmadnav <notifications@github.com>日期： 2020年12月14日周一 11:42收件人： ros2-java/ros2_java <ros2_java@noreply.github.com>抄送： xingjl6280 <xingjl6280@163.com>, Author <author@noreply.github.com>主    题： Re: [ros2-java/ros2_java] Dashbing branch build error: ‘rosidl_generator_java::msg::dds_’ has not been declared (#156)
I am getting the same error, any luck?

esteve, 2020-12-14:
@xingjl6280 the opensplice rmw implementation is no longer supported in ROS 2 (or at least it's not actively maintained), could you try removing `ros-dashing-opensplice-cmake-module ` and any of the other opensplice packages? Thanks.

xingjl6280, 2020-12-15:
thanks Steve,

it's working now, by removing dependence <test_depend>rosidl_typesupport_opensplice_c</test_depend>
from ros2_java/ros2_java/rosidl_generator_java/package.xml

closed 2020-12-15 by xingjl6280