ros2-java/ros2_java #157: Dashing branch ros_java build error

ahmadnav, 2020-12-13:
Starting >>> rosidl_generator_java
--- stderr: rosidl_generator_java                          
CMake Error at /home/loacaladmin/ros2_dashing/ros2-linux/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:86 (message):
  rosidl_generate_interfaces() the passed file
  '/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/share/test_interface_files/msg/Arrays.msg'
  doesn't exist
Call Stack (most recent call first):
  CMakeLists.txt:54 (rosidl_generate_interfaces)

ahmadnav, 2020-12-13:
When using the provided ci script build_java.sh this error occurs

Traceback (most recent call last):
  File "/home/loacaladmin/ros2_java/ci-scripts/output/install_isolated_ament/ament_tools/bin/ament", line 33, in <module>
    sys.exit(load_entry_point('ament-tools', 'console_scripts', 'ament')())
  File "/home/loacaladmin/ros2_java/ci-scripts/output/install_isolated_ament/ament_tools/bin/ament", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/home/loacaladmin/.local/lib/python3.6/site-packages/importlib_metadata/__init__.py", line 573, in distribution
    return Distribution.from_name(distribution_name)
  File "/home/loacaladmin/.local/lib/python3.6/site-packages/importlib_metadata/__init__.py", line 203, in from_name
    raise PackageNotFoundError(name)
importlib_metadata.PackageNotFoundError: No package metadata was found for ament-tools

ahmadnav, 2020-12-14:
Can be resolved by modifying ros2_dashing/ros2-linux/share/test_interface_files/cmake/test_interface_files-extras.cmake to reflect proper paths.

jacobperron, 2020-12-14:
What method of installing ROS 2 are you using? It sounds like you are following the ["install from fat archive" instructions](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Binary/).

ahmadnav, 2020-12-14:
@jacobperron Yes due to #158 I am running an older verstion of dashing (7)

ahmadnav, 2020-12-14:
But I guess all this is irrelevant if opensplice dependancies are removed entirely from both ros_java and ros_android

jacobperron, 2020-12-14:
We can certainly remove OpenSplice as a default dependency from rcljava (and we have in https://github.com/ros2-java/ros2_java/pull/163). AFAIK, OpenSplice is still a supported RMW for Dashing, so it still might be worth investigating this issue. Though since it has been dropped from ROS 2 Eloquent, I'd say it is not high-priority.

jacobperron, 2021-11-13:
Since Dashing is EOL, I'm going to close this. I encourage you to try the `main` branch with ROS Galactic :)

closed 2021-11-13 by jacobperron