ros2-java/ros2_java #158: ROS2 Dashing fresh install, tons of errors on build with colcon

ahmadnav, 2020-12-14:
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:458:19: error: request for member ?uint16_values_default_? in ?dds_message?, which is of non-class type ?int?
       dds_message.uint16_values_default_[i] = ros_message.uint16_values_default[i];
                   ^~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:468:17: error: request for member ?int32_values_default_? in ?dds_message?, which is of non-class type ?int?
     dds_message.int32_values_default_.length(length);
                 ^~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:470:19: error: request for member ?int32_values_default_? in ?dds_message?, which is of non-class type ?int?
       dds_message.int32_values_default_[i] = ros_message.int32_values_default[i];
                   ^~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:480:17: error: request for member ?uint32_values_default_? in ?dds_message?, which is of non-class type ?int?
     dds_message.uint32_values_default_.length(length);
                 ^~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:482:19: error: request for member ?uint32_values_default_? in ?dds_message?, which is of non-class type ?int?
       dds_message.uint32_values_default_[i] = ros_message.uint32_values_default[i];
                   ^~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:492:17: error: request for member ?int64_values_default_? in ?dds_message?, which is of non-class type ?int?
     dds_message.int64_values_default_.length(length);
                 ^~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:494:19: error: request for member ?int64_values_default_? in ?dds_message?, which is of non-class type ?int?
       dds_message.int64_values_default_[i] = ros_message.int64_values_default[i];
                   ^~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:504:17: error: request for member ?uint64_values_default_? in ?dds_message?, which is of non-class type ?int?
     dds_message.uint64_values_default_.length(length);
                 ^~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:506:19: error: request for member ?uint64_values_default_? in ?dds_message?, which is of non-class type ?int?
       dds_message.uint64_values_default_[i] = ros_message.uint64_values_default[i];
                   ^~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:516:17: error: request for member ?string_values_default_? in ?dds_message?, which is of non-class type ?int?
     dds_message.string_values_default_.length(length);
                 ^~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:518:19: error: request for member ?string_values_default_? in ?dds_message?, which is of non-class type ?int?
       dds_message.string_values_default_[i] = ros_message.string_values_default[i].c_str();
                   ^~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:522:15: error: request for member ?alignment_check_? in ?dds_message?, which is of non-class type ?int?
   dds_message.alignment_check_ = ros_message.alignment_check;
               ^~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp: In function ?const char* rosidl_generator_java::msg::typesupport_opensplice_cpp::publish__UnboundedSequences(void*, const void*)?:
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:534:3: error: ?__dds_msg_type_UnboundedSequences? was not declared in this scope
   __dds_msg_type_UnboundedSequences dds_message;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:534:3: note: suggested alternative: ?__ros_msg_type_UnboundedSequences?
   __dds_msg_type_UnboundedSequences dds_message;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   __ros_msg_type_UnboundedSequences
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:535:43: error: ?dds_message? was not declared in this scope
   convert_ros_message_to_dds(ros_message, dds_message);
                                           ^~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:535:43: note: suggested alternative: ?ros_message?
   convert_ros_message_to_dds(ros_message, dds_message);
                                           ^~~~~~~~~~~
                                           ros_message
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:537:37: error: ?UnboundedSequences_DataWriter? is not a member of ?rosidl_generator_java::msg::dds_?
   rosidl_generator_java::msg::dds_::UnboundedSequences_DataWriter * data_writer =
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:537:69: error: ?data_writer? was not declared in this scope
   rosidl_generator_java::msg::dds_::UnboundedSequences_DataWriter * data_writer =
                                                                     ^~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/multi_nested__type_support.cpp:690:42: note: suggested alternative: ?take__MultiNested?
   DDS::OpenSplice::CdrTypeSupport cdr_ts(__type_support_MultiNested);
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                          take__MultiNested
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/multi_nested__type_support.cpp:692:66: error: ?dds_message? was not declared in this scope
   DDS::ReturnCode_t status = cdr_ts.deserialize(buffer, length, &dds_message);
                                                                  ^~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:537:69: note: suggested alternative: ?u_writer?
   rosidl_generator_java::msg::dds_::UnboundedSequences_DataWriter * data_writer =
                                                                     ^~~~~~~~~~~
[... cut ...]
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:1029:66: error: ?dds_message? was not declared in this scope
   DDS::ReturnCode_t status = cdr_ts.deserialize(buffer, length, &dds_message);
                                                                  ^~~~~~~~~~~
/home/loacaladmin/ros2_java_ws/build/rosidl_generator_java/rosidl_typesupport_opensplice_cpp/rosidl_generator_java/msg/dds_opensplice/unbounded_sequences__type_support.cpp:1029:66: note: suggested alternative: ?v_message?
   DDS::ReturnCode_t status = cdr_ts.deserialize(buffer, length, &dds_message);

ahmadnav, 2020-12-14:
Running ROS 2 Dashing Diademata - Patch Release 7 seems to resolve this.

However,

--- stderr: rosidl_generator_java                          
CMake Error at /home/loacaladmin/ros2_dashing/ros2-linux/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:86 (message):
  rosidl_generate_interfaces() the passed file
  '/home/jenkins-agent/workspace/ci_packaging_linux/ws/install/share/test_interface_files/msg/Arrays.msg'
  doesn't exist
Call Stack (most recent call first):
  CMakeLists.txt:54 (rosidl_generate_interfaces)

still persists.

ahmadnav, 2020-12-14:
Can be resolved by modifying ros2_dashing/ros2-linux/share/test_interface_files/cmake/test_interface_files-extras.cmake to reflect proper paths.

jacobperron, 2021-11-13:
If I'm not mistaken, the current `dashing` branch builds successfully. If I am mistaken, pull requests fixing the build are welcome :)

I'm going to close this ticket, since our focus is now on supporting the latest available ROS 2 release (Galactic).

closed 2021-11-13 by jacobperron