ros2-java/ros2_java #159: ROS2 Bouncy, Ubuntu 16.04, compile errors

ahmadnav, 2020-12-14:
home/ubuntu/ros2_java/ci-scripts/output/build_isolated_java/action_msgs/rosidl_generator_java/action_msgs/msg/GoalInfo.java:51: error: package unique_identifier_msgs.msg does not exist
  private unique_identifier_msgs.msg.UUID goal_id = new unique_identifier_msgs.msg.UUID();
                                    ^
/home/ubuntu/ros2_java/ci-scripts/output/build_isolated_java/action_msgs/rosidl_generator_java/action_msgs/msg/GoalInfo.java:53: error: package unique_identifier_msgs.msg does not exist
  public GoalInfo setGoalId(final unique_identifier_msgs.msg.UUID goal_id) {
                                                            ^
/home/ubuntu/ros2_java/ci-scripts/output/build_isolated_java/action_msgs/rosidl_generator_java/action_msgs/msg/GoalInfo.java:59: error: package unique_identifier_msgs.msg does not exist
  public unique_identifier_msgs.msg.UUID getGoalId() {
                                   ^
/home/ubuntu/ros2_java/ci-scripts/output/build_isolated_java/action_msgs/rosidl_generator_java/action_msgs/msg/GoalInfo.java:51: error: package unique_identifier_msgs.msg does not exist
  private unique_identifier_msgs.msg.UUID goal_id = new unique_identifier_msgs.msg.UUID();

esteve, 2020-12-15:
@ahmadnav what branch are you trying this on? `dashing` or `master`?

ahmadnav, 2020-12-17:
@esteve Master

esteve, 2020-12-18:
@ahmadnav I've changed the repos file to pin `rcl_interfaces` to an earlier version that does not contain the action messages, as actions are not supported in the master version of ros2_java. Could you try again? Your local copy of `rcl_interfaces` should now point to `db27f0e8619460848d80c1442f7fec0c56ee63e5` Thanks.

jacobperron, 2021-11-13:
I'm going to close this since we no longer support Bouncy.

closed 2021-11-13 by jacobperron